bigtreetech/BIGTREETECH-TouchScreenFirmware #1608: Probe Z Offset Menu Issue

JerksonS, 2021-02-06:
In touch screen mode, during the Z probe offset process:

After probing the nozzle moves the OPPOSITE direction of the probe offsets in the firmware.

To explain, the probe probes the bed in the very center. Perfect. After probing the hotend should move such that the nozzle is now where the probe was, to confirm nozzle:probe offset at that center point. However, instead of the hotend moving towards the front right, it moves back left. So it is moving the correct XY offset amount but in exactly the wrong direction. It should move 24, -47 (x,y)

kisslorand, 2021-02-06:
You probably defined your probe offset wrong. Define them the other way around, if it's (24, -47) than define it as (-24, 47), if it's (-24, 47) than define it as (24, -47).

JerksonS, 2021-02-06:
Thanks for the reply but they're set correctly.

The machine is a Biqu B1 and it comes with a metal bracket for the Bltouch. In their stock firmware, as well as mine, we have it defined as:

`Assuming the typical work area orientation:
 *  - Probe to RIGHT of the Nozzle has a Positive X offset
 *  - Probe to LEFT  of the Nozzle has a Negative X offset
 *  - Probe in BACK  of the Nozzle has a Positive Y offset
 *  - Probe in FRONT of the Nozzle has a Negative Y offset

 *     +-- BACK ---+
 *     |    [+]    |
 *   L |        1  | R <-- Example "1" (right+,  back+)
 *   E |  2        | I <-- Example "2" ( left-,  back+)
 *   F |[-]  N  [+]| G <-- Nozzle
 *   T |       3   | H <-- Example "3" (right+, front-)
 *     | 4         | T <-- Example "4" ( left-, front-)
 *     |    [-]    |
 *     O-- FRONT --+
 */
#define NOZZLE_TO_PROBE_OFFSET { 24, -47, -4.5 }`

Ignoring the Z offset parameter, since that will vary depending on the screw-spring tightness on the bltouch.

The BLtouch is indeed to the RIGHT of the nozzle, hence the positive X offset, and in FRONT of the nozzle, hence the negative Y offset.

The leveling process understands the offset correctly - if it didnt it would try to probe off the bed. It is the LCD-driven function of Z offset that moves the wrong direction.

digant73, 2021-02-06:
the Probe Offset menu now always netrulizes the configured z offset. It means the initial Z position matches the probing point (0).
This was made because some users complained of the nozzles touched the bed (because their configured Z offset was wrong). If you have a configured z offset to -0.8 you now has to lower again to -0.8 to have again the original absolute Z0. -0.7 means that you raised the absolute z0 of 0.1 (compared to -0.8)

kisslorand, 2021-02-06:
Yup, you're right, {24, -47} is correct for right front position of BLTouch.
I just adjusted Z probe yesterday, it was working as intended, nozzle gone exactly to the BLTouch position for measurements.
I am puzzled how can it work the opposite on two machines...

JerksonS, 2021-02-06:
Again, I have NO PROBLEM with the Z offset.

When you press the Z offset button (bottom right corner) The sequence of events SHOULD BE:

1 - Home all axes.
2 - Probe in center of bed.
3 - Move XY to place the nozzle where the probe just was - at the center of the bed
4 - Then allow the user to select the z offset.

#3 is moving the opposite direction. If the probe XY offset is 24,-47, then the hotend should move +24mm X and -47mm Y to then place the nozzle in the center of the bed where the probe just probed.

The nozzle is moving -24x and +47y, resulting in the nozzle being closer to the back left corner than to the center.

Could you point me to the block of code that handles this sequence of Gcode commands behind this button?

JerksonS, 2021-02-06:
Edit - found it Line 21-24 in here - [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/ProbeOffsetControl.c]
```
mustStoreCmd("G91\n");  // set relative position mode
    mustStoreCmd("G1 X%.2f Y%.2f F%d\n",
                 getParameter(P_PROBE_OFFSET, X_STEPPER),
                 getParameter(P_PROBE_OFFSET, Y_STEPPER),
```

So now i need to figure out why it's retrieving incorrect values in the getParameter functions

JerksonS, 2021-02-06:
So I tracked it back to MachineParameters.c, I think...

`{"M851 X%.2f\n",               "M851 Y%.2f\n",               "M851 Z%.2f\n",               NULL,                NULL},                        //Probe offset`

M851 returns 

`Probe Offset X24.00 Y-47.00 Z-2.90`

Which is correct.

digant73, 2021-02-06:
Which values do you see in the machine->parameter menu for probe offset item?

@Kisslorand yes, after I read again the bu description I removed my first comment

JerksonS, 2021-02-06:
In Machine - Parameter I see the correct values 
X 24.00
Y -47.00

JerksonS, 2021-02-06:
I realized I'm using a build of the fw that BTT shipped with the printer - Dated Aug 2020 - so maybe this was fixed since then?

I'm downloading the new firmware from this repo and re-flashing.

JerksonS, 2021-02-06:
UGH! what a waste of everyone's time. Sorry guys. The new firmware works properly.

Thanks again!

closed 2021-02-06 by JerksonS

kisslorand, 2021-02-06:
Just as I said, yesterday it was just fine. :)

Eibenger, 2023-04-03:
@JerksonS 
Hallo,
habe dasselbe Problem. Kannst du bitte den Link für die neue Firmware angeben? Habe SKR 2, und die passende TFT- Variante dazu? L-G., Gerhard