bigtreetech/BIGTREETECH-TouchScreenFirmware #1610: TFT70 No printer found.

terencepeace, 2021-02-07:
--
I have a CR-6 SE with a BTT CR6 V1 installed.

I connected my TFT70 and flashed the CR6 V 1 with firmware inside the folder (BTT-Touch Screen + 12864-CR10_STOCKDISPLAY)

I then flashed the TFT70 with BIGTREE_TFT70_V3.0.26.x.bin and config.ini and THEME_Unified Menu Material theme.
Everything worked great.

Come to print something today. And I get "No Printer Found"
I have checked all the wires, I can see nothing wrong.

So I checked the CR6 V1 motherboard by flashing the  firmware inside the folder firmware(CR6 Original Touch Screen).
The touch did not work, but in the end I tested it. The CR6 V1 board is ok.

So I disconnected the CR-6 SE screen and reconnected the TFT70 and flashed the fiemware in the folder (BTT-Touch Screen + 12864-CR10_STOCKDISPLAY).

It still says "No Printer Found.
Any help would be great.
Thank you for your time.

oldman4U, 2021-02-07:
Cable issue
Baud rate does not match
Wrong serial port in Marlin

terencepeace, 2021-02-08:
> Cable issue
> Baud rate does not match
> Wrong serial port in Marlin

Thank you for your help.

It was working great, but the very next time I switched it on it doesn't.
I have tried 2 new cables.

How do I check the serial port?
What would change that as it worked?
Thank you

oldman4U, 2021-02-08:
I guess you also believe that what you describe sounds like a hardware issue. But this repository is about the firmware of the touchscreen so I told you about the possible firmware related issues.

Serial port is defined in Marlin.

Please help the community and close the ticket once you do not need it anymore. Thank you

terencepeace, 2021-02-08:
> I guess you also believe that what you describe sounds like a hardware issue. But this repository is about the firmware of the touchscreen so I told you about the possible firmware related issues. [...]

What should the serial port be set to?
I know the old way of editing marlin with arduino.
But not this new way.

Thank you.

oldman4U, 2021-02-08:
This is a Marlin and therefore a mainboard setting and because of this outside of the TFT firmware.

But you wrote you use CR10 display and if this is the correct setting I guess the display is connected using EXP3 and if this is the case I guess the correct setting could be 1 or 2.

Hope this helps

terencepeace, 2021-02-08:
> This is a Marlin and therefore a mainboard setting and because of this outside of the TFT firmware.
>
> But you wrote you use CR10 display and if this is the correct setting I guess the display is [...]

Thank you.
Yes I used the "BTT-Touch Screen + 12864-CR10_STOCKDISPLAY" firmware that was inside.
The other firmware as you know is for the standard CR-6 SE screen.

I will try to find someone that will work this out for me.
On a side note. To make this more popular.
Wouldn't be better to have a simple GUI to edit things so even old people like me don't get put off?

oldman4U, 2021-02-08:
The TFT firmware can be configured editing the config.ini file like described in the ReadMe on the main page.

Do not believe something similar is available for Marlin. 

Greetings 

Oldman😁

terencepeace, 2021-02-08:
> The TFT firmware can be configured editing the config.ini file like described in the ReadMe on the main page.
> 
> Do not believe something similar is available for Marlin.
> 
> Greetings
> 
> Oldman😁

Oldman  :)
It WILL happen to you one day  ;)

Can you link me to older TFT70 updates please?

oldman4U, 2021-02-08:
It already happened to me as you can see in my nickname.

You can find all TFT firmware versions here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commits/master

And there is also a GitHub for the mainboard: https://github.com/bigtreetech/BIGTREETECH-SKR-CR6

Seems it has RS232 and EXP3 so 2 cables connected to TFT.

terencepeace, 2021-02-08:
> It already happened to me as you can see in my nickname.
>
> You can find all TFT firmware versions here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commits/master
>
> And there is [...]

Thank you very much for your help.

I tried some older versions, but I still get the No Printer found error.
Time to send it back and move on. Maybe one day they will have a simple GUI that everyone and his dog can use.

Again.
Thank you.

oldman4U, 2021-02-08:
Like mentioned at the very beginning, sounds like a hardware issue.

Please close this ticket

Thank you

terencepeace, 2021-02-09:
> Like mentioned at the very beginning, sounds like a hardware issue.
> 
> Please close this ticket
> 
> Thank you

Hi.
So I am reporting back.

There was a bad earth on the motherboard and the BTT cables are crap.
But I can't enabled ABL. It just throws up and error when I turn it on and if I try to turn it on in the menu.
Once the ABL is fixed I can use it  :)

Thanks old man

terencepeace, 2021-02-09:
All soted.

closed 2021-02-09 by terencepeace