bigtreetech/BIGTREETECH-TouchScreenFirmware #1611: [BUG] Btt Smart Sensor Does Not Work on the Tft35 E3 V3 with Marlin 2.0.7 (Works with 2.0.6)

YBeyin, 2021-02-07:
I just bought brand new Btt Smart Filament Sensor for my Ender 3 Pro. Everything went perfect untill test it. Upgraded my motherboard's firmware, Tft35's firmware but sensor would not work. It was like dead. At first i had a lot of panic. I did pull off the cable of sensor even my printer while working, there is nothing happened. My printer keept working.

Digged in the github and searches on the internet not found a solution. So i decided to contact with the Rob from 3D Printscape youtube channel, asked for his help and he helped a lot. At final he shared his marlin files with me. The result was incredible. My sensor begin to work perfectly. Then we realise all that was issue about the marlin itself. My sensor did not work with 2.0.7 but worked perfectly with the 2.0.6

Now my sensor plugged on the Tft's Fil-Det slot, using Tft's SdCard input, using in touch mode (not in marlin) and everything perfect.

I just want to let you know if you are having issue about your sensor, try it with marlin 2.0.6

My Printer's Specs:
Ender 3 Pro
BL-Touch
Btt Skr Mini E3 V2.0
Btt Tft35 E3 V3.0
Btt Smart Filament Sensor (Plugged on the tft's Fil-Det slot)

### Steps to reproduce

1. Plugged the sonsor
2. Upgraded the firmware of motherboard
3. Upgraded the firmware of tft.

### Hardware Variant
Btt TFT35 E3 V3
### TFT Firmware Version & Main Board Firmware details

DaStivi, 2021-02-21:
look at the other SFS issue...   when you upgraded to 2.0.7 did you enabled linear advanced eventually? i think SFS does not work with LA enabled..

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1612#issuecomment-782926663

killrob, 2021-03-23:
i made this work, see here
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/907

oldman4U, 2021-03-25:
The information that it does not work is nonsense and could lead users into the wrong direction. Please close this ticket.

closed 2021-04-05 by bigtreetech