bigtreetech/BIGTREETECH-TouchScreenFirmware #1612: [BUG] Filament sensor does not work after the update

Chris1231, 2021-02-07:
i use that BTT TFT 35 E3 V3.0
after the update to the version 3.0.26.x it is no longer recognized that there is a filament in the filament sensor.
no matter whether filament is inserted or not
I use a simple sensor (not a smart one) connected directly to the TFT.
with the old version 3.0.25.3 it worked fine.

do I have to use other settings compared to the old version?

I hope you can help me

YBeyin, 2021-02-07:
Thats about the marlin. Lol i just open a topic about that bug. Compile your motherboard's firmware from Marlin 2.0.6 thats works.

Chris1231, 2021-02-07:
Thanks for the quick reply, I've tried it, but it's still the same problem :(

YBeyin, 2021-02-07:
> Thanks for the quick reply, I've tried it, but it's still the same problem :(

Are you plugged your sensor on motherboard or Tft?

Chris1231, 2021-02-07:
connected directly to the TFT

YBeyin, 2021-02-07:
> connected directly to the TFT

I am pretty sure something wrong about pins. Can you check your pins with multimeter if there is voltage?

Chris1231, 2021-02-07:
i checked.

at the pin PA15 with filament 0V and without filament 3.3V

Apfelbrot34, 2021-02-07:
Good evening I have the same problem too. My hardware is: SKR 1.4, TFT35 V3.0 and BTT Smart Runout Sensor. The sensor is connected to the TFT. The signal output is from 0V to 2.6V. input voltage is at 3.3V at Fil_Det. Firmware is Marlin 2.0 bug fix 20008 according to config. TFT is V3.0.26.x. The sensor is simply not recognized by the TFT. If the plug is unplugged, there is no runout signal or even True or False, there is no reaction. M114 E we output from the Linear Advanced board is switched off.

Grex3D, 2021-02-08:
Same issue, more easy to use with the dedicated  fil-det pin on screen (tft70) but nothing work....

YBeyin, 2021-02-09:
> Good evening I have the same problem too. My hardware is: SKR 1.4, TFT35 V3.0 and BTT Smart Runout Sensor. The sensor is connected to the TFT. The signal output is from 0V to 2.6V. input voltage is [...]

Can you try marlin 2.0 bug fix 20006? i am running my btt sensor on my tft with 20006

Apfelbrot34, 2021-02-09:
Good evening.

marlin 2.0 bug fix 20006 I tried it first without success. The same mistake always came. 
Then I read that it was a Marlin error and so I tried marlin 2.0 Bug fix 20008 with the same problem again !. 

I could now grasp the problem in more detail. Whenever the error comes up with runout. I have now looked in the terminal. And see there it is at M114 Count E: -912. So a minus movement. 

I work with the slicer: simplify3d. Since I had also read that there is a solution. Using the relative position of the extruder. But no, that didn't help either. I just found a function in simplify3d. If I turn it off it just works, I just have to continue testing. The function is called: Force Retraction between layers.

If I turn the function off it works, if I turn it on it doesn't work.

Here the Gcode has no function:

G1 X93.322 Y75.737 E1.7630
G1 X93.049 Y75.679 F11520
G1 X92.899 Y76.383
G92 E0.0000
G1 X93.049 Y75.679 E-0.3751 F11520
G1 X93.495 Y75.773 E-0.6126
G1 X93.345 Y76.477 E-0.9871
G1 X93.792 Y76.570 E-1.2246
G1 X93.927 Y75.935 E-1.5625
G1 E-10.0000 F6000
; layer 3, Z = 0.450
M104 S215 T0
; feature inner perimeter
G1 Z0.800 F4602
G1 X108.127 Y91.498 F9000
G1 Z0.450 F4602
G1 E0.0000 F6000
G92 E0.0000
G1 X89.149 Y91.168 E0.5398 F5616
G1 X92.647 Y74.711 E1.0182
G1 X110.872 Y78.585 E1.5481
G1 X108.335 Y90.520 E1.8951
G1 X108.127 Y91.498 F5616
G1 X105.127 Y91.446 F5616
; feature outer perimeter
G1 X107.939 Y91.951 F9000
G92 E0.0000

and here without:

G1 X93.322 Y75.737 E1.7630
G1 X93.049 Y75.679 F11520
G1 X92.899 Y76.383
G1 X93.049 Y75.679 F11520
G1 X93.495 Y75.773
G1 X93.345 Y76.477
G1 X93.792 Y76.570
G1 X93.927 Y75.935
; layer 3, Z = 0.450
M104 S215 T0
; feature inner perimeter
G1 Z0.450 F4602
G1 X108.127 Y91.498 F9000
G92 E0.0000
G1 X89.149 Y91.168 E0.5398 F5616
G1 X92.647 Y74.711 E1.0182
G1 X110.872 Y78.585 E1.5481
G1 X108.335 Y90.520 E1.8951
G1 X108.127 Y91.498 F5616
G1 X105.127 Y91.446 F5616
; feature outer perimeter
G1 X107.939 Y91.951 F9000
G92 E0.0000

I have a bowden printer.

It seems that there is a problem with the G92 E0 and M114 Count E :. With retract and zeros. I will try this again with marlin 2.0 bug fix 20006.

I apologize for my english right away. But I come from Germany and use Google translator.

Apfelbrot34, 2021-02-11:
So am back with more results. I installed Marlin 2.0 bug fix 20006 and tried again. The bottom line is: it's not Marlin for me at least. Well, if you enable Advanced in both Linear, both don't work. But my reason problem is the firmware of the Bigtreetech TFT35 V3. 

My technique:

SKR 1.4 Marlin 20006
TFT35 V3 Firmware Master CONFIG_VERSION 20210124
BTT Smart Filament Runout

Silcer:
Simplify3d

So now to my solution which is turned off but maybe so that the functions work again.

So I had problems that the sensor has reacted during printing. With the message I looked into the terminal and looked at the M114 E > Count E:. It came each time a Count E: -912 value. So there was always a minus value.

So by searching around and looking at the Gcode I found out. That it was a problem with the E values and G92 E0.0.

So I looked at Simplify3d and turned off the following function:

Wipe Distance 
force retraction between layers

By switching off these two functions no minus steps come in the pressure and the error Count E:-900 minus values did not come any more. 

I could thus print normally and also so that the printer triggers when you deliberately force it with your hands. 

Conclusion: 
It is not because of Marlin with me! Except for Linear advanced!
The firmware of the TFT35 V3 would have to be tackled again. Because the minus values for the release are the reason. Probably the code would have to be adapted again. That he must respond to the Fil_Det pin input signal rather than Count E: or at least plus and minus values are taken into account.  Which are generated by the Silcer by certain extra function. 

The reason is: He drives minus value because of the function and then of course the M114 reads the step and Bam already is a minus value there. 

It still has improvement potential

DaStivi, 2021-02-13:
my filament sensor doesn't work too here.. i've some assumption, but i've to test this... 
@chris1231 did you use the latest nightly build from 6.2.2021?

edit: ok strange thing i did tested with an older nightly build, where i'am almost sure it worked... as i got falsh positive fil_runout warnings...  but even with older firmware i don't get any warning and there is no filament through the sensor :/

i really try to use ah stable tft firmware later... as i got so many issues lately with the nightly builds its mindboggling.

Budis3, 2021-02-18:
I think I have found the problem of this bug. I am probed a lot of firmware on SKR mini E3 v2.0 and TFT35 E3 V3.0 with the sensor connect to the mainboard. It works in Marlin mode, but in Graphycal no unless you print a file of de sd card of the main board. I think that the problem is that the firmware of the TFT and Marlin is not prepared to be able to connect some external devices and manages them in both modes. If you use the Marlin mode you must connect the sensor to de mainboard and if you use the TFT mode you must connect the sensor to the TFT. 

Maybe this gone to be crazy, but, what would happend if I take the signal wired and connect to the mainboard and the TFT? Did you think that it would be a succesful solution? Or maybe it would be better find a solution by the firmware?

DaStivi, 2021-02-18:
do you guys use linear advanced in Marlin??

kisslorand, 2021-02-19:
@Budis3 
You are on a wrong track.
If the FR sensor is connected to the motherboard than it's Marlin's job to report the sensor's state. Marlin will do that only if  a  print job is ongoing. When you print from TFT, Marlin has no way to know that you started a print job unless you tell him. You can do that by sending M75 at the start of the print. You can put M75 in your start gCode in your slicer or your TFT's start gCode.

DaStivi, 2021-02-19:
so guys basically i had ah chat about the SFS not working and Linear Advanced with @looxonline ... and he mentioned following: 

> The TFT measures the amount of extruded filament.
> Using an E0 counter.
> Marlin does not obey the TFT 100% when LA is active because it goes a little faster and a little slower sometimes.
> This gets reported back to the TFT via action commands.
> It seems as if one of those commands was resetting the E0 counter.
> I would need to debug the TFT in real time to confirm using an ST link.
> However it's a really old bug so I would expect it to have been fixed.

and he also suggested me to connect the sfs to the skr board and use the marlin features of sfs... i will try this... but i'd might also  cross check to disable LA in marlin again too see if SFS starts working again... 

but at least for me its sounds legit what he's pointing too, and also i found a couple of older issues, where this was also mentioned by someone that with LA enabled SFS stops working... but there wasn't any further discussion gooing on..

https://l.facebook.com/l.php?u=https%3A%2F%2Fgithub.com%2Fbigtreetech%2FBIGTREETECH-TouchScreenFirmware%2Fissues%2F312%3Ffbclid%3DIwAR31WcXCZgn6nPIBvErzdEh8lZdyx3cHhz8WPqymSenyWWAUQgqlSHvXUSo%23issuecomment-611780891&h=AT15Z9X7lDNYq55hjjBM0dReSQ3w9XbZj-_yPpXQ8e2ld_OWk14XbJSGhWEGOq-eAwwJkz8tRvQBSic3wipH0iTQkoi7iQUs3u8RmSpEVxjhixHSpdFQg3Hcis0sw2MJhZg

Budis3, 2021-02-19:
@kisslorand 
Ok I prove the start G-code of M75 to print with the runout sensor connected to the mainboard and a pendrive with the file to print connected to the touchscreen and it works, but... there is another problem. When I charge the filament (obiusly putting first the filament through the sensor) it start to purge, but after the purge the touchscreen shows constantly a message that says "the print was stopped by user" with a "ok" button. I suggest that if you pussed "ok" on the screen the printing would restart, but no, the message continues showing on the screen. Any idea?

kisslorand, 2021-02-19:
You should open a new ticket with the new issue and with more details (steps to reproduce, HW used, etc) or check if it has been already discussed in another ticket.

Budis3, 2021-02-19:
@kisslorand 
I will open a new issue ticket for the case that said before. But it's importart to remember that the fact is if the runout sensor doesn't work properly and here is a bug to fix. Also, I remember you, that I said in my two comments before the hardware and how i connect them.
But in other case I will open a new case and I will post in this coment if someone have the same problem

DaStivi, 2021-02-21:
> do you guys use linear advanced in Marlin??

for my point i did tried to add the M75 into start code.. but even then with LA enabled, SFS does not trigger ah runout...

YBeyin, 2021-02-22:
> > do you guys use linear advanced in Marlin??
> 
> for my point i did tried to add the M75 into start code.. but even then with LA enabled, SFS does not trigger ah runout...

yes but i am not sure it is the reason or not.

DaStivi, 2021-02-22:
> > > do you guys use linear advanced in Marlin??
> > 
> > 
> > for my point i did tried to add the M75 into start code.. but even then with LA enabled, SFS does not trigger ah runout...
> 
> yes but i am not sure it is the reason or not.

i had SFS working for sure... got some false positives... then i've enabled linear advanced (really in marlin, having LA enabled but kfactor to 0 also breaks SFS!) and it stopped working... and i even wondered how i didn't get any false positives anymore... 🤦‍♂️

so my next step will be to connect the SFS to the board directly... as there isn't any benefit from having it on the TFT so far i can see, and then enable the filament_runout features in marlin and it should good to go even with linearadvance! but still have todo this sometimes and i do sit all the night long with regular filament dial in's and prints and don't wanna hassle again days and days with such and new bugs then 🙈

baikal, 2021-02-28:
I was having these issues too and basically **gave up** using a rotating filament sensor and linear advance together. Iirc, my findings so far were:
- When smart filament sensor is enabled in TFT (and the sensor is connected to the TFT obviously), the TFT asks `M114 E` every second or so to know the real E-coordinate of the machine
- When linear advance is enabled in Marlin, that `M114 E` constantly reports 0 (zero) as if he extruder would never move, therefore the TFT is not even expecting the sensor to move. Even when K-value is set to 0 (de-facto disabling linear advance), `M114 E` still reports 0 all the time. I guess that is a bug (or unavoidable consequence of the LA-logic ? ) in marlin
- I managed to hack around (*) in Marlin to get `M114 E` reporting again, but not exactly the same values as without LA. Maybe pure E-movements (like retractions) do still work and therefore I got a strange mix ?
- `M114 R` does seem to report reliably with LA on, so probably better to change TFT-logic to poll `M114 R` instead and extract E-values from there.

(*) might have these changes around if anybody is interested ...

digant73, 2021-03-16:
To anyone having the problem with the filament sensor not working when connected to TFT, try to enable M114_REALTIME in marlin fw. Please let me know the result

oldman4U, 2021-03-17:
I have been testing filament runout for seversl hours using the TFT firmware from yesterday and it worked great. Sensor connected to TFT - printing from TFT storage it triggers correctly and purges at the pause position. @digant73 Please mention this ticket in your PR.

Thank you

DaStivi, 2021-03-17:
i am currently testing the newest tft firmware... just need some time, and change the setup ah little bit to test the filament runout accordingly, as i print from octoprint right now, and i guess there it doesn't even work when enabled and not printing from TFT...

btw. yes please @digant73  explain what you changed regarding the m114(_realtime) in tft firmware ? i did looked your Pullrequests but didn't find any mention of such ah fix or any change, would be interessting to see what was causing it if its fixed now... the only PR i see is 1716 where it stated with "other fixes"

digant73, 2021-03-18:
@DaStivi What I can suggest you having the problem with the filament runout sensor on TFT is simply to try the following setting in Marlin (Configuration_adv.h):
M114_REALTIME
M114_LEGACY

one, then the other and finally both.

As far I see the output format for M114 or M114 E is not changed (so no need to make changes on TFT fw). The values could be different and solve the missing filament detection.

oldman4U, 2021-03-19:
@Chris1231
Please make sure to activate all Marlin features which are mentioned in config.ini.

Could you please test with the actual TFT firmware and let us know, thank you.

Chris1231, 2021-03-20:
@oldman4U 
I have activated all marlin features mentioned in the config.ini.
I am currently using the TFT firmware from 3/18.
the filament sensor does not work there.
when my printer is ready, i load today's firmware onto the TFT.

oldman4U, 2021-03-20:
Does not work means in detail?

Have you tested the sensor with a multimeter?

Chris1231, 2021-03-20:
the error is that the message on the TFT comes, no filament.
regardless of whether there is filament in the sensor or not.

if I use the TFT firmware 3.0.25.3, everything works without problems.

yes i checked the sensor (old z-endtop from ender 3).
it works great.

I have already bridged the PA15 directly with 5V in order to rule out a cable break.

oldman4U, 2021-03-20:
And you print from mainboard or TFT?

Chris1231, 2021-03-20:
I print via TFT, and the sensor is connected to the TFT

oldman4U, 2021-03-21:
OK, so this tuned out to be a connection issue. Using only two wires for filament detection, GND and SIGNAL must be connected, voltage is not used in such a case.

Please close this ticket, thank you

closed 2021-03-22 by Chris1231