bigtreetech/BIGTREETECH-TouchScreenFirmware #1613: [BUG] TFT70 Azerty keyboard layout

Grex3D, 2021-02-08:
Hello i  have compiled a firmware for my TFT70, all work fine but  it is impossible to get a AZERTY keyboard layout for the terminal.

When i configure the keyboard layout in AZERTY in the config.h file, the keyboard pass in ABC  the  mode after uploding the firmware.

`* QWERTY/QWERTZ keyboard layout
 * Keyboard layout for Terminal Keyboard (Only for TFT70 V3.0).
 *
 * Options: [0: Default:, 1: qwerty, 2: qwertz, 3: azerty]
 *  default: The keyboard has an alphabetically order.
 *  qwerty: The typically keyboard Layout for english.
 *  qwertz: The typically keyboard Layout for german.
 *  azerty: The typically keyboard Layout for french.
 */
#define TERMINAL_KEYBOARD_LAYOUT 1  // Default: 3`

oldman4U, 2021-02-27:
@bigtreetech
@Msq001

Could you please check the reported issue and fix it. I do not have a TFT70 otherwise I would verify it for you. 

Thank you

oldman4U, 2021-02-28:
After compiling the firmware, from where did you copy the compiled firmware from?

bigtreetech, 2021-03-01:
[image]
This is the result of my test, seems to be normal?

oldman4U, 2021-03-01:
Hi.

Thank you for verifying this. This seems to be a user error and the user
does not answer. Maybe you want to close this ticket.

Kind regards

closed 2021-03-01 by bigtreetech

Grex3D, 2021-03-06:
I m so sorry for my late reply im gonna retry to compil the firmware.

@oldman4U from BIGTREETECH-TouchScreenFirmware-master\.pio\build\BIGTREE_TFT70_V3_0

Thanks.