bigtreetech/BIGTREETECH-TouchScreenFirmware #1614: Main Screen Icons

MTUpower, 2021-02-08:
Hello Everyone, I have today tried to update the TFT35-E3 v3.0, software.  I have down loaded the correct files to the SD card and proceeded with the upload.  All goes well except for the first three icons on the main menu, as you can see from the picture? The down load was from the main BBT touch screen firmware page.  I also tied to do it using the latest firmware release icon and using that however that seems to be missing the config.ini file??

If anyone is able to shed some light on the matter I would be very great full

Kind Regards

Duncan

[image]

oldman4U, 2021-02-08:
Please try this firmware: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/923c5e01d41f08a0ea6945b2291ab5665bbf7b5c

SD Card size and format?

MTUpower, 2021-02-08:
Many thanks for the reply, your suggested firmware is the one I have tried, and also the used the link on the right hand side of the page.  No go with either very odd indeed!  

SD card in fat 32 format and about 5 gig

radek8, 2021-02-08:
Did you use the correct icons according to the diagonal of your display?

MTUpower, 2021-02-08:
Radek8 thanks for the suggestion however I’m not sure I know what you mean.  I used the tft35 font and icons and the tft35 bin file

radek8, 2021-02-08:
Then it's okay.
Use another SD card.
The lower the capacity of the card, the better.

oldman4U, 2021-02-09:
Your problem is for sure not a bug, but most likely a problem with the sd card or another hardware. Make sure you follow the instructions in the ReadMe: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

MTUpower, 2021-02-09:
Afternoon all, and many thanks for all the helpful hints.  To be honest I have no idea what changed as I did the exact same thing, however thing morning the sun is shining and the birds are singing!!  It works again.  My only issue and this was happening before the update is the bed levelling feature on the BBT TFT35-E3 v3.0 screen.  Although the Bed settings in Marlin 300 x 300 and in the ini file have been changed to a CR-10 the BBT screen still thinks its an Ender 3.  But by pushing the button and changing it to the Marlin screen it works fine and goes to the correct corners of the bed.  Any ideas anybody??

kisslorand, 2021-02-09:
Changed where and what in which ini file?
Are you talking about the TFT's config.ini, this section:
`#### Machine Size / Build Area`
`# The TFT will auto-detect the machine size (min and max) in Marlin firmware (requires enabling M115_GEOMETRY_REPORT`
`# in Configuration_adv.h in Marlin firmware).`
`#   Format: [size_min: X<minimum distance> Y<minimum distance> Z<minimum distance>]`
`#           [size_max: X<maximum distance> Y<maximum distance> Z<maximum distance>]`
`#   Unit: [mm]`
`#   Value range: [min: -2000, max: 2000]`
`size_min:X0 Y0 Z0`
`size_max:X250 Y250 Z300`

MTUpower, 2021-02-09:
That is correct I adjusted the setting of thr TFT configuration .ini file to the correct bed size.  And also enabled the G115 in Marlin.
But as I said above works fine when the TFT is in marlin mode but not when in touch screen mode very odd!!!

oldman4U, 2021-02-10:
I would recommend to close this ticket as the reported issue is solved.

In case you can not solve the problem with the bed size start a new ticket with more information and config.ini.

MTUpower, 2021-02-11:
Many thanks for the heads up consider the case closed.

closed 2021-02-11 by MTUpower