bigtreetech/BIGTREETECH-TouchScreenFirmware #1616: [BUG] Long Extrusion refused and M701/M702 - unknown!

FlailAway, 2021-02-10:
### Description
Subject says it all

### Steps to reproduce

Marlin
#define PREVENT_LENGTHY_EXTRUDE
#define EXTRUDE_MAXLENGTH 1200

config.ini
custom_label_1:UN-LOAD Filament
#custom_gcode_1:G1 E-805 F1200\n   <<-- "Prevented too long extrusion"
custom_gcode_1:M702 U805\n            <<-- "Unknown command
custom_label_2:LOAD Filament
#custom_gcode_2:G1 E+805 F800\n   <<-- "Prevented too long extrusion"
custom_gcode_2:M701 L805\n             <<-- "Unknown command

**Expected behavior**

What I asked, feed in/out 850mm
**Actual behavior**

Error messages

### Hardware Variant

BIGTREE_TFT35_V3.0

### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0.26.x.bin

Marlin Stable 2.0.7
Marlin Bugfix 2.0.8

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2021-02-10:
check in the Marlin parameter
#define EXTRUDE_MAXLENGTH

oldman4U, 2021-02-11:
Radek.

Please check the beginning of the ticket.

radek8, 2021-02-11:
Sorry, I missed it.

oldman4U, 2021-02-14:
In case you really have to extrude 850mm I would try to uncomment that function in Marlin to see it this solves the problem.

sarvenn, 2021-02-14:
What if I print something in vase mode, will this long extrusion feature kick in?

oldman4U, 2021-02-14:
No.

This is to un(fill) the bowden which makes such numbers unlikely. Have you
tried to disable like I recommended?

guruathwal, 2021-02-20:
it looks like a marlin issue. check and make sure the firmware is building and uploading successfully.

oldman4U, 2021-02-22:
What is the status of the reported issue?

Please help the community and close the ticket once you do not need it anymore. Thank you

oldman4U, 2021-02-28:
Another 9 day wonder which does not know how to behave right not answering his own ticket.

Answer or close this ticket

closed 2021-04-01 by bigtreetech