bigtreetech/BIGTREETECH-TouchScreenFirmware #1617: BTT TFT35 E3 V3 26.X June 15, 2020 - Filament runout only works in marlin screen mode not TFT mode.

telco55, 2021-02-11:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

telco55, 2021-02-11:
Reproduced several times. 
With the runout sensor installed and enabled in the Touch Screen mode will not allow a print to start. 
The mesage is "filament runout" and the print cancels and shuts down the temps.
Fortunately I am able to print in Marlin Mode just fine. 
I will have to use Marlin Mode till this is fixed. 
I need to tell everyone to not buy a BTT TFT screen till it this issue is resolved.
I have updated firmware on the SKR mini E3 V2 board several times with no effect.
   Please fix this issue so I can use the nice color screen. 
There several other issues with the screen like it hangs the screen in "Loading" after a print pause. This requires a print restart(power cycle). 
When switching back and forth between Marlin screen and TFT screen I found that the Probe Offset was not shared between screen modes. I am not sure if that is normal or a bug. I also had in a Marlin mode session during a print the that the screen became unresponsive  I was unable to get the screen menu to change.  
Thanks    Please Help

oldman4U, 2021-02-11:
Please update to the most recent TFT firmware.

Then read previous made tickets related to your issues. 

For those issues you see on your system but you can not find a ticket for make a new ticket per issue. Include all relevant information incl which filament sensor you use and where it is connected.

This is a friendly reminder, that this repository is about the TFT firmware and not a user group.

Thank you

scathappens, 2021-02-11:
I have the same issue, have had a ticket going for this, it was supposed to be fixed however i have updated the firmware on the TFT35 today to the 04/02/2021 release and it has not solved this issue 

when the filament sensor notices a runout (printing from SD) the head parks as it should, then when i press resume it does nothing....then after a pause of about 30-60 seconds it says heater timout, when i press reheat it the heater turns off.
if i hit stop print at this time i get a message saying 'busy processing, please wait'

so the runout sensor is still not functioning as it should in the latest firmware (dated 04/02/2021)

I get the exact same reaction when i use the pause feature aswel, the head parks but then it will not resume in the same manner described as above

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1538

3dfixer, 2021-02-13:
does the filament sensor work when in marlin mode?
where is it connected too? board or tft

scathappens, 2021-02-13:
I havent tested in marlin mode tbh, it is connected to the TFT...as stated the sensor works but i can't resume the print after reloading the filament, and the pause feature also will not allow resume either

oldman4U, 2021-02-14:
telco55 seems to be a nine day wonder and does not answer his own ticket.

For issues related to his topic please read all the existing tickets snd also the pinned verified bugs ticket.

Thank you

OnBudgetServices, 2021-02-15:
Having the same issues. I've installed the BTT sensor on a custom printer, using the SKR 1.4 board and TFT35 V3 touch screen. The sensor seems to work in Marlin mode, but it always activates, no matter what I do. It does not show anything in graphical mode, no matter what settings I use.

Marlin 2.0.7.2
TFT V3.0.26.x

Any advice would be much appreciated, been battling this for three days now. I'm close to removing it and returning it. I don't really print unmonitored anyways. The only reason I got it was to use up close to empty spools of filament.

npcizzy, 2021-02-17:
@OnBudgetServices I'm having the same exact issue. I've updated my screens (TFT35V2) firmware to the latest. Marlins "classic" mode works great but as soon as I switch over to TS/TFT mode... Nothing.

OnBudgetServices, 2021-02-17:
@npcizzy I was able to get it not always to trigger. I changed  FIL_RUNOUT_STATE from low to high, and that solved that issue. Still no dice on the graphical interface, but it does work if you swap over to the Marlin simulator once it's detected a runout state. Not ideal, but at least there's a way to interact and fix it. I'm assuming this is a firmware issue at this point with the TFT35.

oldman4U, 2021-02-17:
And when you print from touchscreen you made sure to have a M75 in the
start gCode of your slicer?

OnBudgetServices, 2021-02-17:
I didn't. I've got a long print running now, but I have a few smaller ones I can test this with after that.

How does adding a print timer help with the filament dialog for the GUI?

oldman4U, 2021-02-17:
It helps in case you print from on board sd in touchscreen mode. Marlin
does not know about the print otherwise

OnBudgetServices, 2021-02-17:
Ah, got ya.

I'll give it a shot, thanks!

Budis3, 2021-02-19:
In another bug report over this same bug I could solve it thanks a comment from a user. If you put in you start G-code M75 the sensor works but another bug appears.
For more details: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1634

oldman4U, 2021-02-19:
This ticket is about the topic mentioned in the headline.

Check if you have HOST_PROMPT_Support enabled in Marlin.

oldman4U, 2021-02-22:
What is the status of the reported issue?

Budis3, 2021-02-22:
I not consider it complete because if you apply this solution appears another bug. More detaills here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1634
And another reason is, if the another not have a correct solution and here find another better then it should been open.

digant73, 2021-03-13:
Try PR #1718 it could solve the problem

oldman4U, 2021-03-17:
I have been testing filament runout for seversl hours using the TFT firmware from yesterday and it worked great. Sensor connected to TFT - printing from TFT storage it triggers correctly and purges at the pause position. @digant73 Please mention this ticket in your PR.

Thank you

oldman4U, 2021-03-21:
@telco55

Please respond to your own ticket or I expect the reported issues to be solved.

Thank you

closed 2021-04-05 by bigtreetech