bigtreetech/BIGTREETECH-TouchScreenFirmware #1619: TFT24 v1.1 does not save touch calibration results

kaktus312, 2021-02-12:
### Description

After screen reboot I have to calibrate it again becouse  some buttons does not work as expected and couse wrong submenus calling. After calibration procedure touch screen works properly till next reboot.

### Steps to reproduce

1. Reboot TFT24
2. Touch menu button, for instance, or heat nozzle button, or "Ok" button at a popup window with message   

**Expected behavior**
Open main menu screen, or nozzle heat screen, or close a message window respectevly

**Actual behavior**
Open nozzle heat screen, or a window with breef, or a message window does not close at all respectevly

### Hardware Variant
BIGTREETECH TFT24 v1.1
### TFT Firmware Version & Main Board Firmware details
Current TFT Firmware version number is V1.1.26 (latest).
Marlin firmware version number is 2.0.7.2

### Additional Information
No Additional Information

oldman4U, 2021-02-12:
reset the TFT according to the description here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Reinstall the firmware

If this does not solve the problem contact the dealer you bought the hardware from for a exchange or refund

Hope this helps

kaktus312, 2021-02-13:
It seems step 6 of the [tutorial](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#updating-firmware-configuration) was helpful. I repeeted the screen calibration procedure and now the touchscreen works properly, although I think it has some hardware problems.

Thank you!

closed 2021-02-13 by kaktus312