bigtreetech/BIGTREETECH-TouchScreenFirmware #1623: tft70 text overlapping

jariddler, 2021-02-13:
tft70 text overlapping in e-steps calibration menu.
in using skr mini e3 v2 with the tft70.
its running marlin bugfix 2.0.x
screen firmware was updated.
when you go to input length of filament left you cant see the numbers you're changing them to because they're behind the old 
e-steps value

[config files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5975054/config.files.zip)
[image]

oldman4U, 2021-02-14:
Please add E-Steps calibration to the title of your ticket.

Thank you

oldman4U, 2021-02-26:
Thank you 

Please verify that the problem still exists with the firmware available since yesterday and let me know.

jariddler, 2021-02-26:
Yes the new update fixed the issue. Thanks for the fast fix and all the dedication. Keep up the good work

From: oldman4U
Sent: Friday, February 26, 2021 5:46 AM
To: bigtreetech/BIGTREETECH-TouchScreenFirmware
Cc: jariddler; Author
Subject: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] tft70 text overlapping E-Steps calibration (#1623)

Thank you
Please verify that the problem still exists with the firmware available since yesterday and let me know.

oldman4U, 2021-02-27:
Perfect. In this case BTT fixed it.

Please help the community and close the ticket once you do not need it anymore. Thank you

jariddler, 2021-02-27:
The new update fixed the issue. Thanks

closed 2021-02-27 by jariddler