bigtreetech/BIGTREETECH-TouchScreenFirmware #1624: [BUG] Rumba32 and TFT 35 E3

DrumClock, 2021-02-13:
Hi, If the TFT is connected using the EXP 1 and 2 connectors,
it is not possible to activate the DFU mode for FW Marlin recording on the MKS Rumba32 board.
When connecting TFT via UART, everything is fine

oldman4U, 2021-02-13:
Hi. 

Where is the problem in the TFT firmware related to your ticket?

Chris1231, 2021-02-13:
it is possible that you have to turn the connector 180 degrees. you have to compare the pin assignment of the TFT and the Rumba board.

DrumClock, 2021-02-13:
Hi, the connectors are right because TFT in "Marlin mode" works fine.
I think there will be a problem in the FW TFT which blocks the Rumba32 board to switch to DFU mode. 
If I connect the old LCD 12864, everything is fine.

DrumClock, 2021-02-13:
I found that the EXP2 connector blocks the start of DFU mode. 
I'll try to figure out which pin is causing it.

DrumClock, 2021-02-13:
When I disconnect pins 3 and 5 on TFT EXP 2 conector it is possible to start the DFU mod on the Rumba32 board

[image]

pin3 - BTN EN1 - PB2 Rumba board
pin5 - BTN EN2 - PB1 Rumba board

radek8, 2021-02-13:
These are contacts from the Encoder.
It's weird that the encoder does it.

oldman4U, 2021-02-13:
I still don't understand how this is TFT firmware related!?

radek8, 2021-02-13:
> I still don't understand how this is TFT firmware related!?

It only has a problem with the connected TFT.
It does not have a problem when connecting a classic display

oldman4U, 2021-02-14:
DrumClock wrote that Marlin mode works fine. Still don't understand how this could be firmware related but it seems you take care. 👍🏻

DrumClock, 2021-02-14:
@oldman4U is that how you understand it? 

[image]

oldman4U, 2021-02-14:
I do not understand it at all. The TFT firmware uses the existing hardware which provides a certain functionality. The TFT works fine with other mainboards but not with this one. So the firmware works.

For me it is like buying any tires and wondering why the do not fit my car and then to blame the car for it.

DrumClock, 2021-02-14:
I just pointed out that TFT blocks the ability to run DFU mod on a Rumba32 board - so it's not compatible with this board.

If I connect pin3 - BTN EN1 on connector EXP2 (which is defined on pin PB2-BOOT1 Rumba board ),
the mod can be started and FW Marlin can be loaded.

Since the LCD 12864 does not cause it ...... so it must do the TFT 35 E3 display .....

DrumClock, 2021-02-14:
The LCD 12864 display has a 0V pin on restart and therefore the DFU mode on the Rumba32 board can be started.
After loading the FW, the value changes to 3.3V

The TFT35 display has 3.3V on the pin when restarted !

kisslorand, 2021-02-14:
If you check Marlin's "pins_RUMBA32_common.h" file you'll see this:
`  #define BTN_EN1                           PB2`
`  #define BTN_EN2                           PB1`
Since Marlin expects that PB1 and PB to be connected to the encoder, I do not see how it is a BTT TFT FW issue.
The only thing I see different is the that "BTN_EN1" and "BTN_EN2" are connected the wrong way. You can try exchanging them.

DrumClock, 2021-02-14:
HI @kisslorand 
Change does not solve anything. 
The problem is that the PB2 pin also has the BOOT1 function and the TFT 35 has 3.3V on the BTN pins when restarted and does not allow starting the DFU mode on the Rumba32 board ..... I have been writing this for several times ....

kisslorand, 2021-02-14:
I am just checking "LCD_Encoder.c" and `LCD_ENCA_PIN` and `LCD_ENCB_PIN` will be always in logical "1" state that is 3.3V in your case.
There is no FW bug.

DrumClock, 2021-02-14:
The LCD display 12864 is in OV during a reset!
If I switch the pin from PB2 somewhere else, eg PD13,
so you can run DFU mod to upload FW to Rumba32 board.

If I'm not a jerk, then the error in TFT is that when restarting the pin is at 0V ..... or do you have another opinion?

kisslorand, 2021-02-14:
My opinion is that the problem can be solved and it is not a FW issue.
Redefine your encoder pins in Marlin and move those two problematic wires to the newly defined pins.

DrumClock, 2021-02-14:
Yes, but the Bigtreetech TFT display doesn't solve the problem.
It's the same with the TFT 50 and I believe it with the others.

**You should think about this and not that the user will remap the pins.**

kisslorand, 2021-02-14:
> You should think about this and not that the user will remap the pins.

If you're talking to me than I have to let you know that I am also just a user who spent 1 hour of his free time on a Sunday afternoon to help you. I do not work for BTT, I never received any payment from them ever.

> Yes, but the Bigtreetech TFT display doesn't solve the problem.

Also true that the Bigtreetech TFT display didn't create the problem. It's an incompatibility issue and that's it., take it or leave it.

DrumClock, 2021-02-14:
> I have to let you know that I am also just a user who spent 1 hour of his free time on a Sunday afternoon to help you. I

Thanks, I spent two days of my time doing that and basically he figured it all out on his own

kisslorand, 2021-02-14:
Chill dude!
Happy printing, have a nice Sunday, stay safe, happy Valentine's Day, etc.!

closed 2021-02-14 by DrumClock

DrumClock, 2021-02-15:
> Later edit: Try to rotate the encoder while powering up, it might do the trick.

Chill dude, kisslorand !

late: I have already done HW and SW modification of the Rumba32 board, when it could not be solved FW TFT35 E3 ....

[image]