bigtreetech/BIGTREETECH-TouchScreenFirmware #1625: default settings?

3dfixer, 2021-02-13:
so i'm following instruction on readme.md. install new firmware on tft. then told to :-

""On the TFT click on Menu - Settings - Feature and navigate to the last page. Click on
"Reset default settings ..."
power cycle and poof, all my settings/changes to config.ini disappear.

question?  where does it get default settings from? 
thought that was what config was for, and that resetting was to take
you back to config settings after you've been buggering about.

**specify the device variant related to your question.**
skr v1.4 & tft35 v3

Also, totally unrelated, i keep getting this pop up, dunno if bigtreetech are cursing me or not :-)
[image]

digant73, 2021-02-13:
The default settings are the hard coded values while the config.ini is used to overwrite the defalut values. If you omit a param setting in config.ini, its default value is used

oldman4U, 2021-02-14:
Please help the community and close the ticket once you do not need it anymore. Thank you

3dfixer, 2021-02-15:
So dont reset settings if you use config.ini.  thanks digant73

closed 2021-02-15 by 3dfixer