bigtreetech/BIGTREETECH-TouchScreenFirmware #1626: random hangups

3dfixer, 2021-02-13:
**Your question or details you need. Please describe.**

today i updated to latest firmware. i am testing different settings, checking things like pause, fil runout, fil change etc etc
i am printing small object to test things - max 10 min prints
the printer will lockup at random times or not at all. nothing short of reset gets out of it.
also sometimes when print finishes, i get continuous beeping. pull out sdcard from tft- beeping stops, put back 
in starts beeping. can keep doing this till i reset. annoys the hell out the dog :-)

my question is, how do i debug this to find out whats wrong/causing the hangup?

**specify the device variant related to your question.**
skr v1.4   tft35  v3    marlin 2.0.7.2
**Additional context**
Add any other context or screenshots about the question here.
who wants a video of the beeping???

radek8, 2021-02-13:
Given your previous ticket, I'd say you have a problem with a lot of interference.
You should use a shielded serial cable to connect the display to the Marlin board. It is also necessary to shield the display from the heated bed, or place it as far away from the bed as possible. PWM control causes interference and the heated bed acts as a huge electromagnet that emits interference.
That's why you have the message: Unknown command

oldman4U, 2021-02-13:
Please close tickets you do not need anymore, like this one:
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/110

Thank you

According to your profile you are not familiar with coding. Please take my friendly advise and start with a setup which includes only the minimum needed to start printing and add additional hardware once you found a stable setup.

Happy printing

3dfixer, 2021-02-15:
prusa like clone, bltouch, fil runout sensor. skr1.4 & tft 35
i think this is about as basic as anyone would want!
would just like it working well

closed 2021-02-15 by 3dfixer

oldman4U, 2021-02-15:
None of my printers has filament sensor and one has no bl touch. Prints like the other.