bigtreetech/BIGTREETECH-TouchScreenFirmware #1628: BTT TFT70 White calibration screen/touch not working

omgwtf2, 2021-02-14:
[image]

BTT TFT70.
Having issues after GitHub update. (BIGTREE_TFT70_V3.0.26.x.bin)
Updated firmware and when the white calibration screen was up, I was able to touch the red dots in top left, top right and bottom right. It never showed bottom left, just the black dot in the center. Calibration failed.

I tried the reset txt,(notepad) it didn't do anything, just showed card inserted.
Tried updating again with different themes, same issue. No response when touching.
I can't seem to get the calibration white screen back, no matter how many times I "update". ( screen worked before firmware update)
I took the TFT out of it's case, incase it was binding. and repeated steps to no prevail. it doesn't sense anything.
Any suggestions on how to reset it so I can get that calibration screen back? 
reset.txt didn't work. Unless I'm using the wrong (notepad) program.

oldman4U, 2021-02-14:
reset has to work. Make sure it is plain text and not formatted. 

There is no lower left red dot.

omgwtf2, 2021-02-14:
> reset has to work. Make sure it is plain text and not formatted.
> 
> There is no lower left red dot.

In my case it's not, I'm doing something wrong. I just went through an update process and the white screen calibration doesn't show during update.
[image]
[image]

omgwtf2, 2021-02-14:
on the SD card, I created new text document, labeled it reset.txt and left the inside blank.
This image shows what the screen display's after powering up.
[image]
[image]

oldman4U, 2021-02-14:
Where is it written that calibration has to be shown during update?

oldman4U, 2021-02-14:
What is the size and format of the sd card?

Have you tried another sd card?

omgwtf2, 2021-02-14:
> Where is it written that calibration has to be shown during update?

It showed during the first update. then never again. Got confused with the dot thing and it failed.
All rest updates never showed the white calibration screen.

omgwtf2, 2021-02-14:
> […](#)

Tried 2 SD cards, 2gb and 64gb

oldman4U, 2021-02-14:
Seems the 2gb does not works and 64 will not work. And format?

omgwtf2, 2021-02-14:
> […](#)

Fat32

oldman4U, 2021-02-14:
the calibration screen is shown after a successful reset. The screen shown in the movie is corrupted. Both points to a faulty sd card.

omgwtf2, 2021-02-14:
When the calibration screen did show on the first attempt, I used the 64gb SD.
And continued to use it 6 more times. 2gb one was used 3 times

oldman4U, 2021-02-14:
Ok

omgwtf2, 2021-02-14:
[image]
This image is with a different miniSD to Standard adapter, powered off for an hour. SD card is inserted turned power on.

oldman4U, 2021-02-15:
Disconnect everything from the TFT except the RS232 connector and try the reset.txt.

omgwtf2, 2021-02-15:
Done, It doesn't power up.

oldman4U, 2021-02-15:
Check the cable with a multimeter

oldman4U, 2021-02-15:
Which mainboard?

oldman4U, 2021-02-15:
And send a picture which shows the connection.

omgwtf2, 2021-02-15:
SKR 1.4 Turbo, Have power now, RS232 connector to board must of gotten lose after several open and closing.
[image]
[image]
[image]

oldman4U, 2021-02-15:
Ok. Reset with only the black cable connected. Nothing else.

omgwtf2, 2021-02-15:
The black cable is the only connection and the sd reset.txt is attached. still the same. last 2 pictures

oldman4U, 2021-02-15:
Ok. No idea left.

Sorry

kisslorand, 2021-02-15:
@omgwtf2 
You connected to the wrong connector the RS232 cable.

omgwtf2, 2021-02-15:
> @omgwtf2
> You connected to the wrong connector the RS232 cable on the mainboard. That is an I2C serial connector, not an RS232.

[image]

kisslorand, 2021-02-15:
Sorry, my mistake, previous picture tricked me.

omgwtf2, 2021-02-15:
Thanks for the assist, oldman4U

oldman4U, 2021-02-15:
The last trick is to take a UDB cable, cut it and to connect the red and
black wire to 5v and GND. This provides the power needed and eliminates
everything else. I still believe the problem is finally the SD card or the
reset file. More like the card.

omgwtf2, 2021-02-15:
Tried 3 cards last night, broke the 8gb after 2 try's.
[image]

omgwtf2, 2021-02-15:
[image]
[image]

omgwtf2, 2021-02-15:
Same outcome.

oldman4U, 2021-02-15:
This will not work in case there is a problem with the power supple or the
mainboard or a cable in between.

omgwtf2, 2021-02-15:
> […](#)

[image]

kisslorand, 2021-02-15:
Where is Rx and Tx connected on the TFT?

omgwtf2, 2021-02-15:
Wanted to confirm it's not the cable.
[image]
[image]

kisslorand, 2021-02-15:
Touchscreen is working now OK? Baudrate is set to match the mainboard?

omgwtf2, 2021-02-15:
> Touchscreen is working now OK? Baudrate is set to match the mainboard?

Still not working.

kisslorand, 2021-02-15:
That's the first step, to make the TFT work.

omgwtf2, 2021-02-15:
oldman4U will have to go walmart to pickup same more sd cards, to rule it out.
USB won't work?

omgwtf2, 2021-02-15:
> oldman4U will have to go walmart to pickup same more sd cards, to rule it out.
> USB won't work?

Nevermind USB didn't work.

omgwtf2, 2021-02-15:
Press and hold knob it shows Marlin and bigtreetech, use knob to select marlin, knob works, shuffle through marlin menu and select things, no problems.
Press and hold knob and select bigtreetech opens screen (like the pictures) knob doesn't do anything anymore.

oldman4U, 2021-02-15:
Connecting the USB like I described has to power up the TFT.

omgwtf2, 2021-02-15:
> […](#)

[image]
Didn't work.
[image]

oldman4U, 2021-02-15:
The RESET and the FIRMWARE UPDATE ONLY works from TFT SD Card, not from mainboard and not from USB stick.

oldman4U, 2021-02-15:
[image]

[image]

omgwtf2, 2021-02-15:
> The RESET and the FIRMWARE UPDATE ONLY works from TFT SD Card, not from mainboard and not from USB stick.

I used the mainboard marlin install only, all tft firmware and what we have been doing has been from tft.
I have to verify the SD card suspicion, will re visit after purchasing a couple at Walmart.
[image]

omgwtf2, 2021-02-15:
[image]
Ok, Start off with reset.txt?

omgwtf2, 2021-02-15:
16gb SDHC  and 32gb sandisk SDHC didn't work

oldman4U, 2021-02-15:
Hope u also got some smaller ones.

omgwtf2, 2021-02-15:
> […](#)

didn't have any smaller

oldman4U, 2021-02-15:
What is with the one you got with the mainboard?

omgwtf2, 2021-02-15:
IDK if I got one?

oldman4U, 2021-02-15:
The SKR comes with a micro sd card

omgwtf2, 2021-02-15:
MIA, Sorry.

omgwtf2, 2021-02-15:
No go with the other 2 remaining cards

Grex3D, 2021-02-16:
Just to know, pc or Mac to create your reset.txt ? (ask you because .DS_Store on mac can cause issues)

My sd car is a Samsung 32go all work fine.

Same issue at the first start, reset.txt solve the issue instantly.

oldman4U, 2021-02-16:
Thank you Grex for helping with this. In my house only macs exist 😁

omgwtf2, 2021-02-16:
Windows 10 pc, just ordered 8gb miniSD from amazon, nothing local.
Big appreciation on the help!

Grex3D, 2021-02-16:
> Thank you Grex for helping with this. In my house only macs exist 😁

i think that your issue.

Try with your Neighbours pc :D (or friend) for sur .DS_Store is the issue.

Format from pc and create the reset.txt, (dont re-put the sd in your mac) and try :)

oldman4U, 2021-02-16:
But I do not have a problem. Only one out of 10  sd cards does not work for
firmware update.

Grex3D, 2021-02-16:
sorry i made a  mistake  your not the autor ! :D

if it work for you, that not the issue ...

[4 more replies]

closed 2021-02-16 by omgwtf2