bigtreetech/BIGTREETECH-TouchScreenFirmware #1629: [BUG] TFT35 keeps freezing

CRCinAU, 2021-02-16:
### Description
When randomly pressing buttons within the touchscreen, after a few seconds, the entire screen will freeze.

When the freeze occurs, the button will be 'highlighted' as if its some kind of touch effect. The only recovery is to reset the screen.

This screen has been functioning properly for months - so this is rather new.

### Steps to reproduce

1. Update to latest firmare
2. Randomly go through menus
3. Observe the freeze

### Hardware Variant
TFT35

### TFT Firmware Version & Main Board Firmware details
TFT App: 5143b8162a927b303f865a149ca0363a47c2e347
Marlin: bugfix-2.0.x

oldman4U, 2021-02-16:
This is most likely not a firmware issue but hardware related. And this is the firmware repository. 

Please help the community and close the ticket once you do not need it anymore. Thank you

CRCinAU, 2021-02-16:
This only started on the latest few revisions - so I'm wondering if backing up through the prebuilt binaries would resolve it.

oldman4U, 2021-02-16:
If it is really software related it has to solve it. You know how to get
older versions?

CRCinAU, 2021-02-16:
Yeah, I'm guessing a `git checkout <commit id>` and then just look through the log for when the binaries were updated.

oldman4U, 2021-02-16:
Perfect.

StefanBiermann, 2021-02-16:
Same problem here - i'd really be happy if one may tell me which commit id is latest 'non freezing' :-(

oldman4U, 2021-02-16:
Steven and Stephan have the same issue😁 Sorry

Maybe you can upload a short movie to check if you have the same issue.

I use firmware from Feb. 3rd without any issue reported here.

StefanBiermann, 2021-02-16:
Updated today from 0.23 version to current 0.26.x version. Since the update screen freezes randomly when pressing a 'button'. The 'button' looks like highlighted then - exactly as described by Steven.

oldman4U, 2021-02-16:
And you both followed the update described in the "ReadMe" on the main page of this repository and you read the "Read me before posting a new Issue" and there especially this part:

Q: Touchscreen doesn’t respond/ stay’s in the “Choose mode” selection

Answer: There are 2 possible problems that can be causing this.
a) The screen stay's in choose mode
The screen is touching the frame where it’s mounted in.
Create more clearance in between the touchscreen and the frame where you mounted it.

b) The screen doesn't respond to any touch
The touchscreen settings could be corrupted. The screen doesn't know at this moment how to correctly respond to a touch.
A way to solve this:
Place a text file reset.txt in the root of a SD-card and insert this in the screen
Reset the screen.
When the screen is booted there will be a touchscreen callibration screen. Calibrate the screen and it should be solved.

Which date is shown under the Info of your touchscreen?

StefanBiermann, 2021-02-16:
My screen isn't mounted yet to a chasis at the moment (and about 15cm away from the original chasis), as i just replaced the board by an SKR 1.4 Turbo. I already tried reseting (and therefore calibration) a couple of times, i changed between the different icons (and thus flashed the firmware more than once). Sometimes it works for 15-20 minutes, sometimes it freezes after the second or third icon touch :-( i'm happy to got my bed leveled (hopefully) after hours ...

CRCinAU, 2021-02-17:
Tested the following commit IDs where the binaries were updated:

Commit: e1002b749b959ba8fff6c65d4c7287c016f6816a - Tue Feb 2 2021 - Hangs after a couple of button presses.
Commit: e20291f2107ddd4f1236023346b884e40c19b48e - Sat Jan 30 2021 - does not hang.
Commit: f9c2f9d786abe046a4d34bced91dbc7d12d9e894 - Sat Jan 23 2021 - does not hang.

StefanBiermann, 2021-02-17:
@CRCinAU Did you use the prebuilt bins?

CRCinAU, 2021-02-17:
> @CRCinAU Did you use the prebuilt bins?

Yes.

radek8, 2021-02-17:
I think there will be some problem in using RAM.
I created a PR with the option to turn on StealthChop mode
Despite the fact that there was nothing conflicting in PR, my display froze when I started printing. So the problem really won't be in the HW but in the current firmware.

StefanBiermann, 2021-02-17:
Rolling back to e2091f, like Steve did, also seems to solve the freeze problem for me (still issues with z leveling, but this also happend before i changed the board).

DaStivi, 2021-02-17:
i had the freezing tft also a couple of times with the latest nightly builds from 04.02.2021... i did had suspected the UPS, what i am also trying to get to work... but also did disabled everything from this and still had the tft hangs... then i flashed back to nightly from 2.2.2021 and hadn't any freeze in two long prints now that last over 24h ...

oldman4U, 2021-02-17:
So do you need more information Radek8?

Are you all using BTT TFTs or B1 or MKS?

DaStivi, 2021-02-17:
in my case, the tft (and the print was gooing) freezed on ah B1 (tft35 varriant) hadn't had the issue in menus... i got it a couple of times in real print jobs...

radek8, 2021-02-17:
I'm not smart enough to solve this problem.
I have a TFT 35 E3 V3.0
With the current configuration, my display does not freeze. But just add the setting of one parameter to the FW and it will start freezing when you start printing from the TFT SD card.

oldman4U, 2021-02-17:
I do not see this problem using the same kind of TFT. What can I do to reproduce the issue?

StefanBiermann, 2021-02-17:
To be a bit more specific:
BTT TFT35 V2.0 on Ender 5 Plus with SKR V1.4 Turbo Board.
Updated via SD card with current BIQU_TFT35_APP1_V2.0.26.x.bin (incl. config & different UI themes).
Randomly freezes when touching random icons, sometimes within seconds, sometimes after 10-15 minutes. E.g. while going to level the printer or home it ... The last touched icon stays 'highlighted', no response until reset. When i was able to start a print it stops after random time, display is frozen. Using exactly the same config & UI theme with the older checkout makes the display operate normaly (since then two successful prints > 4h). 
Switched to the V3.0 display, exactly the same behavior - random freezes with current precompiled firmware. Switching back to the e20291f commit it works perfectly ...

oldman4U, 2021-02-18:
So there is a TFT35 v2 and a B1 TFT35 with the same issue. I have been testing my BTT TFT E3 v3 the whole evening while testing the PR from Gurmeet (https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware.git) without any problem, right now preparing the second 2h+ print.

Will let you know if it is successful or fails.

oldman4U, 2021-02-19:
2 prints each 2h+ without a problem. Also navigation in menus is fast and does not cause any issues.

DaStivi, 2021-02-19:
> 2 prints each 2h+ without a problem. Also navigation in menus is fast and does not cause any issues.

you wrote you're on the guruathwal branch? as at least i don't see the latest commits in there... isn't it?

oldman4U, 2021-02-19:
Yes. You have to go to his PR and click onto his repository.
[image]

CRCinAU, 2021-02-20:
Tried that - it hung on the 4th button press.

andy-miles, 2021-02-20:
> 
> 
> Tested the following commit IDs where the binaries were updated:
> 
> Commit: [e1002b7](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/e1002b749b959ba8fff6c65d4c7287c016f6816a) - Tue Feb 2 2021 - Hangs after a couple of button presses.
> Commit: [e20291f](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/e20291f2107ddd4f1236023346b884e40c19b48e) - Sat Jan 30 2021 - does not hang.
> Commit: [f9c2f9d](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/f9c2f9d786abe046a4d34bced91dbc7d12d9e894) - Sat Jan 23 2021 - does not hang.
> 
> With the build on [e1002b7](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/e1002b749b959ba8fff6c65d4c7287c016f6816a), I did the thing with `reset.txt`, and then recalibrated the touchscreen. It still hung after less than a minute of randomly going into menus and submenus.
> 
> Rolling back to [e20291f](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/e20291f2107ddd4f1236023346b884e40c19b48e) and things work fine again...

I can confirm that I can reproduce the same behavior on a TFT35 v2.0 based on the cited commits.  I took a closer look at [0606d0a](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/0606d0a831941e55d5fb7248e232cbca831a4c4c#diff-b5eb94c87e81ca9ef828444fb153588227f045eab8e507cca99bfd38702bb4c0L762-R772) with menu.c.  It appears that there was a check on the state of `tempKey != KEY_IDLE` that was removed.  I lack the full context of the code base, so I'm testing a weak hypothesis based on adding the check back in with the following code snippet in `void loopCheckBack(void)`:
```
if (tempKey != KEY_IDLE && curMenuItems->items[tempKey].label.index == LABEL_BACK)  // check if Back button is held
{
  BUZZER_PLAY(sound_ok);
  backHeld = true;
  infoMenu.menu[1] = infoMenu.menu[infoMenu.cur];
  infoMenu.cur = 1;
  if (infoMenu.menu[1] == menuPrinting)
    clearInfoFile();
}
```

I've since applied this update based on [923c5e0](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/923c5e01d41f08a0ea6945b2291ab5665bbf7b5c) and appears to have fixed the hang for me.  To be sure, I reverted my fix and can consistently reproduce the issue.  I don't have the context on the implications of adding back this check with its impact to the behavior for other device types.  If anyone else can confirm my hypothesis, please let me know and I can submit a pull request.

oldman4U, 2021-02-20:
Would be good to add the developer of the original change.

andy-miles, 2021-02-20:
> Would be good to add the developer of the original change.

@kisslorand, you mind taking a look at my previous comment?

CRCinAU, 2021-02-20:
> For reference, here is:
> 
>     1. [The commit with the potential fix in my forked repo](https://github.com/andy-miles/BIGTREETECH-TouchScreenFirmware/commit/99cdc12804b41cffb402ade51793cef99d13455c)
> 
>     2. [Prebuilt images with the above commit available for testing](https://github.com/andy-miles/BIGTREETECH-TouchScreenFirmware/tree/TFT35v2-Hang-Fix-Images/Copy%20to%20SD%20Card%20root%20directory%20to%20update)

For reference - I started setting up my PIO buildchain until I read you already had the binaries. :facepalm: 

Good news, I have not been able to make the screen freeze after this patch...

kisslorand, 2021-02-20:
> @kisslorand, you mind taking a look at my previous comment?

Yes, I will do that.

andy-miles, 2021-02-21:
> Thanks for everyone for finding the bug and the cause of it and I apologize for it, it is my mistake. Sorry.

@kisslorand
No worries and no apology necessary. Bugs happen. Glad I was able to help.

DaStivi, 2021-02-21:
hey guys, did a couple of prints now... also dialed in ah new woold filament and got another hang now... i did initially right after updating another hang but didn't ensured too much about this.. but now it happend in mid print again for me.... and hadn't anyissue before... i do try again the version from 3.2.2021 6976fa8ebad634c60c896bda37e66e835f0946fe 

and i think i had the e1002b749b959ba8fff6c65d4c7287c016f6816a version before now where i hadn't an single issue... strange thing is that @andy-miles noted he had ah hang with the version though ?!

DaStivi, 2021-02-22:
and just got another freeze with previous mentioned commit... go this this version now fb8628700df219faf0d75d5b4efc15315eb705ea

andy-miles, 2021-02-22:
@DaStivi
Your freeze does not appear to not be related to the same root cause that was previously identified.  Does it freeze randomly or only when the continue printing dialog is shown?  Which version of Marlin are you using? I recommend to **not** change Marlin versions as it would introduce a lot of unknowns.  It's difficult to find the root cause of your issue without identifying repro steps as well as establishing the last known working TFT35 build for your use-case.

DaStivi, 2021-02-22:
> @DaStivi
> Your freeze does not appear to not be related to the same root cause that was previously identified. Does it freeze randomly or only when the continue printing dialog is shown? Which [...]

yeah sadly its randomly.... just had also freezes on the latest nightlys, then tried your binarys from your fork....
did flashed back to one of the pre-builts from 2.2.2021... but there where a couple of pre-builts that day... so i'm not sure on what version i was before i tried yours...

PR #1607 (2021-02-24): Filament info for multi-filament print, possible fix for Load/Unload hanging popup, fix TFT freeze
### Requirements

BTT TFT or MKS TFT

### Description

This PR adds support for multi-filament prints. It enhances the print summary so it shows the sum of each  filament data present in the gCode. (sum of length, sum of weight, sum of cost)

Later on added a possible fix for the Load/Unload popup hanging issue.
...and more later added a fix for TFT freeze.

### Benefits

It benefits those who use the filament data embedded into the gCode and use multi-filament print.

### Related Issues

Impleme [... cut]
closed 2021-02-24 by bigtreetech