bigtreetech/BIGTREETECH-TouchScreenFirmware #1630: [FR] Scroll long file names in TFT mode while printing

sarvenn, 2021-02-17:
**Is your feature request related to a problem? Please describe.**
While printing a thing long file names scrolls in Marlin mode (relevant feature is enabled in my marlin config), but in TFT mode file name does not scroll.
My request is to have long file name scroll feature in print screen.
[image]

kisslorand, 2021-02-18:
IIRC it was like that but it could hurt the print quality so it was removed.
During printing it's best to leave the TFT alone and let it do the most important job, to send printing gCodes.

sarvenn, 2021-02-18:
Oh ok, so that request will have a load on the TFT processor and may cause issues during printing, right?
If so, it is better to leave like that :) It was just a "nice to have" idea.

kisslorand, 2021-02-18:
Yes, scrolling text adds load to the TFT.

closed 2021-02-18 by sarvenn