bigtreetech/BIGTREETECH-TouchScreenFirmware #1631: [Q] (How can i use turkish charset)

EmA04, 2021-02-17:
**Hello,
ı am using SKR V1.3 motherboard and TFT35 v2.0 touchscreen  i need use turkish charset How can i do it?**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**TFT35 V2.0**

**Additional context**
Add any other context or screenshots about the question here.

EmA04, 2021-02-17:
> **Hello, ı am using SKR V1.3 motherboard and TFT35 v2.0 touchscreen i need use turkish charset (example: ı İ ç Ç ğ) How can i do it?**
> A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.
> 
> **TFT35 V2.0**
> 
> **Additional context**
> Add any other context or screenshots about the question here.

radek8, 2021-02-17:
What do you need it for?
Terminal mode is used to send commands not to write novels :-)

radek8, 2021-02-17:
My language also uses special accents and characters, but I don't miss the display.

EmA04, 2021-02-17:
I want to use this characters for menu.

radek8, 2021-02-17:
They are used in the Turkish menu

[image]

EmA04, 2021-02-17:
those characters are not visible

radek8, 2021-02-17:
[image]
[image]
[image]
[image]
[image]
[image]

EmA04, 2021-02-17:
I could not. I choose the language as Turkish but different characters come out

radek8, 2021-02-17:
You have an old FW

EmA04, 2021-02-17:
I am using (THEME_Unified Menu Material theme) theme.

radek8, 2021-02-17:
Did you update the font as well?

EmA04, 2021-02-17:
No, i can't updated FW

EmA04, 2021-02-17:
I'm getting an error while updating

EmA04, 2021-02-17:
[image]
[image]

radek8, 2021-02-17:
You don't have to compile your own FW.
Use precompiled and configure ini file.
Follow the instructions on the title page

sarvenn, 2021-02-17:
It is explained in the README section of the main page. Check the firmware update section in that text.

EmA04, 2021-02-17:
The screen is not with me right now. I'll try again tomorrow and return

EmA04, 2021-02-17:
peki menüden led menüsünü nasıl kaldırabilirim

EmA04, 2021-02-17:
So how can I remove the led menu from the menu

radek8, 2021-02-17:
That has already been solved here ...
If you don't know what you're doing, don't do it

radek8, 2021-02-17:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1053

EmA04, 2021-02-17:
Sorry but my english is not very good. I looked a little and couldn't find a Turkish source and had to ask here. Sorry to keep busy

oldman4U, 2021-02-17:
In the ReadMe it says clearly self compiled OR a pre-compiled firmware, the config.ini and a TFT folder as well as OPTIONALLY a language pack.
[image]

Please help the community and close the ticket once you do not need it anymore.

Thank you and happy printing

closed 2021-02-18 by EmA04