bigtreetech/BIGTREETECH-TouchScreenFirmware #1634: [BUG] The runout sensor doesn't work correctly

Budis3, 2021-02-19:
I have bough the btt skr mini E3 v2.0 whit the tft35-e3 v3.0 few weeks ago. When I start to configure them I have an issue with a basic runout sensor. This sensor is connected to the mainboard and when the sensor is triggered works well until you want to restart the print. When you want it, the screen in graphycal mode show a message that sais "The printing was stopped by user" with an "ok" button. I spect that when I press the button it will restart, but the message continuously are showing and the print doesn't restrart. In marlin mode, this isssue doesn't happend but if I bougth the tft is for use them whit the graphycal mode.

So, the question is, how I can restart the printing after this message? how I can fix that bug?

For more details I use the version of Marlin 2.0.7.2 that appears in the BTT github and the firmware of the TFT is the version 0.26.x of Feb 4 2021 of that model.

NOTE: If you come to the bug that happens with this sensor that even if it was triggered doesn't work as I sais (like happen in this bug report: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1612) you must know that the solution is put as G-code start M75 in your slicer. This is another bug report for another bug.

digant73, 2021-02-19:
Verify that all the features reported at the beginning of config.ini file are enabled in Marlin

kisslorand, 2021-02-19:
I cannot reproduce the issue, as **digant73** suggested, probably `HOST_ACTION_COMMANDS` and `HOST_PROMPT_SUPPORT` are not enabled in Marlin (Configuration_adv.h).

Budis3, 2021-02-19:
> Verify that all the features reported at the beginning of config.ini file are enabled in Marlin

But @digant73, the config.ini is a file of the firmware of the TFT, not of Marlin. And if is in that case that the file was on this firmware. What I have to do? Uncomment deleting the lines in the "Supported Marlin Firmware Versions"? Or only make as @kisslorand says in the configuration_adv.h?

kisslorand, 2021-02-20:
"config.ini" is a file for the TFT. In that file it is written what are the recommended settings in Marlin.
Check in the TFT's "config.ini" what are the recommended settings and enable them in Marlin's "configuration_adv.h".

Budis3, 2021-02-20:
I configured all as you said and, it is done but something wrong happen. When I was printing and trigger the sensor all work as I spected, but when the printing is ending the buzzer not stop to make sound and don't let me do anything on the screen, not let me continue whit another printing, I have to restart the print power. And that's strange because if the sensor not triggered in the printing process this issue don't happend... Maybe it would solve it with a G-code at the end of the printing no?

kisslorand, 2021-02-20:
I accidentally also triggered the same beeping stuff the other day. I will look it up.
I do not think (but possible) that any gCode at the end will eliminate that continuous short beep series.

Budis3, 2021-02-20:
@kisslorand 
Maybe (if the solution is with a G-code) we must search someone that finish the job, stop it or restart the print as the state before the start printing. I say that as a suggestion.

kisslorand, 2021-02-20:
Nah, it's something fishy that happens during print pause/filament runout. It's not new, I came across these before.

Budis3, 2021-02-20:
@kisslorand 
If we couldn't use a G-code with that mission, then the issue probaly is on Marlin.

Budis3, 2021-02-20:
@kisslorand 
When you press the stop button what gcode send to the mainboard?

3dfixer, 2021-02-21:
I too get the continuous beeping at end of print

Budis3, 2021-02-21:
@kisslorand 
If you put M410 at the end of the printing, the 3d print restart. But I not consider that as a final solution. Is there a gcode that restart without reboot?

digant73, 2021-03-13:
@budis3 @3dfixer try PR #1718 it should fix the problem

closed 2021-03-15 by bigtreetech

digant73, 2021-03-18:
> I configured all as you said and, it is done but something wrong happen. When I was printing and trigger the sensor all work as I spected, but when the printing is ending the buzzer not stop to make [...]

Solved by PR #1718. Try it and let me know the result please

digant73, 2021-03-18:
> I too get the continuous beeping at end of print

Solved by PR #1718. Try it and let me know the result please

PR #1718 (2021-03-15): Configurable emulated M600 / real M600 + bugfixes + cleanup
* Added parameter "emulate_m600" on config.ini to select between emulated M600 / real M600 handling.
With emulated M600, the TFT intercepts the M600 gcode (filament change) and emulates the logic instead of demanding it to Marlin firmware.
Enable it, in case Marlin firmware does not properly support M600 on the mainboard
* New param emulate_m600 is also selectable at runtime from Features menu
* fixed missing "break;" for new configurable RRF Macro param
* fixed XYZ min limit to support also Delta printers
* fixed a bug on Marlin prom [... cut]