bigtreetech/BIGTREETECH-TouchScreenFirmware #1635: [BUG] Manual Bed Mesh Edit doesn't save

TobseF, 2021-02-20:
### Description
I try to edit the bed mesh. But I cannot save my edits.

### Steps to reproduce

1. I leveled the bed by hand with the 4 screws and a paper. 
2. To compensate my extremely uneven, wrapped bed, I started the manual bed leveling (I don't have a BLTouch).
3. I did some test prints and now I want to finetune the mesh. But every time I change any value, in the _Mesh Bed Leveling_ menu and saves it to the EPROM, it gets lost in the 8Bit digital nirvana. 

You can follow the process in the video:
### 🎞 https://youtu.be/_9Il-2r0p7E

**Expected behavior**
After hitting the save icon, the value should be present in the mesh.

**Actual behavior**
The value resets every time to the previous value.

### Hardware Variant
Artillery Sidewinder X1 4 default TFT. STM32F 107
### TFT Firmware Version & Main Board Firmware details
**TFT**: `0.26.1_patch_9` **Board**: `2.0.7.2_mbl`

### Additional Information
### 🎞 https://youtu.be/_9Il-2r0p7E

digant73, 2021-02-20:
the problem could be in the precision used on Marlin fw. When saved, Marlin fw could approximate 0,550 to 0.600.
Also it seems that before using the Mesh Editor it is better to disable BL (at least using Marlin 2.0.7.2). It was reported that otherwise wrong z heights are saved. Possibly your problem is due to this second problem

TobseF, 2021-02-20:
I retried it with disabled BL. It really seems a better option, when editing the mesh. But It's still the same problem, that the value get's written to the point below. This works and no rounding was applied.

digant73, 2021-02-20:
well it seems that the J index is wrongly reported. You selected I2 J1 but it is displayed J2 in the Mesh Tuner. Probably a bug when using a not square grid (your is a 5x4).
I would fix that problem asap but unfortunately in this monent my TFT is dead and I'm waiting for a new one.

digant73, 2021-02-21:
I see on Marling github that it could be a bug on Marlin 2.0.7.2. If you can, try on Marlin 2.0.6.1. It should work properly on that fw.
There are a lot of bugs introduced in 2.0.7.2. Also the need to disable BL seems a bug on Marlin

TobseF, 2021-02-21:
I checked the mesh output. The mesh size seems to be out of sync:
**Display**: `5 x 4` VS **Bed**: `5 x 5`

So maybe I should report it also to the [Molise 1.2.5 Firmware ](https://github.com/Dtcreation/Firmware-Molise-Artillery) team. Because grid settings from TFT and firmware should match.

Is there an option to change adjust the mesh size in the display?  
The only mesh setting if found is `mesh_min_color`.

### Mesh Bed Data
```
Send: M420 V1 T1
Recv: Mesh Bed Level data:
Recv: 5x5 mesh. Z offset: 0.00000
Recv: Measured points:
Recv:         0        1        2        3        4
Recv:  0 -0.65000 -0.57000 -0.56000 -0.52000 -0.46000
Recv:  1 -0.65000 -0.63000 -0.68000 -0.54000 -0.45000
Recv:  2 -0.70000 -0.69000 -0.67000 -0.58000 -0.54000
Recv:  3 -0.70000 -0.70000 -0.67000 -0.62000 -0.52000
Recv:  4 -0.53000 -0.58000 -0.61000 -0.58000 -0.54000
```

### TFT Values
[image]
[image]
[image]
[image]

digant73, 2021-02-21:
ok. the row 0 is not displayed (that should be displayed as the last row in the editor). That's the cause of the wrong settings when you edit a point. I will see in the code.

thineight, 2021-02-27:
Hi, I experienced the same issue.
When I set a 4x4 mesh for MBL on my Sidewinder X1 the mesh edit function shows me a 3x4, now I uploaded the firmware with 5x5 configured.. and the display shows me 4x5. Always 1 row less.

From these pages looks like there are some bugs in the Marlin 2.0.7.2 and perhaps it is convenient to use the 2.0.6.1 in the meantime?
Thanks

digant73, 2021-02-27:
@thineight I know you experienced the same issue. As I already told you, I will fix it when I will receive my new TFT. Unfortunately without a TFT I cannot do anything. In the meanwhile you can stay on Marlin 2.0.5.1 if it's working on that

thineight, 2021-02-27:
Sorry I didn't see the answer on thingiverse.
I am coming from the stock firmware (got the printer few weeks ago) so I am not experienced in this field.
I see in your repository there is also a v.2.0.6.1 MBL fix.. can this be used instead.. waiting for the 2.0.7.2 bug fixed? Or do you still recommend the 2.0.5?

digant73, 2021-02-27:
@thineight no problem. I was joking. Yes, I suggest you to use the 2.0.6.1 waiting for a new Marlin fw fixing a lot of bugs

thineight, 2021-02-27:
Hi @digant73 , just for info to help debugging..I just loaded the following fw:
- Marlin 2.0.6.1 from your thingiverse repo.
- for the TFT:  the bin files from BTT repo (24 Feb) + your config.ini from your repo.

Unfortunately I still see 4x5 on mesh edit despite I have configured a grid 5x5.
I am afraid that the issue is not on the 2.0.7.2 then...

digant73, 2021-02-28:
I developed and tested the mesh editor on marlin 2.0.6.1. So if not working now, something was changed in the TFT fw. I will see it when possible.

thineight, 2021-02-28:
I see.. thanks!
possibly I should try with a older TFT fw. I will try with all the files from your repo (2.0.7.2 hex and TFT bin).
In the previous trials I got the latest bin files from BTT GitHub.

digant73, 2021-02-28:
I think the problem was introduced with some of the last changes in the TFT fw. I have some suspects. You should try my version related to marlin 2.0.6.1. If you don't have it I can upload it again. However the bug should be easily identified and fixed once I will have a working TFT

thineight, 2021-02-28:
I remember in the repo there is 2 set of marlin fw (2.0.7.2 and 2.0.6.1) and 1 for TFT, not sure if it is the one ok for the 2.0.6.1.
If not.. it should be great if you upload it.
Thanks

digant73, 2021-03-01:
@thineight I uploaded the previous TFT fw version (8.8) that should be working. I would also test it with Marlin 2.0.7.2.

thineight, 2021-03-01:
Just a feedback: with the Marlin 2.0.6.1 and the TFT firmware 8.8 the MBL mesh edit shows the correct size (5x5 in my case).
Will try some print test tomorrow.
Hope you can sort this out with the latest marlin fw when you get the mks28 display.
Thanks

digant73, 2021-03-01:
Try marlin 2.0.7.2. I suspect that the TFT 8.8 fw version will work. In that case I know the cause of the bug although the editor was not changed at all

thineight, 2021-03-03:
What is the correct procedure for the MBL in order to obtain a precise result?
After the reset of the eeprom, I follow these steps:
1. Level the 4 corners with the skrews + center point. Check done with the paper sheet.
2. Enter MBL menu and start leveling the 5x5 points with the up/down soft buttons. Data saved in eeprom.
3. Switch on the BL button

At this point I expect that if I test the level , e.g. the 4+1 points or the mesh edit, I should feel the "perfect" height for each point, which is not.
Am I missing something?

In addition, when I start my print I see the nozzle high up and I have to tune it down with the babystep.
Any suggestion?
Many thanks

digant73, 2021-03-03:
You also need to preheat nozzle and bed during leveling. Use 200 and 60 for pla. If during a print the first layer is not sticking to bed use babystep to find the good z offset. Save to eeprom when you find a good adhesion

thineight, 2021-03-03:
Yes, forgot to mention that I also preheat everything.
Is there any setting to start print with the settings as per my calibration? I see there is a +0.1 offset displayed every time a new printer starts.. is that normal?

digant73, 2021-03-03:
You simply need to enable BL to apply the meshes during a print. If you see 0.1 it means you set 0.1. See on "probe offset" entry on "parameters" menu. You should see z offset set to 0.1. If you have Marlin 2.0.7.2 it is possible it is a bug. There are open bugs on wrong z height with bl.

thineight, 2021-03-04:
No, I am still on 2.0.6.1 from your repo as I understand it is to date the most stable.

I found the setting "z offset" on the Mesh Bed Leveling" menu.. hope it is the right one...
Not sure all the settings are saved correctly as, if I do a second print, I always need to lower down the nozzle with the babystep..

digant73, 2021-03-05:
If you need again to lower down the nozzle with babystep you probably didn't save it before. The Z offset displayed in babystep menu is saved on EEPROM (while babystep is reset after each G28).
I set safe homing in Marlin fw even for MBL. For MBL it could be set on bottom-left corner

thineight, 2021-03-05:
Ok I will try to change the Z_HOMING_X(Y)_POINT variables and recompile.

I will also change the mesh grid for MBL from 5x5 to 4x4.
On this respect I see in the code of that section that there is a variable called MESH_EDIT_Z_STEP set to 0.025.. where in the TFT menu the steps are shown as 0.01-0.1-1 (0.01 looks too small to me, in fact looks like my z motors moves a little only every 2 clicks).
Is it correct? Should in the TFT be shown as 0.025 as well?
Thanks

thineight, 2021-03-05:
Update: 
I changed the  Z_HOMING_X(Y)_POINT variables from X(Y)_CENTER to a fixed value like 10, but the result is that the printer was not able to lower down the extruder to reach the z limit (the bed), only perhaps those 10mm i set.
Therefore I switched back to the original code, I changed only the MBL grid to 4x4 (that worked).

I am afraid I have to go back to the school and learn some more if I want to change the homing point 😄

digant73, 2021-03-05:
probably due to inset limits. Try with 20-30.

thineight, 2021-03-05:
Thanks for the suggestion. I tried as suggested with the following code.
`#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
//  #define Z_SAFE_HOMING_X_POINT X_CENTER  // X point for Z homing
  #define Z_SAFE_HOMING_X_POINT 30  // X point for Z homing
//  #define Z_SAFE_HOMING_Y_POINT Y_CENTER  // Y point for Z homing
  #define Z_SAFE_HOMING_Y_POINT 30  // Y point for Z homing
#endif
`
If I try to send the nozzle to HOME it goes ok for X and Y, then the following message appears: "Z probe past bed".

Then I tried to increase the two values values from 30 to 50.
It worked, but surprisingly the home position is now like +50mm in Y but close to 0 in X, looks like there is an offset along X somewhere in the code.
Now I understand why, when homing with the previous code #define Z_SAFE_HOMING_X_POINT X_CENTER, it was not to the center but slightly to the left.
Is this behavior correct?

digant73, 2021-03-05:
Hmm with the MBL version you should have "#define NOZZLE_TO_PROBE_OFFSET { 0, 0, 0 }". I probably forgot to reset it on the MBL version
Try that and it should work also with Z_SAFE_HOMING_X_POINT 0.
I will fix it in the next release

thineight, 2021-03-05:
Interesting stuff...
With the following settings:
#define NOZZLE_TO_PROBE_OFFSET { 0, 0, 0 } // was 33, -33, 0
  #define Z_SAFE_HOMING_X_POINT 40  // X point for Z homing
  #define Z_SAFE_HOMING_Y_POINT 20  // Y point for Z homing

when I home the z the nozzle goes at coordinates 7, 53mm (coordinates are shown in the Move menu)
Looks like there is another 33 and -33 somewhere..
In fact:
40-33 = 7
20+33 = 53

Looks a bit odd for a novice like me.. I let you experts to sort this out, at least now it homes near the origin and not somewhere in the center.

digant73, 2021-03-05:
reset eeprom

thineight, 2021-03-05:
Aaaahhh you are damn right. After reset the it homes at 40,20..
Thanks!!

thineight, 2021-03-09:
Is it a good way to get noticed when you fix the bugs on the 2.0.7.2? Should I see updates here or on your repo on thingiverse?
Thanks

digant73, 2021-03-09:
check here first

oldman4U, 2021-03-11:
Fix is available. Please check

PR #1716 (2021-03-11): Fixed Mesh Editor + other fixes and cleanup
* fixed wrong mesh grid visulaization for MBL (#1706) in  Mesh Editor (e.g. a 5x5 grid was displayed as 5x4)
* fixed missing labels for many languages for last merged PR  #1706
* reviewed string parsing order in parseAck.c to improve general performance
* Added new param "z_raise_probing" on config.ini to support also Delta printers on menus Probe Offset / Mesh Editor for 
  the Z offset / Z height tuning process. It improves PR #1670
  **NOTE:** It MUST be negative (e.g. -50mm) for a Delta printer to avoid crashing into the top of the tower.
* some  [... cut]
closed 2021-03-11 by bigtreetech