bigtreetech/BIGTREETECH-TouchScreenFirmware #1636: TFT24 v1.1 | SKR 1.3 | TMC2130 | No printer attached

Wyso77, 2021-02-20:
Hi. Please help me configure the screen TFT 24 v1.1. For a week I have trying to configure it unsuccessfully. The problem is "No printer attached". I used various videos on YT, help on GitHub and nothing...

I have LCD: TFT 2.4 v1.1, board: SKR 1.3 with TMC1130 and I check (old and new) original LCD firmware and Marlin firmware (old and new - 020007). My settings:

#define SERIAL_PORT -1 (or mix like e.g 0)
#define SERIAL_PORT_2 0 (or mix like e.g uncomment or -1)
#define BAUDRATE 115200 (or 250000)

[image]
[image]

In screen is TMC2208 and reset pin is disconnected
I have no ideas how to fix it. Please help me.
Best regards.

radek8, 2021-02-20:
Take a better photo of the connection, including the display on the side of the connectors.
Submit configuration files to Marlin

Wyso77, 2021-02-21:
> Take a better photo of the connection, including the display on the side of the connectors.

[image]
[image]
[image]

> Submit configuration files to Marlin

[configuration_configuration_adv.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6016929/configuration_configuration_adv.zip)

Wyso77, 2021-02-21:
> Take a better photo of the connection, including the display on the side of the connectors.

[image]
[image]
[image]

> Submit configuration files to Marlin

[configuration_configuration_adv.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6016945/configuration_configuration_adv.zip)

radek8, 2021-02-21:
It looks fine
Make sure you have the same baud rate set.
Depending on the file you sent, there will be 115200.

Try two tests.
1. If you click Home, will the printer execute the command and find the home position?

2 connect the PC, connect with the Pronterface program to the printer. they sent orders one by one
M503,
M114
M115
Does "No printer attached" disappear?

oldman4U, 2021-02-22:
Connect 12-24v power to the mainboard and try it again.

Wyso77, 2021-02-22:
> It looks fine
> Make sure you have the same baud rate set.
> Depending on the file you sent, there will be 115200.

Yes. TFT and mainboard is set on 115200.

> Try two tests.
> 
> 1. If you click Home, will the printer execute the command and find the home position?
> 2. Connect the PC, connect with the Pronterface program to the printer.
>    Gradually went commands from the PC
>    M503,
>    M114
>    M115
>    Does "No printer attached" disappear?

1. Printer no react. LCD12864 mode is work fine but touch mode still "No printer attached".
2. In Pronterface I have answer from mainboard.

M503:
`SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 T0 D1.75
echo:  M200 T1 D1.75
echo:  M200 S0
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00
echo: M92 T0 E100.00
echo: M92 T1 E100.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00
echo:  M203 T0 E25.00
echo:  M203 T1 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X1500.00 Y1500.00 Z100.00
echo:  M201 T0 E5000.00
echo:  M201 T1 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R2000.00 T4000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Material heatup parameters:
echo:  M145 S0 H200 B40 F0
echo:  M145 S1 H240 B90 F0
echo:; PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:  M304 P10.00 I0.02 D305.40
echo:; Stepper driver current:
echo:  M906 X800 Y800 Z800
echo:  M906 T0 E800
echo:  M906 T1 E800
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 T0 E
echo:  M569 S1 T1 E
echo:; Tool-changing:
echo: Z2.00`

M114:
`SENDING:M114
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0`

M115:
`SENDING:M115
FIRMWARE_NAME:Marlin 2.0.7.2 (Feb 21 2021 02:27:18) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:XXX EXTRUDER_COUNT:2 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:RUNOUT:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:1
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:AUTOREPORT_SD_STATUS:1
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
area:{full:{min:{x:0.00,y:0.00,z:0.00},max:{x:200.00,y:200.00,z:200.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:200.00,y:200.00,z:200.00}}}`

I tested this TFT on SKR 1.4 (another board) and dont work. Maybe TFT is damaged?

--------------------------------

>    Connect 12-24v power to the mainboard and try it again.

My printer work in 24V power but I testet in 12V and TFT touch mode no reaction :(.

radek8, 2021-02-22:
Have you tried what I wrote?
How did it end up?

Depending on the configuration files, you will not see the Marlin mod either.
In the configuration.h file
uncomment
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
and comment
//#define REPRAP_DISCOUNT_SMART_CONTROLLER
//#define ST7920_DELAY_1 DELAY_NS (0)
//#define ST7920_DELAY_2 DELAY_NS (250)
//#define ST7920_DELAY_3 DELAY_NS (250)
Compile a new Marlin and upload to SKR.

In the config.ini file, set
marlin_type: 0
and upload the configuration to TFT

After connecting the EXP1 and EXP2 cables, you will operate the Marlin Mod and test whether the Marlin is working properly.

radek8, 2021-02-22:
If it does not work with another board, the display or RS232 cable will be defective.
Try another cable

radek8, 2021-02-22:
The submitted configuration does not correspond to the parameters you specify.
Are you really set to the correct speed? You have sent another printer configuration.

The extruder steps are different in Configuration.h-46 and in response M503-100.
You state that you have TMC2130 and you have 2 extruders in the configuration, so you should define 5 TMC2130 drivers.
But you only have 4x TMC2208 in the configuration

How you want to help when you give bad information.

Wyso77, 2021-02-22:
> Have you tried what I wrote?
> How did it end up?

In Pronterface working all from Windows system
In TFT nothing happen:

[image]

> Depending on the configuration files, you will not see the Marlin mod either.
> In the configuration.h file
> uncomment
> #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> and comment
> //#define [...]

Result of the changes (in particular "marlin_type: 0"):

[image]
[image]

> If it does not work with another board, the display or RS232 cable will be defective.
> Try another cable

I tried on my pin cable:

[image]

I think in my TFT is a corruption.

> The submitted configuration does not correspond to the parameters you specify.
> Are you really set to the correct speed? You have sent another printer configuration.

Yes, is correct. 100%.

[image]
[image]

> The extruder steps are different in Configuration.h 46 and in response M503 100.
> You state that you have TMC2130 and you have 2 extruders in the configuration, so you should define 5 TMC2130 [...]

Because we testet in SKR 1.3 with TMC2208 now. Related problem too SKR with TMC2130. I send for you Configuration.h and Configuration_adv.h with TMC2208 stepper driver. I can not carry massive printer 3D with TMC2130 - but I would be glad if it worked in the same board with TMC2130 which I use always.

[image]
[image]

radek8, 2021-02-22:
I compared the description on the display with the tft35 display
I noticed that your description on the display is a confused description of the RX and TX pinet. RS232 connector.
Can you try to swap them and see if it has the opposite connection?

Wyso77, 2021-02-24:
> I compared the description on the display with the tft35 display
> I noticed that your description on the display is a confused description of the RX and TX pin. RS232 connector.
> Can you try to swap them and see if it has the opposite connection?
> 
> The order of the pins on the SKR is 5V, GND, TX, RX, RST
> The order of the pins on the TFT35 is 5V, GND, TX, RX, RST
> The order of the pins on the TFT24 is 5V GND RX, TX, RST

No. Still nothing.

I bought a new TFT 24 board. Everything work fine. Touch mode too. The "No printer attached" is gone.  I think that old board has burn UART and does not work (maybe my fault). I close this topic. Very thanks for help radek8 and oldman4U.

Best regards.

oldman4U, 2021-02-24:
Happy printing

closed 2021-02-24 by Wyso77