bigtreetech/BIGTREETECH-TouchScreenFirmware #1637: Load/unload menu Nozzle button does not change the displayed Nozzle Number (T0 etc)

jimhiggs, 2021-02-20:
Searched but could not find this issue listed. 
Board: TFT35V2 
Firmware:  26x, Current?

### Description
Load/unload menu Nozzle button does not change the displayed Nozzle Number (T0 etc)

Solution:
In LoadUnload.c Line #100 is:
`curExt_index = (curExt_index + 1) % infoSettings.hotend_count;`
Should be:
`curExt_index = (curExt_index + 1) % infoSettings.ext_count;`

kisslorand, 2021-02-20:
Good finding but hard to know which nozzle has to be heated. You can have IDEX or only DEX.
Someone introduced nozzle temp check before Load or Unload and it assumes you have IDEX, you have to have the nozzle attached to the extruder heated.

jimhiggs, 2021-02-20:
I only have one mixing nozzle but three extruders so its great this now works for me.

kisslorand, 2021-02-20:
I do not think it can be made "one size fits all". What if you have 2 carriages, one regular and the other one a mixing one with 5 etxruders? It's a nightmare to implement every possible situation/config.

jimhiggs, 2021-02-20:
That is true. Seems to me we have mixing nozzle and multi-extruder fairly well sorted. At least at my fairly basic level of understanding.

oldman4U, 2021-02-24:
Hi jimhiggs

Please help the community and close the ticket once you do not need it anymore. Thank you

jimhiggs, 2021-02-25:
Has the bug been taken up for fix?

oldman4U, 2021-02-25:
@kisslorand 

Anything you pr another developer can fix here?

oldman4U, 2021-02-28:
No reply is also an answer.

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-02-28 by jimhiggs