bigtreetech/BIGTREETECH-TouchScreenFirmware #1639: [Q] Unable to get 3DPROBE to HOME Z-Axis

doug-scott, 2021-02-21:
Using the following hardware: 
MKS SGEN_L V2.0
MKS TMC2208 in ANSI mode via jumprers
BIGTREETECH TFT35 E3 V3.0 Graphic Smart Display Control Board 
Generic 3DTouch probe 

I can move the hotend around the printer via g-code commands, except for the homing commands, G28 and G29.  Both commands will home Y and then X axis, then move hotend to midpoint of print bed where it will deploy and stow the probe twice.  It will then produce a very meaningful error message that states 
    !!STOP called because of BLTouch error - restart with M999 

I have been trying to find a solution to this issue for quite a while now.  It fails in both modes, touch screen and 12864 modes.

[Configuration_files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6018351/Configuration_files.zip)

digant73, 2021-02-21:
You probably wrongly connected the probe cable or you have the cable with no continuity

doug-scott, 2021-02-21:
I have been through the 5 wires many times thinking the same thing.  Is there a listing of what each wire does at what point in time?  Like a debugging logging option that can be set to capture what the probe is sending/receiving/expecting. etc ?  Perhaps an option to capture the real time activity and capture the error that is generating the error that requires the M999 to reset?

digant73, 2021-02-22:
in my case the problem was due to a broken wire (the withe or black cable). That cable was the probe signal and I got the same problem as you. However, if the wires are ok, it seems that you connected the 2 probe wires in the wrong connector. From your configuration, the 2 probe wires (balck and white) must be connected to the Z min endstop

oldman4U, 2021-04-08:
@Msq001 For sure not a BTT TFT firmware issue.

closed 2021-04-08 by bigtreetech