bigtreetech/BIGTREETECH-TouchScreenFirmware #1642: TFT43 Screen goes to static when using BLTouch menu (Touch Mode)

agenttatdaddy, 2021-02-22:
### Description

TFT43 Screen gets all static and unusable causing me to reset the screen. Using SKR E3 Mini V2.0
### Steps to reproduce

1. Turn printer on
2. Enter "Menu"
3. Enter "BLTouch" menu
4. 
**Expected behavior**
I expect it to enter the BLTouch menu as usual

**Actual behavior**
The screen starts to get static and fuzz at top of the screen and then works it way down the entire screen cause me to have to use the reset button to restart Touch screen.

### Hardware Variant
TFT43 V3.0 
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT43_V3.0.26x & Marlin 2.0.7.2 configured for Ender 5 Plus

### Additional Information
[configh.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6025397/configh.zip)

oldman4U, 2021-02-24:
Hi.

2 days no answer could mean,

a. that nobody else has the same issue
b. that the bot had to close your last ticket and radek8 never got an answer from you to his last reply, which is unfriendly somehow.

agenttatdaddy, 2021-02-25:
Yeah, I was having an issue with not getting the emails whenever someone replied. But that is corrected now.

oldman4U, 2021-02-25:
Wonderful.

I would repeat the firmware update with another supported sd card.

agenttatdaddy, 2021-02-25:
I flashed it with the current Unified Theme and it works fine now. It would seem the rounded theme has a bug somewhere that messes the screen up when using ABLE.

oldman4U, 2021-02-26:
Ok

Please help the community and close the ticket once you do not need it anymore. Thank you

agenttatdaddy, 2021-02-26:
Will do! And thank you for the help and advice. I really appreciate it. I am still kind of new to the community. So it is a learning curve for me.

agenttatdaddy, 2021-02-26:
Thanks again!

closed 2021-02-26 by agenttatdaddy

oldman4U, 2021-02-26:
Happy printing

oldman4U, 2021-02-26:
Hi.

Wanted to let you know that I verified that all 3 themes have a complete set of bmp's in the firmware released 2h ago.

Happy printing