bigtreetech/BIGTREETECH-TouchScreenFirmware #1644: Update fails due to "Outdated data" TFT35 v3 E3

phongshader, 2021-02-23:
After downloading the most current, as of 2/22/21, zip file and copying over "Copy to SD Card root directory to update"
BIGTREE_TFT35_V3.0_E3.26.x.bin
config.ini
Language Packs
THEME_Hybrid Red Menu Material theme
THEME_The Round Miracle Menu Material theme
THEME_Unified Menu Material theme
to the root of a 8gb fat32 formatted sd card I get this error:

I cannot complete the update with the most current files from github
```
Found: Outdated data:
Fonts: OK
Config: OK
Language: Update Required (Optional)
Icons: Update Required

Insert SD card with required files
```

### Steps to reproduce

1. download zip file from github, expand
2. copy 
BIGTREE_TFT35_V3.0_E3.26.x.bin
config.ini
Language Packs
THEME_Hybrid Red Menu Material theme
THEME_The Round Miracle Menu Material theme
THEME_Unified Menu Material theme
to the root of a 8gb fat32 formatted sd card
3. reboot
**Expected behavior**
firmware/config is updated

**Actual behavior**
update fails due to "Outdated data"

### Hardware Variant
tft35 v3 E3
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.26.x.bin 02/02/2021

[image]

radek8, 2021-02-23:
You must copy the TFT35 folder to the card
Depending on which icons you want to use, you can pull TFT35 from that folder

radek8, 2021-02-23:
If you want to use a language pack, you will load a specific language pack into the root of the SD card.
Not the whole folder with all the packages and all the topics ...

phongshader, 2021-02-23:
That did the trick, Thank You!!

closed 2021-02-23 by phongshader