bigtreetech/BIGTREETECH-TouchScreenFirmware #1645: [FR] (G26 mesh validation )

labotecno, 2021-02-23:
hello
 in marlin we have G26 mesh validation its a usefull option  , but when i get enable , i get a windows tft error :"G26 repeat must be specified when not using on lcd"

labotecno, 2021-02-23:
i was see i need to put G26 R1  but now its give me "not sd printing ?"

labotecno, 2021-02-23:
ok its working like this way G26 R99  if you put only one its not working 
it s possible to put one icone on abl menu ? thank you

labotecno, 2021-02-25:
someone here ?

labotecno, 2021-02-26:
[https://github.com/labotecno/BIGTREETECH-TouchScreenFirmware](url)

oldman4U, 2021-02-26:
You are not alone 🤚🏻

labotecno, 2021-02-27:
> 
> 
> You are not alone 🤚🏻

can you tell me if PR is ok ?

oldman4U, 2021-02-27:
is it available? can not see it in the PRs...;-(

oldman4U, 2021-02-28:
Do you need help pushing the PR? On Mac you can push the local repository directly to a PR on GitHub. It's magic how this works. 🤩

labotecno, 2021-03-06:
> 
> 
> Do you need help pushing the PR? On Mac you can push the local repository directly to a PR on GitHub. It's magic how this works. 🤩

ok i think i get it now , but i have file on conflict i hope its not a problem , if is it how to resolve it ?

Guilouz, 2021-03-06:
> > Do you need help pushing the PR? On Mac you can push the local repository directly to a PR on GitHub. It's magic how this works. 🤩
> 
> ok i think i get it now , but i have file on conflict i hope its not a problem , if is it how to resolve it ?

Because you don't use latest master with your pull request.

labotecno, 2021-03-07:
> Because you don't use latest master with your pull request.

where is the latest master ? thank you

kisslorand, 2021-03-07:
Here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

ETE-Design, 2021-03-07:
@labotecno Seems like your PR disapeared?

labotecno, 2021-03-07:
> 
> 
> @labotecno Seems like your PR disapeared?

i get some problem with conflict but now i think its ok ! thank you every one [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1688](url)

digant73, 2021-04-06:
@labotecno please close this FR now that it was merged to main branch

labotecno, 2021-04-06:
> 
> 
> @labotecno please close this FR now that it was merged to main branch

ok ready !

closed 2021-04-06 by labotecno

ETE-Design, 2021-07-26:
@labotecno Did you ever got G26 merged so it has an Icon on the TFT?