bigtreetech/BIGTREETECH-TouchScreenFirmware #1646: [FR] Use probe to level corners

Guilouz, 2021-02-23:
Possible to use probe to level corners in Marlin Mode when #define LEVEL_CORNERS_USE_PROBE is enabled in Marlin but there is no option to do this in Touch Mode. It would be nice if it could be implemented.

**Description**
Add a option to level the bed corners using a probe. Instead of using the nozzle to paper ajust the screws, it will probe the first corner (or the center) and save the z position. Then will go to the next corner to check if is whitin tolerance range. If is in tolerance, will go to next corner. If is not, will leave the probe deployed waiting the user to raise the bed util the probe is trigered. When all corners are within tolerance, will present a option to execute bed leveling.

https://github.com/MarlinFirmware/Marlin/pull/20241

oldman4U, 2021-02-24:
It seems, that you are a very smart developer....

;-)

Seelenkind, 2021-03-03:
> #define LEVEL_CORNERS_USE_PROBE

is implemented in Touchscreen 3.5 E3 V3

Cal1sto, 2021-03-21:
Feature probe level corner #1742 : Will give you the following function , it will probe all the 4 corners , display the Z value measured on each corner, then you could adjust each corner individualy and you could use your own tolerance based on the probe precision that you could get with the M48 (this test with M48 will be added in this Level corner function later if needed). You could scan all corner again and again or check and adjust individualy the 4 corners in cross adjustement or linear adjustement manually. Like the manual leveling function you have also the possibility to set the corner position test, this is limited already with a minimum value set depending of the probe offset to the nozzle to reduce the risk the printing head is going outside the X and Y bed area, set a reasonable lowest value to measure nearest the 4 corners leveling screw of the bed.

oldman4U, 2021-03-21:
@Guilouz

Could you please test the PR  and let us know.

Thank you

Guilouz, 2021-03-21:
> @Guilouz
> 
> Could you please test the PR and let us know.
> 
> Thank you

Nope, not working, see my post on this PR.

oldman4U, 2021-03-28:
Is it better now?

closed 2021-04-07 by bigtreetech