bigtreetech/BIGTREETECH-TouchScreenFirmware #1647: TFT35 3.0 No option for Rotary knob LED

martyz242, 2021-02-24:
Just got the new TFT35 V3.0 along with SKR V1.4T. Uploaded the most current firmware, and updated the config file to turn on the Rotary Knob LED, however, the LED is not working. Also, the 2 options (color and sleep) for "Rotary Knob LED" in Menu-Settings-Features are gone. Is this a bug in the latest firmware build?

oldman4U, 2021-02-24:
No bug but wrong TFT without LED. 

Please help the community and close the ticket once you do not need it anymore. Thank you

martyz242, 2021-02-24:
I purchased it directly from biqu. Are you telling me that they have multiple versions of the same screen with no indication on there site of which one is being sold?

oldman4U, 2021-02-24:
Please contact the manufacturer. I do not have such a printer and this repository is about BTT Touchscreen. There is a repository for B1 printers.

bigtreetech, 2021-02-24:
Hello, Please confirm your hardware version. TFT35 V3.0 has no "Rotary Knob LED", and the "Rotary Knob LED" version is TFT 35 E3 V3.0. If your have TFT35 E3 V3.0, please select the correct firmware named " BIGTREE_TFT35_V3.0_E3.26.x.bin" to update

oldman4U, 2021-02-24:
Sorry, my fault, not printer related.....

Like BTT wrote, there are several versions of TFT's also in 3.5", with and without LED. Seems you bought the one without which is easily to figure out when you look at the knob itself.

martyz242, 2021-02-24:
Ok, so only the E3 version has the knob led and the regular version does not? That would have been nice to know prior to me ordering since there in no indication of that on the biqu site.

I will close this issue now. Thanks.

closed 2021-02-24 by martyz242