bigtreetech/BIGTREETECH-TouchScreenFirmware #1648: BTT TFT35 E3 v3.0 - Backlight Flashes but Screen Blank after Firmware Update

Heratiki, 2021-02-24:
The screen worked as designed before the firmware update.  config, BIGTREE_TFT35_V3.0_E3.26.x.bin, TFT35 from THEME_Unified Menu Material theme placed on an SD Card.  Screen displays correctly during the firmware update.  After the Fonts and Icon's are displayed during the update the Screen turns dark and begins flashing the backlight.  This stays like this until the power is cycled.  At which point the backlight flashes a few times and then stays on but the screen is still blank.

BTT SKR v1.4 Turbo + 4x2209 Stepper Drivers running Marlin-bugfix, BTT TFT35 E3 v3.0, Wired correctly with the Serial cable correctly installed.  Custom Marlin on the SKR Turbo working.  I've included my config file just for posterities sake.  Looked all over for this issue but haven't found anything.  This is after trying to understand why the Marlin Simulator Mode wasn't working with the firmware on the display.  I'm ready to throw it all in the trash considering I've already been through 5 bad stepper drivers from BTT and on the second SKR Turbo in just 3 weeks. 

[config.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6038613/config.txt)

radek8, 2021-02-24:
Did you also connect the EXP1 and EXP2 cables?

radek8, 2021-02-24:
Did you use a hybrid red theme? :-)

Heratiki, 2021-02-24:
I do have both cables connected via EXP1 on the display to EXP1 on the SKR Board and EXP2 on the Display to EXP2 on the SKR Board.  I did not use the Hybrid Red Theme though when I tried that theme the same issue happens.  I'm currently attempting the firmware update from a different computer as I think I could be getting corruption while transferring files to the SD Card.

Tested using my Macbook Pro and now instead I get Illegal FLASH app at the end.  Sigh.  Maybe its the SD Card?  Who knows this is frustrating.

radek8, 2021-02-24:
2 ions were removed prematurely
Copy these 2 icons to the BMP folder.
[Ikony.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6038702/Ikony.zip)

Heratiki, 2021-02-24:
Testing again.  I've included the 2 BMP files you zipped and linked.  Waiting on the update. I've reverted the config file back to it's default for now just to rule out any further issues.  I've also got another SD Card I can use for testing as well.

oldman4U, 2021-02-26:
I just verified that the PR from 2h ago has all bmp's of all 3 themes correctly added to the firmware.

Happy printing

Heratiki, 2021-02-26:
Yes it finally flashes the firmware but I was getting only a flashing back light after  when using the MASTER branch. I had to work late last night so I haven't had a chance to do much. I plan to do more tonight. 

I found a method of modifying the EXP3 port on the display to use a single cable rather than 2 that I'll be testing as well. Apparently with the SKR 1.4 Turbo and the TFT35 E3 v3.0 it gets caught in a reset loop due to a strange way the two interact. I'll explain more later. 

But as of right now flashing the 26.x master branch to my TFT35 E3 v3.0 connected to my SKR 1.4 Turbo and setup as REPRAP DISCOUNT SMART DISPLAY CONTROLLER in Marlin just causes my screen to flash the backlight when connected via the EXP1&2 ports and the RS232 port. 

The display will function as a touchscreen only when i flash it with the release-xx.26.1 branch. Marlin Simulator does not work with that firmware for some reason.

radek8, 2021-02-26:
You must connect the EXP1 and EXP2 connectors on the SKR V1.4 to the EXP1 and EXP2 connectors on the TFT
If you have SKR V1.4 YOU MUST NOT USE the EXP3 connector on the display !!!

Heratiki, 2021-02-26:
I understand how to connect it correctly but my display does not function correctly when I connect it this way. With its current firmware connecting it this way only allows touchscreen use. The Marlin mode does not work. I'll take a video and show later.

radek8, 2021-02-26:
I guess I found where your problem is

radek8, 2021-02-26:
You have the wrong version of the CONFIG.INI file.
You need to set
marlin_type: 1
You have changed the comment on this item and therefore have the wrong value set.

[image]

Heratiki, 2021-02-26:
I'll give that a shot in just a moment and test.

radek8, 2021-02-26:
The condition is the connection of EXP1 and EXP2
In Marlin permit
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Heratiki, 2021-02-26:
Ok so now I have a different issue. The current firmware flashes correctly but now the touchscreen does not work at all.  The firmware flashing goes through Marlin Mode appears to work but the touch screen does not.  And unlike the last time I got release 26.1 working it doesn't ask to do the touchscreen calibration after it finishes flashing the firmware. What exactly is going on?

radek8, 2021-02-26:
perform calibration using the reset.txt file

kisslorand, 2021-02-26:
If you keep pressing the touchscreen does it go to the Marlin/Touch selection menu?
If yes, than put an empty reset.txt file on your SD card and reboot the TFT.

Heratiki, 2021-02-26:
I'll try the reset.txt file I believe that's probably the issue. I'd completely forgot considering all the other issues I'd had.  I changed nothing in my config.ini from previously radek8 but just used the firmware binaries that were available today.  The display flashed correctly and appears to be working correctly across the board.  Thank you so much for sticking with me through all this. I'll do what I can to try and clean up the firmware flashing instructions to try and make it easier for those that have the most basic of issues.  No clue what fixed my issue but no more flashing backlight. I'll be closing this now.

closed 2021-02-26 by Heratiki