bigtreetech/BIGTREETECH-TouchScreenFirmware #1649: [Q] BTT TFT35 E3 v3.0 Blank Screen and Flashing Backlight after Firmware Update

Heratiki, 2021-02-24:
**Your question or details you need. Please describe.**
How do I go about troubleshooting and fixing this issue and getting my display working again.

**specify the device variant related to your question.**
BTT TFT35 E3 v3.0

**Additional context**
I've flashed my display several times now with different SD Card Readers, Different size and style and type SD and CF cards.  After successfully flashing the firmware the screen shows nothing on it and the backlight just flashes. Powering the printer off and removing the SD card then powering it back on causes the display to be blank and the backlight to flash 5 times then stays on.  The touchscreen does nothing and the encoder does nothing.  I've used the included BIGTREE_TFT35_v3.0_E3.26.x.bin file as well as compiling my own.  The printer functions normally via Octoprint even though I can't see the display.  The display is connected using the 3 cables (2 EXP cables and the Serial RS232 Cable).  The display was working fine before I attempted the firmware update.  Holding down the encoder wheel does not allow me to switch to the Marlin Simulator (in case anyone wanted me to try that).  Any help would be greatly appreciated.

radek8, 2021-02-24:
After the flash, will the files be overwritten to the CUR extension?
BIGTREE_TFT35_V3.0_E3.26.x.CUR
config.ini.CUR
TFT35.CUR

Heratiki, 2021-02-24:
Yes they are all overwritten and have the .CUR extension.  I'm attempting to flash the 26.1 branch instead of the current MASTER branch from GitHub.

radek8, 2021-02-24:
For Marlin mode to work, Marlin must have a display driver enabled in Marlin
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
other drivers may be commented.

radek8, 2021-02-24:
You can shoot a video with the progress of the flash?

Heratiki, 2021-02-25:
> You can shoot a video with the progress of the flash?

I'll do that this afternoon after work so you can see. I'll also show you my connections as well.

Heratiki, 2021-02-26:
I'll shoot a video tonight when I get home. Work ran late and by the time I was home I just didn't have the energy to mess with it.

oldman4U, 2021-02-27:
Please see #1648

Heratiki, 2021-02-27:
> Any news?

After downloading today's firmware and flashing my display everything seems to be working fine now. No clue what changed but something was just playing havoc with the last bin file I had.

closed 2021-02-27 by Heratiki