bigtreetech/BIGTREETECH-TouchScreenFirmware #1650: [BUG] TFT35 V3.0 Stop printing random when printing from TFT35 USB

Rodov, 2021-02-25:
### Description

TFT35 V3.0 Stop printing random when printing from TFT35 USB

### Steps to reproduce

1. Start to print

**Expected behavior**
Normal printing expected

**Actual behavior**
Printing stops at random point and TFT hangs.

### Hardware Variant
TFT35 v3.0 and SCR 1.4 turbo

### TFT Firmware Version & Main Board Firmware details
TFT firmware - 26.1
Mainboard firmware - Marlin 2.0.7.1

### Additional Information
The issue is similar to **TFT35 V3.0 Stop printing random when printing from TFT35 SD #244** 

My USB drive has a led and I noticed, that USB drive has no power when TFT hangs. If I remove USB drive then TFT is going alive and can hold commands. Though reinsert the USB drive back is not lead to normal print - printer can't print at all. But can pause the job.
After powercycle, the printer can continue the job and finish the model correctly.

I have tried to change the USB drive with no result.
Marlin config has this set up:
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
```
Also I have reduced the baudrate from 250000 to 115200 with no effect.

At the same time printing is going fine without any bugs in LCD simulation mode.

oldman4U, 2021-02-25:
Which power supply and voltage do you use?

Is there any indication that the firmware of the TFT causes this. I ask because this is the repository related to the firmware which is running on top of the screens hardware.

radek8, 2021-02-25:
I think it's the same problem with the display freezing as here
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1629

oldman4U, 2021-02-25:
Only in case he presses buttons during the print.

Most of those problems in the past came from PC power supplies used in self made printers.

radek8, 2021-02-25:
I do not agree
I noticed it myself.
I have a quality Ender3 source.
After an hour of printing without clicking on the display, the display froze and the printer froze. Marlin continued to heat the pad and nozzle, but the display did not send new coordinates.

oldman4U, 2021-02-25:
My ender must be different then. I haven't seen the freezing display so far but will investigate further.

Rodov, 2021-02-25:
I use computer ATX PSU 650 watt. I am not touch any buttons on the screen during print.

Rodov, 2021-02-25:
> Is there any indication that the firmware of the TFT causes this.

I think it's about TFT's firmware, because in LCD simulation everything is going okay.

borupdk, 2021-02-26:
I have almost the same issue, but here the culprit is the SD card. In fact both SD card and USB fails, but not always..!

Rodov, 2021-02-26:
> I have almost the same issue, but here the culprit is the SD card. In fact both SD card and USB fails, but not always..!

As I mentioned above (in description) the issue is similar to issue #244

oldman4U, 2021-02-26:
And how many solved the problem there by using a certain TFT firmware version?

Those reported the issue and found a solution for it found something in Marlin or hardware.

If it is only in the TFT firmware every user would have it and this only with certain versions.

I have an Ender3 Pro with SKR E3 DIP and TMC 2208 as well as TFT35 E3 v3 and never saw an interrupted print or something similar. The only issue is with purge more and this is STM32 related.

borupdk, 2021-02-26:
I have tried many different tactics, changing the serial_port, increased the buffer size in marlin, and yet it continues to fail. I have also tried different SD cards, and usb drives, but the print seems to fail approximately around the same area every time (for one specific g-code!) i tried a different gcode recently, which seems to go OK. 

Printing using the Marlin mode / 12864 simulator works every time. the UART communication simply drops. I have also tried disabling all other unused UARTs on the TFT50 (which I use)

Rodov, 2021-02-26:
> [skipped]  the print seems to fail approximately around the same area every time (for one specific g-code) !

Can you show the G-code which is lead to fail the print?

Rodov, 2021-02-26:
> If it is only in the TFT firmware every user would have it and this only with certain versions.

As you can see, many people are experiencing problems with printing via USB or SD (see issue #244 )

borupdk, 2021-02-26:
[cable-train1v4.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6049610/cable-train1v4.zip)

It fails between line 1500 and 2000 (approximately) its a rather straight forward print, no weird geometry or anything like that.

Rodov, 2021-02-26:
Maybe it's long extrusion?

radek8, 2021-02-26:
@Rodov ,
don't refer to the 2 year old problem all the time. Since then, FW has undergone great development and today it is completely different.
But I also have to confirm that I also have a problem with occasional freezing. :-(

oldman4U, 2021-02-26:
Taking the number of TFTs sold, the number of users having issues is ignorable.

Even more as the number of users reporting issues here where the problem is really the TFT is also ignorable.

I only say this to point you into the right direction.

Radek. Think about all the issues we have seen here related to bad storage or user errors. 😀

borupdk, 2021-02-26:
Should I attempt at updating the bootloader? I got an ST Link v2 at hand anyway.

radek8, 2021-02-26:
No. Botoader can't blame it

borupdk, 2021-02-27:
I recorded two videos, part 1 is just before it acts up, and part 2 is when it stops printing, I hope its the same issue that other people are facing. 

Part 1 : https://youtu.be/_eLvH_c4OJk
Part 2: https://youtu.be/6JILM9xLTeI

fmarkf, 2021-02-28:
I also have this issue when printing from USB or SD Card. 
Tried to troubleshoot it with different baudrates, USB drives, SD cards. Nothing seems to help. 
Would appreciate any help. Printer Ender 3. Bought directly from BIG TREE TECH CO.,LTD Store on AliExpress. 
Unfortunately seller does not reply to my messages. TFT is connected to SKR MINI E3 V2.

oldman4U, 2021-02-28:
The problem I have with this ticket is, that another ticket with the exact same title and topic already exists and instead of using it the information what you can do is now cluttered and split.

I wish you all that you find the reason for your problem. It is highly unlikely the firmware and this repository is about the firmware and nothing else.

Good luck

fmarkf, 2021-02-28:
Unfortunately for us (having this issue) there is no much more of what we can do to find the solution. So we ask Firmware community for help. If this is a hardware problem, we need to figure that out somehow. I think, tickets like this could help with the investigation. Other tickets (referenced in this issue) are similar, but not the same.

Rodov, 2021-03-01:
> Most of those problems in the past came from PC power supplies used in self made printers.

What hardware and firmware do you use?

Anyway, you didn't give us any clue or solution to our problem.
Most of us (or all) checked any of circumstances which is possible to check for us and it didn't help.
So, we decided to ask firmware community (and you) to check any other ways to solve the problem.
If the problem is in hardware or any configuration then you can point it to us, it will help alot.

oldman4U, 2021-03-01:
I use TFT35 E3 v3 on both printers with SKR E3 DIP/2208 and SKR PRO V1.1/2209 mainboards. Marlin from last week and TFT firmware one week old and an unreleased PR on the other. Both printers with Meanwell 24/350 power supplies.

Because I do not have this issue using the same TFT firmware, I can tell you that the firmware is the less likeliest reason for it. If I would exactly know what it is I would tell you and I tried to point you in the direction which is the most likely for me.

borupdk, 2021-03-01:
I think that's anecdotal evidence, at best, but OK lets assume for a second that its not the firmware thats at fault, I just have to mention, that i have installed a ferrite choke on the UART cable, and wound it around a few times, in order to hopefully mitigate this issue. I also connected an external wallplug 5V power supply to the LCD, but yet, it still fails.

oldman4U, 2021-03-01:
Maybe you have a faulty TFT?
Or the firmware got corrupted because the sd card was wrong or broken or the firmware was the wrong one, or....

borupdk, 2021-03-01:
I have mentioned, previously, that i have successfully update the firmware before, i have performed the update at least 3 times, with no issues.

oldman4U, 2021-03-01:
.... and you still believe that it is the firmware which is the problem. How do you explain yourself that there are users using the same firmware without the issue you see?

borupdk, 2021-03-01:
oldman4U, I would appreciate constructive criticism, but you have yet to provide any actual insightful knowledge regarding the problem. I am not sure what your agenda is, but you are helping no-one. 

I will counter your argument, what is the reason to why the firmware is 100% bug-free when it comes to printing from the screen? Are all cases covered? is every single conceivable scenario of G-code execution carried out through rigorous testing? 

When you actually provide us a shred of explanation as to why the firmware is NOT the issue, I will hold my tongue, but for now, you have only managed to slightly annoy me.

oldman4U, 2021-03-01:
I try to help by telling you that you are looking for the problem at the
wrong place and I have to provide nothing.

Let me know once you found the problem in the firmware and I will spend you
a beer or two.

Good luck

oldman

Rodov, 2021-03-02:
I am going to change the power source of my printer, so will share the result soon.

DaStivi, 2021-03-02:
i had exact the same issues (but really there could be ah couple of hang issues... but with the known bug of freezing fixed, it should be something else...)  

i just changed from SD Card to USB Pen, what worked greate a dozen of prints... then it started again with the USB Pen... resliced the models, and sometimes it just worked, but sometimes nothing helped... then i just copied the same gcode to another USB Pen and tested with that and voila it just worked again... so just try another USB Stick it might help!

borupdk, 2021-03-04:
I am officially stumped. 

I moved my machine from my workplace, to my home, plugged it in, and ran the same G-CODE, and.... it worked....????????

oldman4U, 2021-03-04:
No. You are not.

But you have an on/off problem which is the worst case which can happen.

But I will drink your beer tonight 😁

DaStivi, 2021-03-04:
@borupdk  i had another freeze today.... even without printing from USB or SD.. i printed from Octoprint instead... BUT i thought twice about this... i did something on the tft while printing with octoprint.. you could please try this....

i got into the babystep menu changed something and saved it!
and i did this also quite often with printing from TFT (SD/USB) just to adjust slightly the first layer... so maybe, maaybee , because its the only thing i really did on the tft now, there is some issue in there...

oldman4U, 2021-03-24:
Hi Rodov. Have you changed the power supply in the meantime?

Rodov, 2021-03-25:
> Have you changed the power supply in the meantime?

Not yet. I have bought a PSU but have no time to change it... ((

Rodov, 2021-04-08:
Didn't changed the PSU on time - the SKR 1.4 turbo was burned out by power surge (I think). Now I installed another SKR board (SKR mini 1.1) and still have not meet interrupted print process. It seems to me it was an hardware problem with SKR 1.4t.
I will keep a close lookon this problem. If something will happend I will let you know.

oldman4U, 2021-04-08:
Seems your problem is not really urgent. Make sure to install the v27 TFT firmware which was released 2 days ago and the new PSU.

Rodov, 2021-04-08:
> Seems your problem is not really urgent. Make sure to install the v27 TFT firmware which was released 2 days ago and the new PSU.

Sure.

billyd60, 2021-04-18:
I have this problem. It is absolutely a serial port connection issue, IMO

oldman4U, 2021-04-20:
Serial always on activates the serial connection also while using Marlin emulation mode.

What is your setup? SKR mainboard, BTT TFT and nothing else connected? This, and an ESP01-S connected to the TFT is my setup and I never saw this issue.

borupdk, 2021-04-20:
Serial always on didn't do a lick of difference for me, only after i moved the printer from one electrical outlet to another it changed.

oldman4U, 2021-04-20:
This is the problem with such tickets. It makes the user believe that there is really a problem with the TFT which can be solved in the firmware. Besides the users which start such tickets but never close it.

billyd60, 2021-04-20:
The only problem I have is I can't use the touch screen to print anything because the serial connection is garbage.  It works when it feels like it. I use a shielded cable too.

radek8, 2021-04-20:
Very often it helps not to place the display close to the heated bed.
another option is to place the display in a metal box.

Be aware that the heated bed is a pulse-powered large-area coil with a current flowing around 15 Amperes. This makes it a frequency interference transmitting antenna, which, if it encounters the frequency of the serial port, can interfere with it by oscillating the voltage on the TFT printed circuit board.

radek8, 2021-04-20:
> Jediný problém, který mám, je, že nemohu použít dotykovou obrazovku k tisku čehokoli, protože sériové připojení je nesmyslné. Funguje to, když na to máte chuť. Používám také stíněný kabel.

Shielded cable is great, but sometimes it's not enough

billyd60, 2021-04-20:
My display is mounted on the acrylic enclosure I use to surround the printer. It is in a 3d printed case. The touch screen is at least 24" away from any electronics (other than it's own). Ironically the IDC cables are unshielded and Marlin mode works perfectly. Maybe I will make up an unshielded serial cable lol.

nobaj, 2021-04-22:
some printers have the same issue printing via octoprint with a raspberry pi, maybe is something with the usb/serial comunication ?

Rodov, 2021-04-23:
Haven't changed a PSU but replaced burned out SKR 1.4 with SKR mini 1.1.
SKR mini 1.1 running with TMC2209 and UART so it must use touch function of the display becouse SKR mini 1.1 can not use UART and 12864 at the same time.
After some hours of printing and some printed things I can assume, that TFT is working properly now.
Haven't seen any hangups or freezes.
So, the problem in my case, was probably in out of order SKR 1.4t board.
Can close the issue now.

closed 2021-04-23 by Rodov

MLiv79, 2021-05-07:
Same issue here with a TFT 70, SKR 1.4T and TMC2209. If i start a print from the TFT (touch mode) using the sd card of the SKR no issue, all the prints finish without problems. If i use the TFT internal usb or SD card reader TFT during print freeze and nozzle stop over the part without any way to move it. I've to restart the printer to move Z up. Power supply is the one included with the printer.

magnificu, 2021-10-31:
I have the same issue with TFT50_v3.0 and TMC5160_v1.2.
When I start the print form the TFT SD Card, randomly the print freeze (sometimes after one or two hours of printing), but the hot end and the bed keeps the temperature. The only way to recover is to press the restart button.
I use the latest TFT firmware version BIGTREETECH-TouchScreenFirmware-master from 30.10.2021.

borupdk, 2021-11-09:
@magnificu - If you ask oldman4u then its a user error, he seems to have an infinite amount of hindsight, and also massive backlog of being a total fucking knobhead without any real inputs regarding how to solve the issue.

borupdk, 2021-11-09:
with all that being said, try moving the printer to a different outlet it worked for me, with absolutely no explanation what so ever.

radek8, 2021-11-09:
@borupdk 
If moving the problem to another outlet has been solved, electromagnetic interference must have been the cause of your problem.

mikaeltarquin, 2022-04-10:
I am also having this issue. Having read through the thread, it is very disappointing to see the replies from oldman4u. I am trying to print from a USB drive (new, high quality aluminum encased 32 GB drive from Samsung), and at random times several hours into the print it freezes. Head and bed stay hot, head stays in place on model, touch screen unresponsive. The last time it happened I was able to keep going by unplugging and plugging back in the USB drive. It unfroze and continued printing immediately.

novygh, 2022-05-05:
sadly it happens to me as well ... read many threads, having latest build but still cannot leave machine on its own! bcs of screen upgrade i made? realy... pls respond...

godl3ss, 2022-06-13:
I solved this printing from Onboard SD, not the TFT one. In order to to this, in the firmware of the TFT, file config.ini and config rff.ini you have to change the SDONBOARD from 2 (auto detect) to 1 and Long names from 2 to 1 as well. 

Once done this i have been able to print from SD onboard with no issues!

dkt0591, 2022-07-26:
Wow, and til now, they didnt fix this bug :))))) good job BigTreeTech :)))))

[6 more replies]