bigtreetech/BIGTREETECH-TouchScreenFirmware #1651: [BUG] No notification when new binaries are available

adelyser, 2021-02-25:
### Description

There is no way to track when new binaries are available or what changes are included in the new binaries. 

### Steps to reproduce

Just watch the repo.

**Expected behavior**

New binary builds would be tagged as a release with an accompanied change list so people can know what's included. People can also watch for when new releases are available using the GitHub notifications.

**Actual behavior**

Nothing.

### Hardware Variant
N/A
### TFT Firmware Version & Main Board Firmware details
N/A
### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2021-02-25:
Is this a bug or an offer to help and provide what you mean it should be done?

oldman4U, 2021-02-28:
Seriously, what would you expect? A letter from the CEO? There is a notification available which you have to turn on and if this does not meet your expectations you should contact the GitHub makers.

What changed you can see easily when you click on "Code". If this does not meet your expectations and you would like a plain textfile I hope that you understand that everything done here is made by users like you and me. So finally you tell others what to do what you could also do.

Please think about it again

adelyser, 2021-03-21:
Sorry, been away for a bit, but seriously, a letter from a CEO? No, be a little more realistic. I'm not asking for a plaintext file of all the changes either, but you do take the time to update https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1371.

It's really rather simple. When the compiled binaries are committed, update the version and tag it as a release. That makes it easy for people to track using the Github notifications. Github doesn't allow you to track individual file commits, which is what those are. Maybe this is a question oriented more towards BTT repo owners?

I'm surprised that BTT doesn't want to track the versions it provides, for troubleshooting issues. How can you help someone that is running version 3.0.26.x, when there is almost a year's worth of binary commits, all with the same version?

The history of the binaries is difficult to follow. I don't know what changes are included in the new ones to determine if I should update, they all state "Updated binaries and config". If there's another manner you are using to track the binaries and what's included, how about a bit of guidance as to where that might be? The readme on the main page has a link for the current version history. This takes you to the releases page, where the version history is extremely outdated, and is anything but current.

kisslorand, 2021-03-21:
If you go to the repo's main page there you have a button "commits". Just check out what it does. ;)
It's not what you've asked for but I hope it helps.

adelyser, 2021-03-21:
Thanks, I guess that sort of works. I have to look at the date my firmware was compiled to see where it relates to the newer versions, since the version doesn't actually change.

oldman4U, 2021-03-21:
Sad enough the version history is up to date;-( but this will change very soon. 

Great you found the notification.

If you think you can help -  please do it. Every helping hand is very appreciated.

kisslorand, 2021-03-21:
> Thanks, I guess that sort of works. I have to look at the date my firmware was compiled to see where it relates to the newer versions, since the version doesn't actually change.

Same goes for versioning, you have to check yours to see if the actual one is newer or not.

closed 2021-03-21 by adelyser