bigtreetech/BIGTREETECH-TouchScreenFirmware #1657: [BUG] (short description)

DaStivi, 2021-03-02:
### Description

If TFT is set to Marlin Mode (LCD Emulator/Simulator) and printer is reseted or power cycled, marlin does not boot, as soon as TFT is switched to touch mode marlin does boot up correctly!

### Steps to reproduce

1. switch do Marlin-LCD Emulator Mode
2. Reset/Power-Cycle

**Expected behavior**
Board/Marlin should boot up normal and just connect in Marlin/LCD-Emulator Mode (subsequently OctoPrint is able to connect)

**Actual behavior**
Board does not boot

### Hardware Variant
B1-TFT35 
### TFT Firmware Version & Main Board Firmware details

Marlin: 2.0.7.2
TFT: prebuilt from 24.2.2021 (d3453505ab3fae7e758d3adbbb402e78b8b66a12)

gordo3di, 2021-03-02:
Try turning the knob after it boots. My screen is dark unless I turn the knob so it looks like it didn't boot.

DaStivi, 2021-03-02:
> Try turning the knob after it boots. My screen is dark unless I turn the knob so it looks like it didn't boot.

nope, thanks for the hint.. but also don't work or change anything on this issue....
i've made ah short video to show you... but nothing special too see marlin just not initializing when TFT boots in LCD-Mode...

https://1drv.ms/v/s!AiIZqPOtqD4XqNBYvE9c6s-JwWAtrg?e=Gn3zkB

sarvenn, 2021-03-03:
I have the same issue here. When I switch to marlin mode and then shut down and power on my custom bootscreen stays on the screen. When I push and wait the encoder mode switch page does not appear. I tried to click and turn the knob from my memory to do a homing. It does. The only way to switch to tft mode is to copy my config file to an sd card and reboot my printer. Then it opens wirh tft mode.
[image]

sarvenn, 2021-03-03:
I have tft35 e3 v2 26 feb firmware. Now I will install latest firmware 3 March.

sarvenn, 2021-03-03:
I also did a test with my stock ender 3 display. Everything is perfect. So I'm sure that it is 26 feb firmware related.
Another issue is that, I opened my printer then switched to marlin mode and started a print from onboard sd of my skr mini e3, screen switched to tft mode.
I also had printer killed halted issues when I started a print using my tft sd card slot.

sarvenn, 2021-03-03:
> Try turning the knob after it boots. My screen is dark unless I turn the knob so it looks like it didn't boot.

That doesn't work my friend. It happens sometime when I switch to marlin mode from tft mode, but this is something really different as I explained my experience above.

oldman4U, 2021-03-03:
Try to start print with serial always on turned off.

DaStivi, 2021-03-03:
> Another issue is that, I opened my printer then switched to marlin mode and started a print from onboard sd of my skr mini e3, screen switched to tft mode.
> I also had printer killed halted issues when I started a print using my tft sd card slot.

yeah this is realted to serial always on... as @oldman4U mentinoed ... runinto this issue also before... 

> Try to start print with serial always on turned off.

@oldman4U i've doubled checked it just to be sure... serial always on is disabled on my side.. so the board not booting in marlin mode shouldn't b e an issue from this other issue..

oldman4U, 2021-03-03:
I mentioned this somewhere else, but as far as I remember the problem with Marlin not starting has been there before and has been fixed app 2-3 months ago. So it would be important to find a working version and also which version broke it. Then we can ask the developer to fix it. Maybe I find some time tomorrow to check it.

DaStivi, 2021-03-03:
some other guys in the biqu b1 Facebook group use ah firmware from @looxonline ... thats from 3.12.2020 (compiled, so not sure what base branche clone he's using and if he's changed other things in his branches too...) but there it's still working

oldman4U, 2021-03-03:
Maybe you can ask him or try a firmware from this repo from beginning of December?

oldman4U, 2021-03-04:
OK, I just tested the startup issue with my TFT35 E3 v3 connected to an SKR E3 DIP Marlin Feb. 11th and it works fine. Sure there was no problem during the update of the firmware and have you uploaded firmware, config.ini and TFT folder using a 8GB or smaller, FAT32 formatted sd card?

sarvenn, 2021-03-04:
> OK, I just tested the startup issue with my TFT35 E3 v3 connected to an SKR E3 DIP Marlin Feb. 11th and it works fine. Sure there was no problem during the update of the firmware and have you uploaded firmware, config.ini and TFT folder using a 8GB or smaller, FAT32 formatted sd card?

Hi @oldman4U , could you try with tft35 e3 v2 26 feb firmware? Switch to marlin mode and shut down your printer and re-open it. Before 26 Feb firmware there wasn't any problem. I haven't tried last two commits after 26 of Feb yet and will not be able to try for a week at least.

oldman4U, 2021-03-04:
Hi. I am just downloading the build from today 4.3.2021 and let you know asap.

sarvenn, 2021-03-04:
Could you also try 26 february firmware? I suppose that was guru's update. As I up to remember my serial always on was disabled as well.

oldman4U, 2021-03-04:
So I tested with the build from today with serial always on DEactivated and there was no problem to start in Marlin mode and also no problem to start a print using Marlin - it stayed at Marlin.

Then I activated serial always on, went back to Marlin and restarted the printer - it worked. So far so good.

Next I tried to print from on board sd card using Marlin and this is where the nightmare began. The TFT switched to the touchscreen where only the heaters got an update, everything else was frozen. The print started but when I tried to stop it it went to the parking position resume was not possible. I tried to press some other buttons and also the encoder and sometimes there was a kind of reaction but not really something I would call a normal operation. So I turned the printer off and when I turned it on again it booted in Marlin mode (even the printer was in touch mode when I turned it off) showing only the Marlin Mode Title. NO Marlin menu and also no way to go back to the Touchscreen using the encoder because it does not show the selection. The only way out of this is to reinstall the firmware.

The good thing is, that this is exactly the behaviour we saw a few months ago and I am 99.999% sure that we fixed it.

sarvenn, 2021-03-04:
You may also just copy your config.ini and boot your printer. If the default mode is tft mode, printer switches to tft mode and boots successfully. No need to flash the firmware.
I doubt that something went wrong at #1633 .

sarvenn, 2021-03-04:
I also had printer killed halted issues while printing from tft sd. I don't know if it is related with the tft. I may try with my ender3 stock display with the same gcode. But a week later at least.

@guruathwal hello. Could you have a look to this ticket?

oldman4U, 2021-03-04:
I haven't tested it but look here Jan. 4th: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1433

Hey it is me, your olddatabase;-)

sarvenn, 2021-03-04:
Relevant?

sarvenn, 2021-03-04:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1633/commits/cc30818bbfa2d59e91944445d1f8593a34a551bc

DaStivi, 2021-03-04:
whats gooing on here... :) 

strange with with this serial_alwayson "bug"... i even did the reset.txt on each flash to newer version to just rule out anything else, and i'm almost sure i didn't enabled serial always on since them, and also never tried to print from marlin mode then...

BUT, i got another freeze today from the TFT while i was printing from octoprint right now! so printer is still working, skr1.4/marlin is working normal, TFT jost froze and does not react to anything :/  really strange because this rules out almost everthing i thought could be the issue, like havein bad USB/SD ... the only thing still was gooing on, was that that marlin sent M114 to got some status update on the TFT as its not recognizing the print from octoprint host.. so either its ah hardware fault, temp, issue, but i doubt that, because i had again ah couple of successfully prints (even with octoprint) with no tft freeze ...

oldman4U, 2021-03-04:
Just tested https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware.git which is the last build on Feb. 24th before #1633 was released and it switched also to Touchscreen when starting a print from Marlin mode, but this time the Touchscreen updated regularly and after stopping and restart it restarted in Touchmode.

Last test for today is #1468 from Jan. 7th https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/c367983689afc3cd4fe1f7b230f8a02f32b6b47d

oldman4U, 2021-03-04:
So using  https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/c367983689afc3cd4fe1f7b230f8a02f32b6b47d
I can start in Marlin Mode and also print using Marlin Mode without the TFT switching to Touchscreenmode once the print is confirmed all with serial always on activated. Now we have to find the PR which broke it.

guruathwal, 2021-03-05:
The boot issue is related to LPC based mainboard. There is some issue with these mainboards which is preventing booting if the state of serial pins on TFT controllers is changed while the mainboards is booting. I had already found the boot issue in marlin mode while working on #1633 and tried to fix it there, but it looks like the fix is causing other issues.
To be able to solve the issue it is important to know the details of both Mainboard hardware and software, which no one has provided till now. Please provide the details so I can further look into it.

sarvenn, 2021-03-05:
@guruathwal 
skr mini e3 v2 ==> Marling bugfix 2.0 Commits on Feb 28, 2021 (custom bootscreen is enabled as you can see above, freezes on that screen while last state was marlin mode before switching off).
tft 35 e3 v3 ==> 26 Feb commits

By the way in config.ini file default mode is set to touch mode. Shouldn't it restart in tft mode regardless of the last state of the screen mode? I expect to always open in tft mode when I choose this setting to touch mode as default.

#### Default Mode
#   Options: [Marlin Mode: 0, Touch Mode: 1]
default_mode:1

oldman4U, 2021-03-05:
sarvenn

This option is only for the first time it is turned on but the TFT (usually) remembers the mode after that.

oldman4U, 2021-03-05:
Have you found the PR which broke it?

sarvenn, 2021-03-05:
> Have you found the PR which broke it?

I'm away from home for drive tests etc. So I'm unable to try but @DaStivi has 24 feb firmware, I had 26 feb firmware. Something must be in between. 

My "speaker" is enabled at marlin mode. Also powerloss recovery was enabled at marlin and tft. Any realation possibility?

oldman4U, 2021-03-05:
You better drive faster...;-)
[image]

DaStivi, 2021-03-05:
i've got an SKR 1.4 LPC1768 (marlin 2.0.7.2)

and if i understood oldman4U's findings correctly the issue started somewhere after 7.1. commits...

oldman4U, 2021-03-05:
This
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/c367983689afc3cd4fe1f7b230f8a02f32b6b47d
is the PR which solves the problem you describe here and the problem that starting a print in Marlin Mode makes the TFT switch to the Touchscreen. I have both issues with the TFT firmware from yesterday using the same Marlin on a SKR E3 DIP (STM32).

DaStivi, 2021-03-05:
i looked into the files.. and one thing i found regarding serial (have to admit didn't through all files) was this here.. what have been changed in this PR #1468 

[image]

and in #1633 the same file changed slightly again and moved on function into an if function:
(but there are ah lot other changes too... and also i've to say i am no where near a developer... i can read the code ah little but have absolutly no knowing on how all that add up together, sorry.)

[image]

guruathwal, 2021-03-05:
@DaStivi `Serial_ReSourceInit()` was moved inside `(mode == MODE_SERIAL_TSC)` because it is preventing some LPC based boards from booting if the TFT is booted in marlin mode.

DaStivi, 2021-03-05:
> @DaStivi `Serial_ReSourceInit()` was moved inside `(mode == MODE_SERIAL_TSC)` because it is preventing some LPC based boards from booting if the TFT is booted in marlin mode.

yeah and thats excactly my issue i've with the the latest builds again now! :)

guruathwal, 2021-03-05:
@DaStivi ok, which is the latest one of these commits that was working?
[image]

guruathwal, 2021-03-05:
> > @DaStivi `Serial_ReSourceInit()` was moved inside `(mode == MODE_SERIAL_TSC)` because it is preventing some LPC based boards from booting if the TFT is booted in marlin mode.
> 
> yeah and thats excactly my issue i've with the the latest builds again now! :) so even if its (Serial_ReSourceInit();) in the if-loop again the board don't boot.. and if i understood the timeline now correctly, it worked before the code was inside the if-loop?!

When the line was moved outside the loop I did not test by booting in marlin mode.  With SKR 1.4 it did not boot before #1633

oldman4U, 2021-03-05:
I sent Gurmeet the # of the PRs related to the two issues reported with this ticket. Hope he will fix them soon.

Thank you

DaStivi, 2021-03-05:
i've got to test by my own now... so... with the firmware before #1633 from 26.2. booting in marlin mode didn't worked!
with this commit/pr now, it indeed boot in marlin mode and even i can start the print from onboardSD in marlin mode.

all was with serial_always_on = disabled
after switching to serial_always_on = enabled
switch to marlin lcd mode, and reset, it won't "boot" anymore... as i do have neopixel LEDs, i can tell you that marlin at least initialize, as i see the LED Test sequence, this didn't worked with the commits before #1633 there i didn't even got this led init

but at this state the screen is frozen! no switching back to touch mode possible... i did also ah test, no.. keep the rotary encoder pressed and then reseted... i got the mode-select screen and could switch but wasn't fast enough to confirm, as the screen freezed again!

and as i've connected Octoprint, i also see that in octoprint the board is connected and i can work with it! with an ESP01s conected onto the TFT, i got no connection to Marlin!

noticed something other too, when in serial_always_on mode enabled, you can switch to marlin mode, while printer is initialzied.. but then assoon as you reset its broken..
while in serial_always_on and marlin mode, and trying to print, the issues happens that tft switches back to touch mode!

ive to admit, in the past where i noticed this when-print-in-marlin-back-to-touch-mode thing happend, i'd might never ever booted from marlin, just switches everytime from touch to marlin without reseting...

so one thing looks like to be fixed, what was initially the issue in here,.... and basically all whats still left is the, when serial_always_on breaks marlin-lcd mode (and it might just be ah wrong turn, that with the special case where you just switch the modes while in operation and than starting ah print, the tft still believes half way to be in touch mode...)

i've also made ah video, but its still uploading..

oldman4U, 2021-03-07:
Please check Gurmeet's #1680 as it should fix the reported issues.

🙏

PR #1680 (2021-03-07): Bugfix: boot issue in marlin mode and prevent mode switching while printing, wrong keywords in Popup dialog
- Fix freeze/hang/boot issue if booting in marlin mode with serial_alway_on active. (#1657)
- Fix mode switching issue if printing from onboard SD with Marlin Mode active due to changing menu index.
- Fix wrong keywords in Popup dialog. (#1676)

Resolves: #1657, Resolves: #1676
closed 2021-03-07 by bigtreetech