bigtreetech/BIGTREETECH-TouchScreenFirmware #1659: A few minor bugs

AntoszHUN, 2021-03-02:
I have attached a couple of (picture) glaring bugs. These errors are quite noticeable on the TFT70 display.

[image]
Does not show the height of the first layer when you start printing. Only update when you have finished 1-2 layers.

[image]
Caption location is an obvious error.

Made By: AntoszHUN

oldman4U, 2021-03-02:
Maybe you want to add TFT70 to the title.

AntoszHUN, 2021-03-02:
> 
> 
> Maybe you want to add TFT70 to the title.

Ohh sorry.

AntoszHUN, 2021-03-03:
During printing, there are moments when the subject cooler does not turn on or change gear.

closed 2021-03-04 by bigtreetech