bigtreetech/BIGTREETECH-TouchScreenFirmware #1660: [Q] Toxic behavior of users here. -- RE: Do you want new developers here or not?

Domush, 2021-03-03:
**Toxic behavior of users here**
Admins here should really consider whether or not they want a toxic person like @oldman4U chasing away developers. Looking at the pull requests it looks like you're down to only a couple of contributors remaining.

I've spent the last year completely rewriting the BTT TFT codebase to be more developer-friendly, stable, far easier to read (all variables, functions, and files have been refactored to be intuitive), documented, and modular, and came here to begin merging that code into the main branch. I use these touchscreens in my CNC controllers, and so have engineered the code to be adaptable to both 3d printing and CNC use.

Roughly 70K lines of code have been completely redesigned in my fork, including config handling, far better error handling, modular gCode processing, UI upgrades, submenu breadcrumbs, pause management, tool change features, and a professionally designed UI icon set. And while I'm happy to begin merging this code, I'm not going to waste my time if toxic buffoons like @oldman4U are going to cyber-stalk and harass incoming contributors. I'll simply continue modifying my private fork and the dozens of feature requests here that I've already enacted in my rewrite can remain open here forever.

Your choice. 

Keep the toxic users or create an inviting development hub where contributors feel valued instead of insulted and attacked upon entry.

oldman4U, 2021-03-03:
It is about me and therefore I will not be part of this ticket. Just to add where this all started: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1170

filip-smid, 2021-03-03:
@Domush I suppose you are free to show the rest of GitHub your 70k lines of code you did on this repo...or not... Up to you... But given the number of contributions between you and @oldman4U ... Well.. good luck..

Portzal, 2021-03-03:
@Domush 
I occasionally drop back in to see if things have changed, seems not. Dysfunctional is the first thing that pops into my head every time I come back for a look.

bigtreetech, 2021-03-03:
I'm sorry that this kind of thing happened. It's my responsibility. I should learn how to manage open source projects on GitHub.

And oldman4U is actually a very enthusiastic user. He helped test a lot of PR and wrote a lot of readme documentation. He put  all the bugs together and all the FR together to make it easier for developers to fix and implement. the intention is good. But  I didn't fix and implement it in time. It's my fault.

We plan to test and fix existing bugs in the next two weeks, and then release a new stable version.  Maybe the new bugs and FR tracing methods will be used in the next development branch ? These can be discussed,

All in all, We all hope that this project can develop better and more actively. The root of the problem lies in me. I should learn how to manage GitHub projects.
I apologize to all users who supports BTT, hope we can have a patient discussion. Thank you all！

sarvenn, 2021-03-03:
@Domush I hope you contribute and help users like me to have a better experience with my btt products. For my short experience here @oldman4U helped me a lot and show the right way to solve my issues. I really wish you to see your improvements here. I don't understand from coding but I'm an early adapter and user for tech stuffs. Also I have UX/UI/HMI knowledge from my daily job at Ford. So I would be happy to try your PRs and give you my feedbacks. No need to punish the users like me, right? We all can talk and solve our problems.

Domush, 2021-03-03:
I've made my fork public for those who want an early preview of the eventual pull requests.

https://github.com/Domush/CNC-firmware-for-TFT35-Touchscreens

It's currently CNC specific (as I don't use it for 3d printing), but a vast majority of the changes are universal (I designed it that way).

I originally made this private because it's for personal development and I often submit changes that don't compile or have known issues. Feel free to adopt code changes (most code changes have accompanied notes), but expect the entire fork to be alpha.

I'm currently deep in serial gCode response handling, as the current factory implementation is hard coded and resembles a bunch of hacks rather than a designed implementation.

readeral, 2021-03-04:
Thanks @Domush this is excellent

readeral, 2021-03-04:
> I'm sorry that this kind of thing happened. It's my responsibility. I should learn how to manage open source projects on GitHub.
>
> And oldman4U is actually a very enthusiastic user. He helped test a [...]

It would be a shame to continue to work against the way that GitHub is designed - Domush and I both have recommended elsewhere that issues remain open so that the open-source community can interact with them in the way that the majority of other repos work.

oldman4u clearly has a lot of energy and wants to participate - but because of the free-reign he's been given, it's largely scaring people away - I would recommend that he divert his energy to the questions asked under Discussions, given he is self-confessed not a programmer, it would be better if his management of this library was revoked to enable it to operate correctly.

It is disappointing to see testimony from a number of developers across a number of issues/posts that feel this repo is toxic . I've felt no desire to actually work on the source and fix my issue with its Klipper compatibility as I have no confidence the code would be reviewed, good feedback would be given to allow me to iterate, and any successful work would be merged. Without developers here to encourage others, there's very little point contributing.

bigtreetech, 2021-03-04:
> 
> It would be a shame to continue to work against the way that GitHub is designed - Domush and I both have recommended elsewhere that issues remain open so that the open-source community can interact with them in the way that the majority of other repos work.

So we should keep unresolved issues in the state of being opened.  but we need to fix the bug in time to avoid opening too many issues (Maybe oldman4U can collect bugs into one link for developers can easily grasp what problems have not been solved, but he shouldn't suggest close issues that aren't fixed?)

> oldman4u clearly has a lot of energy and wants to participate - but because of the free-reign he's been given, it's largely scaring people away - I would recommend that he divert his energy to the [...]

In fact, the account of oldman4U have not the management of this project. As I and other users have said, he is a very enthusiastic user and has helped a lot of people in this community. The root cause of this conflict is the management of issues, That's what we discussed above, I believe we can reach a consensus. 
I apologize to you If you feel offended.  The users(Including you and oldman4U) in this community are all out of love for open source, so we can't blame anyone.  We all hope that this project can be better and more active.  And I apologize again for making you unhappy.

guruathwal, 2021-03-04:
As everyone is stating the facts here is one: It was @oldman4U who helped me test the firmware when no one was bothering to check the issues in this repo. It is because of him that so many issues are resolved which no one else cared to look at. There will always be disagreements at places where people get together from different parts of the world. It is just a matter of one's perception of what is right and what is wrong. So much unnecessary fight but no one has tried to understand why some things are being done in a way they should not be. I agree that it is not the right way to work on GitHub but given the circumstances, some workaround is needed until there is propper management done by @bigtreetech. 
Be careful while using harsh words for anyone because once written they are permanently stamped on the public image on both sides and there is no coming back from that. 
I disagree with the claim that people are being scared away. What I see is that in comparison to Git repo for other similar Touch screen controllers, this is the most actively developed repo which tells the story otherwise. 
I would suggest @bigtreetech to take at least 30 min every other day to sort the new/pending issues and keep the repo clean. The root cause for this conflict is that most of the issues are either left abandoned due to a lack of response from the persons opening the issues or are unrelated to the Firmware.

oldman4U, 2021-03-04:
Seems this discussion takes another direction, so I allow myself to add my thoughts.

@reader and Domush

I tried to explain to both of you, why we introduced this structure but you did not understand. So I want to try it again, using a different approach and hope this will allow you to see the things different afterwards.

I WOULD AGREE WITH YOU, that the way this repository is used is against the way GitHub is designed if:

1. we would have used other methods than those which are available using GitHub
2. the owner of this repository would actively maintain it
3. it would not be used like a Facebook or similar user group.

About me. I am 52 years old, working for the same company in the production and broadcast industry for more than 25 years now and more than half of this time as a production manager. Not sure when GitHub was available for the first time, be we use it since approximately 10 years together with various Atlassian tools and a HelpDesk system for developing and maintaining software for the broadcast industry. Most likely we use GitHub the way you feel it is the only right way to use it, but this would be completely impossible here and there is not a single right way to use it but different ways. None of those ways is right or wrong but good or bad, depending on the needs and the possibilities of the given project.

When I started here as a newbie 1.5 years ago, I dived in slowly and learned a lot by answering every ticket I believed I could help. Sometimes I was right, sometimes I was wrong.

But what catches my eye immediately was the number of open tickets and especially the number of tickets with NO answer at all. I always take this repository as an example, not only because it is from the same manufacturer, but because it shows very nicely how this repository would look like without the system we introduced: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues

Looking at the tickets in this repository you will find many tickets labeled as a Bug. Check them and see by yourself that almost all of them are not bugs but user errors or, that several tickets are about the same issue. Using this GitHub "the right way" this would not be possible, because the owner would take care that user errors are not handled here and that duplicates are removed. Introducing the "system" here was a lot of work and is ongoing work but under the given circumstances it was the only way to keep it usable. The better is the enemy of the good and if you know a better solution you are free to introduce it and to take care that it works.

Personally I am able and willing to work with others, does not matter if they are like a friend to me or if I do not like them, as long as it helps the project. What I do not like are users contributing here for the first time and all the have to add to this project is to tell what a nightmare this is and how it has to be done right. Do it, but do not tell others what to do and also not what is right and wrong. Otherwise you end up as a useless user for this project, something especially you readeral are at the moment. Think about how you can help, contribute but do next expect that others will pave the way for you, so that you readeral can work here the way you expect it. That you can not contribute here because of oldman4u sounds like an excuse to me, even more because there are many other BTT repositories where I am not active, but also there I can not see you contributing.

Finally, it is all about the success of this project and how the firmware can be improved. In case someone has an idea how a certain part can be improved - it should be done - even it is probably not the same person. What we need in such a case is someone who decides or, that the majority of the users contributing to this repo, agrees. This includes not exclusively but especially the part how incoming tickets are handled as they are the base of everything else.

As always my2 cents

oldman

Domush, 2021-03-04:
_”Looking at the tickets in this repository you will find many tickets labeled as a Bug. Check them and see by yourself that almost all of them are not bugs but user errors or, that several tickets are about the same issue. Using this GitHub "the right way" this would not be possible, because the owner would take care that user errors are not handled here and that duplicates are removed. ”_

The solution to this is **not** closing every issue whether or not it's been resolved. All that does is ensure the issue search feature, which developers rely on and is a core feature of GitHub, becomes useless.

If issues can be closed, then they can be marked duplicates in proper fashion and relabeled as user errors in proper fashion. The same repository rights handle both cases.

The fact the SKR 1.3 repository contains many duplicates is how developers prioritize and verify bugs are truly bugs, you would know this if you had any development experience.

When a hospital is full of patients, you don't simply marked them all as cured and create a side list. If the medical world was run as backwards as this repository then a pandemic would never be detected because only one single case of COVID-19 would be listed.

Triage is as important in development as it is in the medical community, and destroying all of the data by prematurely closing issues destroys any hope of proper triage. This can't be spelled out any chase clearer.

It's great you took the initiative when nobody else was seemingly stepping up (although I have no way to verify you didn't chase those developers away, as well, through personal insults and accusations of being ”fake” the second they joined like you did with me), but in doing so without GitHub experience you actually reduce the future success of this repository, not improve it.

If someone did to the mozilla repository what you did to this one Firefox development would grind to a halt. It's not ”better as the enemy of the good”, it's **toxicity, personal insults, and improper actions is the enemy of success.**

You want to help without development or GitHub experience, @oldman4U ? Great! Answer user error questions. Link duplicates. Perform 1st round testing of pull-requests. All of these things can be performed by a non-coder.

Leave the development decisions and tracking to the developers, **even if there aren't any**, because sooner or later there will be some, and they need to be able to use the GitHub tools available to them without getting sabotaged, and in your case, chased away by toxicity.

guruathwal, 2021-03-04:
@sarvenn are you working on Sync 3?

oldman4U, 2021-03-04:
@Domush Got it, thank you.

oldman

sarvenn, 2021-03-04:
> @sarvenn are you working on Sync 3?

No @guruathwal . I mostly involve in heavy commercial trucks which does not use sync but something similar to that, in house developed. Cluster stuff etc. Mostly function/feature developments.

For the topic, gentlemen please don't use words that may hurt others. We may criticize each other of course but please be polite and respectful. Otherwise discussions will not end. This will definately harm basic users like me waiting for new features, bug free firmwares, improvements. etc. I kindly request it from all of you.
Stay safe and healthy.

Portzal, 2021-03-04:
From my experience in this repository, I actually caution people about even raising bug reports or issues here, I found it to be patrolled by a few bully's that try to squash or shout down anything they do no agree with or understand. The constant pestering to close tickets is also very strange behavior. I would very much like to be active in this community, but it seems to have evolved into someone's empire building exercise as opposed to an open and inviting community.

rightawy, 2021-03-05:
> Roughly 70K lines of code have been completely redesigned in my fork, including config handling, far better error handling, modular gCode processing, UI upgrades, submenu breadcrumbs, pause management, tool change features, and a professionally designed UI icon set. 

What my dear friend @Domush is claiming here seems to be a false claim and more like intimidation. I believe the 70k redesigned lines in https://github.com/Domush/CNC-firmware-for-TFT35-Touchscreens he is claiming is about the removal of all the code related to 3d printing, all languages and by adding/removing whitespace. 36k "redesigned" lines can be seen in a single commit here https://github.com/Domush/CNC-firmware-for-TFT35-Touchscreens/commit/a09da2e7c8b3dd6652140fe58b41a62250e29367 which just involves only indentation and whitespace changes and other small commits like https://github.com/Domush/CNC-firmware-for-TFT35-Touchscreens/commit/1f6aa4aca2f5bcd39f2e32e16282358e6a81fdb6 (3k) & https://github.com/Domush/CNC-firmware-for-TFT35-Touchscreens/commit/3db3a84abbd433efe438668e8a7137af593a43c7?branch=3db3a84abbd433efe438668e8a7137af593a43c7 (7k)

Below is the comparison of his fork showing just 45K "redesigned" lines. The last sync on Domush's fork was 1 year ago and a lot has changed on this repository. [image]

>,
> and a professionally designed UI icon set. 
>

**Why do you call these professionally designed?**
There is not even a single bit of consistency between the icons of the same category.
[image]

> I've made my fork public for those who want an early preview of the eventual pull requests.
>
> https://github.com/Domush/CNC-firmware-for-TFT35-Touchscreens
>
> It's currently CNC specific (as I don't [...]

Ok, a piece of small advice here before writing a lie you should do your homework. another small advice is to never lie.
Your fork was never private. How I know this is because a few months ago I was looking for CNC support on BTT's controllers and at that time I saw your fork.
Also, Github does not allow to change the visibility of a fork of a public repository. Can you see the irony here?
[image]

[image]

``` 

```

@readeral Here is the highlights of your first comments in this repository (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1170#issuecomment-741652314). They are very professional and non-toxic.
[image]

``` 

```
@Portzal Can you elaborate on your experience? All I could find about your "experience" in this repository apart from this nonsense crap is only two issues.
One of which (#1179) was opened and **abandoned** by you and never bothered to reply to the person (@radek8) who was trying to help you. There is no involvement of @oldman4U in that issue.
The second "experience" is in #1186. Where you wrote only one comment https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1186#issuecomment-713500473 about the people involved in that issue and called them "clowns"
[image]

[image]

@bigtreetech  Please close this nonsense. The only thing these guys are doing is littering this repository and spreading rumors.

bigtreetech, 2021-03-05:
I hope our goal is to help this project better. I agree with my criticism, because BTT has not kept a continuous follow-up on this project. We have contacted active members about this project and created a contributor team with the intention that this project can be active and better.
  After tell us what we did wrong. We will listen carefully, if there is improvement, it will be better. And words that vent anger are useless for improving the project. So we need to communicate patiently and continuously.
  I just want to say we're keep going, And I hope all of us can make this project progress together.

  Finally, This topic seems to have strayed. Not "Discuss the management of issues ", but there is a tendency to attack each other, so I will close this topic for the time being. But It not mean that we are silent. Welcome to continue to publish here about "Discuss the management of issues ". I'd like to apologize to the injured again. I hope everyone can calm down and communicate patiently.  Because we are all contributors to open source, we are like-minded people, aren't we?

closed 2021-03-05 by bigtreetech

oldman4U, 2021-03-05:
Thank you rightawy for clarification and all the time you spend on this.

Sincerely

oldclownmaybe

Domush, 2021-03-05:
The fact that @rightawy actually took more time to attempt to portray 70k lines of actual written code as merely whitespace changes, instead of actually reviewing the code, shows just how toxic this repository has become.

And @bigtreetech ignoring it shows that nothing will ever be done about it.

Enjoy your toxic fiefdom.

I tried, I gave @bigtreetech a chance to step up and do something about it. They seem to enjoy their dead project remaining dead.

I'm out.

bigtreetech, 2021-03-05:
My first reply under this topic is about the issue that you said before.   the issue should not be closed  before it be fixed, I agree with you and indicate that it will follow this way in the future. I also shows that I am willing to learn the management of GitHub open source projects. accept your positive suggestions.
The fact is that the people under this topic are only looking for each other's fault, not a positive attitude. 
Since we have reached an agreement on the issue management, why are we still entangled here.
If you're willing to give me a chance, we should move on to the next one

> 
> And @bigtreetech ignoring it shows that nothing will ever be done about it.
> 
“ This topic seems to have strayed. Not "Discuss the management of issues ", but there is a tendency to attack each other, so I will close this topic ” This is my reply before closing this issue. How can I ignore it？

If you're willing to give me a chance, Let's stop the attack each other here please.

Domush, 2021-03-05:
> My first reply under this topic is about the issue that you said before. the issue should not be closed before it be fixed, I agree with you and indicate that it will follow this way in the future. [...]

You're right, @bigtreetech , it's about the issue I was discussing _before_, in a **different** topic, not this one. In fact, you went so far to ignore the main topic of this thread that you _literally changed the topic title itself_ away from the topic I brought up.

**Toxic behavior of users here** was the original topic, and it has still not been addressed by you. Clearly "Let's stop the attack each other here please" is not working as a behavioral modification strategy on the toxic regulars here.

Unless toxicity here is dealt with, in a manner better than literally changing the topic to something else, then I can assure you this project will not attract professional developers.

> The fact is that the people under this topic are only looking for each other's fault, not a positive attitude.

The "not a positive attitude" is literally the reason for this entire issue thread. I can see how that would be confusing given the fact someone (_you_) changed the title of the thread.

> Since we have reached an agreement on the issue management, why are we still entangled here.

Because that was never the reason for this specific thread, toxicity and attacks on incoming people was, and still is. I don't know how to make this any more clear.

Until real steps to clear out the toxic people takes place, this repository will be a ghost town that professional developers avoid. It's been spelled out by multiple developers already. And that's a shame, because I like BigTreeTech components, and these TFTs could have a much better, well-documented, modular, easy to approach and modify codebase than they currently have. 

That said, I guarantee none of the toxic regulars in this thread, or any of the other threads, are going to make it that way. Quite the opposite, their toxic behavior will ensure it never becomes a mature, professional codebase. How do I know this? Because no professional bothers with toxicity, because we have better things to do.

I've attempted to address this problem from every direction I know how, so this is my last post here. I have paid work I'd rather be doing.

Best of luck.

bigtreetech, 2021-03-06:
I'm sorry that this happened. I just changed the title of this topic after we reached an agreement on the initial issue. It's true that I've ignored the  previous communication has led you to be very disappointed with the communication environment here.  So I didn't get to the point of this topic before.  Sorry again for that. Your point in this topic is that we need a good environment for this project. 

I thought before that the communication environment will automatically become friendly in the future  when we reach an agreement.  the "toxicity" you said won't happen again here. So I changed the title here. 

Anyway, I beg you don't give up on me now. At least you can observe this project for  a period of time. If one day you find that our environment has improved, you are welcome to contribute for this project. Thank you in advance!