bigtreetech/BIGTREETECH-TouchScreenFirmware #1661: [BUG] (short description)

burakturkoglu54, 2021-03-03:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

burakturkoglu54, 2021-03-03:
TFT70

oldman4U, 2021-03-03:
Looks like an SOS...

How can we help you!?

gordo3di, 2021-03-03:
> Looks like an SOS...
> 
> How can we help you!?

Haha. Must have been a rough night for them.

sarvenn, 2021-03-03:
@burakturkoglu54 Merhaba Burak, problem açarken ayrıntılı bir text yazar mısın?

closed 2021-03-05 by bigtreetech