bigtreetech/BIGTREETECH-TouchScreenFirmware #1664: [BTT TFT35-E3 v3.0 power connector is capable to power RPI4B?] (Your Question?)

IdeYoda, 2021-03-03:
Hi, I have an Ender 3 pro with your great BTT SKR Mini V2.0 together with the TFT35-E3 V3.0. I also upgraded this config with a Raspberry PI4B. My basic mission was connect those two boards inside the house using the UART3 connector and meanwhile i discovered this power connector on the TFT panel. I'm wondering if this is can be used maybe to power up the RPI4 which is needs exactly 5V? Can the system give the 2A which is needed for RPI4? Is this connector is capable to serve that? Also, can the two boards communicate trough the UART3? What do i need to change in the setup maybe? Thanks anyway for all answers.

radek8, 2021-03-03:
The power supply for BTT SKR Mini V2.0 has a fuse on the 5V branch 2.5A
Therefore, I would not recommend taking 2A from it

[image]

kisslorand, 2021-03-03:
There are no power output connectors on the TFT, only power input.
You'll be better off using a cheap buck converter and take the power from the PSU.

oldman4U, 2021-03-03:
Hello IdeYoda.

Please keep in mind that this repository is primary for the TFT firmware and not a user group. Once your question has been answered it would be great if you could close this ticket.

Thank you

IdeYoda, 2021-03-03:
> Hello IdeYoda.
> 
> Please keep in mind that this repository is primary for the TFT firmware and not a user group. Once your question has been answered it would be great if you could close this ticket.
> 
> Thank you

Ok, sorry, i'm a newby and still learning the way... so closint this. Thanks for the answers!

closed 2021-03-03 by IdeYoda