bigtreetech/BIGTREETECH-TouchScreenFirmware #1670: [FR] Option to drop to a safe Z-height after Homing during bed leveling

lightmaster, 2021-03-05:
**Is your feature request related to a problem? Please describe.**
Unable bed level from touch screen when using a Delta printer, since it tries to move along X/Y before dropping down to where X/Y can move freely. This causes the belts to grind against the stepper gear repeated and afterwards the printer thinks that it is at a location that its not actually at.

**Describe the solution you'd like**
There needs to be an optional z-height to drop by before attempting to move laterally to a X/Y location.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

radek8, 2021-03-05:
Do you also have this problem in Marlin emulated mode?

lightmaster, 2021-03-05:
Tried to test that and stumbled upon #1657 bug. Just found that report and how to get the screen working again, so I'm gonna test it in a minute.

As a temp fix, I changed this line from https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/c367983689afc3cd4fe1f7b230f8a02f32b6b47d/TFT/src/User/Menu/MeshEditor.c#L893

to:
```c
mustStoreCmd("G28\nG1 Z150 F1500\n");                         // only the first time, home the printer
```

Not an eloquent solution by a long shot, but it does work to keep it from messing up the belts at least.

lightmaster, 2021-03-05:
In Marlin Mode, UBL Mesh Edit works great. It drops maybe 30 in height so it gets down to a safe area, then moves out to the point you selected

radek8, 2021-03-05:
Maybe @digant73 could help.
But I don't think it currently has a working display and I don't know if it has a Delta printer available to simulate the problem.

lightmaster, 2021-03-05:
> But I don't think it currently has a working display

Don't know if what has a working display? All of the display screens dealing with UBL seem to be working when I compiled the latest from the repo.

oldman4U, 2021-03-05:
digant has a broken display and is waiting for a replacement.

radek8, 2021-03-05:
digant73 is a programmer who created the Mash Editor

digant73, 2021-03-05:
As far as I understand the problem is when you move to mesh tuner and the nozzle is moved on XY coord.. The tuner uses G42. It is strange that the problem is not avoided by marlin. No I don't have a delta printer but I understood the problem.If an indication of delta printer is returned by marlin, the fix is easy. Otherwise a new configuration param is needed

lightmaster, 2021-03-05:
In Marlin, it looks like it issues a G1 command to drop to a lower Z first, and then a G42 to move to the mesh spot, then G1 again to get to the final height for the spot.

digant73, 2021-03-06:
@lightmaster comment it out only line 12 in MeshTuner.c such as:

// if (infoMachineSettings.zProbe == ENABLED)
    probeHeightStop();

so probeHeighStop() is always executed even if you don't have any probe (as it should be in your case).
Please let me know the result. Don't forget to rollback your changes in MeshEditor.c (restore "mustStoreCmd("G28\n");").

If I'm not wrong I added "if (infoMachineSettings.zProbe == ENABLED)" because other users had the opposite problem. I have to check in the history of my PRs

lightmaster, 2021-03-06:
> so probeHeighStop() is always executed (it raises the nozzle by the retract value configured in config.ini)

If that raises the nozzle, wouldn't I want it not to be executed? The problem is that it does a homing, which raises the hotend to its max height, and then it doesn't drop down any before moving to the XY coord.

Also, if I'm reading this code right, this is the code that is run every time you go to a new point, and I think this line is getting run each time, cuz it does jump up a little when I tell it to go to a new point.

digant73, 2021-03-06:
probeHeighStop() is not executed in your printer because you don't have a probe I suppose. If you comment the line
if (infoMachineSettings.zProbe == ENABLED)
then probeHeighStop() (it executes G1 Z...) is executed so you should not have the bug.
Yes, it will be executed for each point.

lightmaster, 2021-03-06:
Did not help.

[Hotend crashing into top of frame and then into bed](https://drive.google.com/file/d/1ytqReGYUiGMMHlO3ekQIlyB_EiudZByH/view?usp=sharing)
As you can see in the video, after clicking Mesh Edit, it homes and then tries to move laterally without first dropping straight down. This causes the belts on the towers to jump on the gear, potentially damaging the belts and making it the hot end is not where the printer thinks that it is. Then when it goes down to the bed, with the hotend lower than it thinks it really is, it rams into the bed.

By first dropping down by 30mm after homing, it can then move wherever it wants to laterally without the towers crashing into the top of the frame. ie:
```gcode
G28            ; Home
G91            ; Set relative positioning
G1 Z-30     ; Drop 30mm below current position
G90           ; Set absolute positioning
```

lightmaster, 2021-03-07:
Feature added and merged in PR #1677

closed 2021-03-07 by lightmaster

digant73, 2021-03-11:
@lightmaster please refer to PR #1716 making configurable the Z height applied after G28. It improves your PR and fixes the crash you verified on Probe Offset menu.