bigtreetech/BIGTREETECH-TouchScreenFirmware #1671: [FR] reopen #1179, add other tmc driver to the menu for current.

BlackDragon2020, 2021-03-05:
I found ticket number 1179, but could not comment.

Here is my comment to the ticket.

Going to see if I can reopen this. I for one am back to modding my rig. I have bought some motors that are different than stock and would benefit from being able to set current per driver. I set my z2 with the gcode.

If we are still space limited. Setting a few variable and if statement using preprocessing #define , #if etc.. could easily add N number of stepper drivers with very minimal extra space.

If this is something of interest and no one wants to take up I can see if I can go through the code and get back into the flow of this project, but someone who knows the tmc section and has a good grasp with c++ could bang this out in about 30min I bet. It would take me a few days of review I bet before I could start any coding.

guruathwal, 2021-03-06:
I am working on separating dual/triple stepper for all X Y Z and E axes which would allow setting allof the parameter variables individually for each motor. You have to wait a little bit more as currently some code cleanup and bug spray is required before proceeding further.

BlackDragon2020, 2021-03-06:
> I am working on separating dual/triple stepper for all X Y Z and E axes which would allow setting allof the parameter variables individually for each motor. You have to wait a little bit more as currently some code cleanup and bug spray is required before proceeding further.

You have been kicking @$$. Keep up the good work.

PR #1884 (2021-05-06): Clean-up parameter settings and display only axis/stepper supported by the machine
- Clean-up parameter settings.
- Display only axis/stepper supported by the machine.
- Fix some bug in parameter parsing in parseACK.c and interfaceCmd.c
- Workaround to display stealthchop settings if not reported by Marlin firmware when it is turned off.
- Separate persistent info code from parameter settings.

Build report:

Device | Old RAM Usage | New RAM usage | RAM Change | Old Flash Size | New Flash Size | Flash Size Reduced
-- | -- | -- | -- | -- | -- | --
TFT35 V1.1 | 17544 | 17596 | 52 | 148188 | 1476 [... cut]
closed 2021-05-06 by bigtreetech