bigtreetech/BIGTREETECH-TouchScreenFirmware #1673: [BUG] Printing from OctoPrint does not show up in touch mode

lightmaster, 2021-03-06:
### Description

When I initiate a print from OctoPrint, the touch mode does not show any printing information and doesn't seem to notice that a print has been initiated. I can still control babystepping from the touch screen and I can see the hotend and bed temps during a print, so clearly there is still communication between the screen and board.

### Steps to reproduce

1. Connect OctoPrint to main board
2. Initiate print over USB serial
3. Observe that the TFT does not show a printing screen

**Expected behavior**
When a print is started from any source, including TFT SD card, main board MicroSD card, and main board USB, the TFT touch screen should show the printing screen with various bits of information about the ongoing print.

**Actual behavior**
TFT touch screen fails to detect that a print has been initiated.

### Hardware Variant
Using TFT43 and SKR 1.4 Turbo
### TFT Firmware Version & Main Board Firmware details
TFT Firmware compiled from `master`, branch on 2021-03-05
SKR 1.4 Turbo firmware compiled from `bugfix-2.0.x` branch on 2021-03-05

### Additional Information
[TFT43 Configs 2021-03-05.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6094760/TFT43.Configs.2021-03-05.zip)
[SKR 1.4 Turbo Marlin Configs 2021-03-05.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6094762/SKR.1.4.Turbo.Marlin.Configs.2021-03-05.zip)

Foxcor, 2021-03-06:
This is not a screen or firmware issue.
Install additional plugins in Octoprint.
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress for example

oldman4U, 2021-03-06:
@Foxcor

Would you be able to provide some basic information regarding the things necessary to make Octoprint work with the TFT? I would add them to the ReadMe then. Thank you

lightmaster, 2021-03-06:
Ok, that makes sense then since the print I was trying was a bed leveling test print and has no layer information, therefor DisplayLayerHeight doesn't do anything for this print. iirc, the MKS Robin Nano I was using before would still show the print screen regardless of whether or not DisplayLayerHeight was going. Could be wrong though.

lightmaster, 2021-03-07:
Just printed a temp tower that does have layer information in it and OctoPrint showed the layer information itself, but the display never reacted to it printing. Just kept showing the same information start page.

lightmaster, 2021-03-07:
[image]

Possibly related to whatever is keeping the display from showing any printing updates, but the fan is on at around 20% and the TFT says 0%.

lightmaster, 2021-03-07:
Didn't wanna mess with it while it was printing, but yes, Marlin mode is getting the printing updates:

[image]

XYZ are fine, they were flashing due to steppers being off.

lightmaster, 2021-03-07:
PR #1681 claims `PR #1680 introduced a bug where if printing from SD remotely (Octoprint. Pronterface, etc) the printing menu would not activate`, however TFT mode is still not entering into the printing menu for me when I initiate a print from Octoprint's "local storage".

kisslorand, 2021-03-07:
It will not do that until someone implements it in the FW. As it is at the moment the TFT doesn't "know" that someone from the outside is printing and not just sending gCodes.

lightmaster, 2021-03-07:
I'm 99% sure that Octoprint sends out a message as soon as it begins a print that the TFT could use as a signal that a print has been initiated. I'm in the middle of a print right now but will check when its done.

kisslorand, 2021-03-07:
No need to check anything. I said it is not implemented, not that it cannot be implemented.

DaStivi, 2021-03-07:
there also an feature request allready...  #1175 and #810 i guess the main issue is as long as another print host is sending gcode, the tft cannot display the full "print-menu-Screens" because there would be buttons that just cannot work... and with that, there is quite some work todo, to achieve it.. but i've allready voted for that too, would be really ah great addition.. 

even though, as i started printing with octoprint lately, i see some potential, but still not fully commited to octoprint, rather it is the need to install 30ish plugins to get to the point what the btt tft can achive, and still missing one of the biggest features yet on octoprint is powerless return... even the simpliest version of it, as i found out recently :(

DaStivi, 2021-03-07:
i've even had another idea... i thought about having the Pi act as an USB Device to the TFT screen... so basically it would not be possible to start the print from the pi on the TFT (still on the board directly) but it would be possible to start the print from the TFT, reading from the Pi... didn't tested what happens when starting in normal condition now from TFT, if the pi is getting the correct info that some print is ongoiing... even can't imagine what would happen then when pi stops the print... anyway my idea also stopped immediatly as i just found out that the pi can't act as an USB Slave/OTG device :(

lightmaster, 2021-03-07:
@DaStivi Not sure what #1175 has to do with this...

I understand this hasn't been implemented yet, but not sure what buttons would not be able to work. Even just being able to see the basic printing information screen and being locked out of certain critical functions (ie: can't start print while one in progress, can't use movement commands, etc). As far as detecting a print is in progress, Octoprint sends a `M73` as soon as you press the Print button, and can't see why a user would be sending `M73` by hand.

And if Marlin on the main board can detect a print in progress and react accordingly, why wouldn't the TFT that's connected via serial to the main board?

kisslorand, 2021-03-07:
For example "Pause" would not work, "Stop" neither.

lightmaster, 2021-03-07:
> For example "Pause" would not work, "Stop" neither.

https://docs.octoprint.org/en/master/features/action_commands.html
Marlin can send action commands back to Octoprint to tell Octoprint to stop, pause, resume, etc. I do not currently know much about how the TFT FW interacts with Marlin, but I assume its possible for the TFT to forward supported action commands to Octoprint.

Job percentage would just be showing the received M73 and M117 commands. The current information screen can already show those M117 updates.

Admittedly, this is no longer a Bug Request as I had originally thought, but now a Feature Request. As for the reasoning behind it, I often look in on long running prints while they are printing and usually I do not have my laptop opened up. Being able to use the TFT to pause or cancel a print would be immensely helpful, especially if my wife looks at it while I'm not home and she sees its FUBARed the print. I also will sometimes initiate multiple prints from work with my wife removing the resulting parts in between print jobs.

I have to assume that I am not the only one who frequently uses a host program like Octoprint and would benefit from being able to also have local control of the print without the need for logging into the web interface.

lightmaster, 2021-03-07:
Also, as far as stopping and pausing, sending custom gcode like `M118 A1 action:cancel`, at least according to https://community.octoprint.org/t/how-to-stop-octoprint-from-lcd/7586/5

kisslorand, 2021-03-07:
I understand the need for such things but as I said, it needs another menu/screen. The present menu is not suitable for printing from outside from an outside storage.
Anyway, what if you print from Pronterface, what if from Cura, Fusion, etc, etc?
The TFT is made for printing from its interface.
For your wife you can always install OctoDash or similar.

lightmaster, 2021-03-08:
Custom Cancel GCode is already supported, which could stop Octoprint then, although there doesn't appear to be any Custom Pause GCode yet.
Pronterface supports action commands, at least according to that octoprint doc about the action commands. No idea about Cura or Fusion as I didn't know you could connect them direct to a printer without something like Octoprint being a middleman.

I'll look into OctoDash and also into learning C/C++ to be able to implement it myself.

lightmaster, 2021-03-08:
Closing this since in hindsight its not actually a bug

closed 2021-03-08 by lightmaster

oldman4U, 2021-03-08:
May I can ask two probably stupid questions?

Why would I like to do everything related to a print using the TFT but to
start the print?

As far as I know there is a screen available from BTT which allows to work
with Octoprint? Why not using this screen instead?

Thank you

DaStivi, 2021-03-08:
> @DaStivi Not sure what #1175 has to do with this...

sorry looks like there was ah typo... have meant #1170 (Featurerequest List)

btw. i still didn't understood what happens... i've re-read your posts, and do i understand you correctly that the "detailed progress" and "display layer progress" plugins doesn't show the updates in the "notification" window on the btt tft??

i also don't understand what the bugfix #1681 is no dooing?

i've just tested this now, still with the #1680 commit compiled and i just have the M117 status update on tft screen:

lightmaster, 2021-03-08:
> do i understand you correctly that the "detailed progress" and "display layer progress" plugins doesn't show the updates in the "notification" window on the btt tft??

M117 messages do show up in that box in the middle. I mistakenly thought that PR was supposed to allow all prints started remotely to initiate the printing menu, whereas I guess it just means all SD card prints started remotely.

I'm dropping this idea anyways, since clearly I'm in the minority that would care about this and don't want to get in the way of development of features that everyone will use. Sorry for wasting people's time.

DaStivi, 2021-03-08:
> M117 messages do show up in that box in the middle. I mistakenly thought that PR was supposed to allow all prints started remotely to initiate the printing menu, whereas I guess it just means all SD [...]

you're definitely not wasting anyones time, and its ah great idea, as i posted it has been requested ah couple of times allready :)

i gonna test later something... the 1680 does fix something that caused the touch mode print display shows up when printing from marlin mode onboardSD when serial_always_on = enabled ...

this has been fixed, and also there is bugfix now..

my idea is now, i go back to ah version with this bug again and like to see what happens when starting ah print from octoprint, in marlinmode display, would be interessting if the tft switches to touch print display then 🤣

oldman4U, 2021-03-08:
I can not speak for the others but you are not wasting my time!

There is always something to learn and just because I do not use it or understand what it is good for does not mean it should not be implemented.

lightmaster, 2021-03-08:
> my idea is now, i go back to ah version with this bug again and like to see what happens when starting ah print from octoprint, in marlinmode display, would be interessting if the tft switches to touch print display then 🤣

It's not a bug.... It's a feature 🤣🤣🤣

oldman4U, 2021-03-08:
Sometimes it is hard to know....

And sometimes it is a feature for one and a bug for the others. 🤯

lightmaster, 2021-03-08:
> I can not speak for the others but you are not wasting my time!
> 
> There is always something to learn and just because I do not use it or understand what it is good for does not mean it should not be implemented.

My thoughts about it were simply consolidation. It seems there are ways of communication between remote hosts like Octoprint and Marlin (action commands), so why not use that to allow easier control? Regardless of how a print is initiated, the screen physically attached to the printer could have at least basic control such as pausing, stopping, and resuming. The print screen would already be able to control fan speed and temp adjustments (at least until gcode applies new values for these). The printing source like Octoprint can send out M73 progress updates for the progress bar and time remaining type stats.

I have basically no specific knowledge of programming in C/C++ and only understand a tiny amount of the code for this, so maybe it's just not possible, idk. But I feel like there has to be a way for the screen attached to the printer that has serial communication with it to be able to have at least basic control regardless of where the print commands are coming from.

DaStivi, 2021-03-08:
yes i definitely agree...   and i really love the bTT TFT... evenmore as i just found out what everything does not work from octoprint (like powerloss recovery)   i search of ah way to combine each other at possible best solution to assist my printing..

and i also have almost no knowledge about coding so i really can help finding bugs, this what i am good at looks like, because i try ah lot of things and really master into stuff :)

kisslorand, 2021-03-08:
It is possible to have a print status/control screen when printing from USB, it just have to be implemented. The current menu during print from TFT is not suitable for this.

oldman4U, 2021-03-08:
Not sure I remember right but some users used a UART connection for this I believe 🤔

DaStivi, 2021-03-08:
@oldman4U, thx i did find that... https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/281

so it looks like it is possible to connect the pi with uart directly to the tft... so what i still don't see if the tft would then show the printing menu correctly?

DaStivi, 2021-03-08:
and further more some interessting comment caught my eye...
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/281#issuecomment-650721626

there @kind3r is reports random freezes of the tft when octoprint is connected... interesstingly as i also still have random freezes and i've connected the ESP01s module to the uart port on the tft all the time! so this is still might be an issue and every user caught that just moved on and didn't use it anymore ?!

DaStivi, 2021-03-08:
> my idea is now, i go back to ah version with this bug again and like to see what happens when starting ah print from octoprint, in marlinmode display, would be interessting if the tft switches to touch print display then 🤣

i gotta tested this now... and its broken as hell... also printing from octoprint directly does not trigger the bug... so only printing from pi from the onboardSD have triggered it.. so octoprint did displayed the ongooing print job, and also the TFT did swicht from marlin to touch print display so far so good.. but of course everything else gooing crazy... as one of the first issues is that when printing fom onboardSD processing of adjustment commands is put into the queue, so octodash/print and TFT rather displays different thing when trying to change anything from on or the other side...

interesstingly pausing the print did worked from TFT, strange thing is that the print head moved to ah position where i never saw it at pause?! 🤷‍♂️

[image]

unpausing didn't worked, neither from TFT nor Octodash...
[image]

also canceling, TFT doesn't even do anything, octodash displayed canceling but looks like never do anything...

[image]

so funny little test but not to recommend in anyway 🤣

kind3r, 2021-03-08:
Indeed, there were issues when connecting via the screen (`PI <-> TFT <-> Board`). It has been a long time and I did not get a chance to followup on the various fixes since then (tho like mentioned it's a minority issue so it was not getting enough attention) so don't know if this is fixed or not. I tried to understand what could cause the TFT to stop forwarding but I got nowhere. Still, even in this proxy mode, you would not get a printing screen and controls for pausing/stopping etc., but you could at least monitor the progress of your print sent via the `M117` commands.

I ended up connecting both the PI and the TFT directly to the Board as it has 2 serial ports. This does not allow me to monitor the printing status or pause/stop the print on the TFT (except in Marlin mode ... yuk), but I can still change most params on the fly while it is printing. What does bug me tho is that using the rotary encoder during printing changes the feed rate and if I'm not careful that could mess up my print sometimes.

Now that I have a JTAG debugger I might be able to dig a bit deeper into the uart issue, as time permits. If this issue is fixed, there are 2 options:
- Do nothing and just monitor the progress via `M117` sent by OctoPrint, this should work without any other intervention.
- As @kisslorand mentioned, it would be possible to create a new screen (or customise the current printing screen) that gets triggered by something OctoPrint sends (like for instance line numbered commands).

Going one step further, as a crude version, I belive pausing can be achived by not forwarding OctoPrint's g-code to the Board and sending busy responses back. Stopping could just disconnect OctoPrint and cause it to abort printing.

DaStivi, 2021-03-08:
> but you could at least monitor the progress of your print sent via the `M117` commands.
this works with Pi <-> Board <->TFT too... there are plugins on octoprint to send the M117 commands, and TFT does display it (as long as the status "screen" is enabled)
[image]

kind3r, 2021-03-08:
In the case of `Pi <-> Board <->TFT` the TFT will not receive the `M117` commands as they would go directly to the Board. Unless you use the same serial port on the Board, but I don't think that will ever work as it would confuse both the TFT and the Board.

DaStivi, 2021-03-08:
> In the case of `Pi <-> Board <->TFT` the TFT will not receive the `M117` commands as they would go directly to the Board. Unless you use the same serial port on the Board, but I don't think that will ever work as it would confuse both the TFT and the Board.

that definitely works as expected! you can see my pictures in this topic here! https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1673#issuecomment-792581425

Guilouz, 2021-05-30:
@digant73 Do you think it's possible now to switch to printing menu when printing with Octoprint with this PR :

https://github.com/MarlinFirmware/Marlin/pull/21987

[image]

digant73, 2021-05-30:
@Guilouz no. It only informs the host (TFT, Octoprint etc...) if host actions are supported (enabled) or not by Marlin. That information can be used by the host just to understand if it can use features requiring host actions or not

Guilouz, 2021-05-30:
> @Guilouz no. It only informs the host (TFT, Octoprint etc...) if host actions are supported (enabled) or not by Marlin. That information can be used by the host just to understand if it can use features requiring host actions or not

Ok Thanks.