bigtreetech/BIGTREETECH-TouchScreenFirmware #1676: [BUG] Popup error.

radek8, 2021-03-06:
Pop-up error when printing was not completed.
If my primary language is set to English, a popup window will appear normally.
If the secondary language is set, both the window name and the dialog buttons have the wrong description.

EN
[image]

PL
[image]

CZ
[image]

kisslorand, 2021-03-06:
Can you check by using second language also English (or any other)?

radek8, 2021-03-06:
Secondary English

[image]

kisslorand, 2021-03-06:
Thx for the test, the result makes it a general bug.

PR #1680 (2021-03-07): Bugfix: boot issue in marlin mode and prevent mode switching while printing, wrong keywords in Popup dialog
- Fix freeze/hang/boot issue if booting in marlin mode with serial_alway_on active. (#1657)
- Fix mode switching issue if printing from onboard SD with Marlin Mode active due to changing menu index.
- Fix wrong keywords in Popup dialog. (#1676)

Resolves: #1657, Resolves: #1676
closed 2021-03-07 by bigtreetech