bigtreetech/BIGTREETECH-TouchScreenFirmware #1679: TFT43_V3.0 no serial comms after update

Ashmarhan, 2021-03-07:
I am running a BTT TFT43_V3.0 touchscreen on a BTT GTR_V1.0  motherboard.
I updated the firmware on the TFT43 to the precompiled firmware on this site, and immediately lost serial communication.
The screen and motherboard communicate perfectly in marlin mode.
I have tried adding a third port to the GTR, changing port numbers, changing baud rates, connecting the other UART pins on the screen board, but nothing has worked

I am at the point now where I believe that the new firmware has bricked the screen.

The box has these numbers:
ZZB000416
BTT TFT43 v3.0
D/C:2020.7.21

lightmaster, 2021-03-07:
IDK about the factory firmware cuz mine came with some super outdated firmware on it. If you don't have the previous firmware you used, head to https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commits/master and click the `<>` icon to the right of one of the commits titled "Update prebuilt binaries and config" and you can grab the older binary you had used before to test.

Ashmarhan, 2021-03-07:
Thanks, I'll try the last firmware before the production date of my screen. Hopefully I get comms back

lightmaster, 2021-03-07:
Do you know what the Baud Rate of your main board is? They should be set to the same. I'm using 250,000 on both and they get comms. And idk much about ports, or which one does which, but `SERIAL_PORT -1` and `SERIAL_PORT_2 0`.

I've got the same TFT as you btw, but using it on a SKR 1.4 Turbo

Ashmarhan, 2021-03-07:
Out of the box it was working until I updated it.

I tried all baud rates after the update, tried adding SERIAL_PORT_3, tried changing all the port numbers around, and tried the alternative UART ports on the TFT to make sure it wasnt being ported down the wrong one.

Bad things get worse though, the older firmware hasn't worked. Now im going to have to try reflashing my GTR, and maybe use an alternative UART. I hope the TFT firmware hasn't somehow damaged my GTR.

Ashmarhan, 2021-03-07:
Ok, WOOHOO
Went back to the old firmware from 2 July 2020, then reflashed the GTR with my current 2 port setting, without the screen connected to the TFT plug of the GTR. Now I'm back to square one... but this time I have a firmware to fall back on.

Thanks for spending some time to help this noob

lightmaster, 2021-03-07:
Check your settings that are working, and then work your way forward in builds until it breaks. Then you'll know that something that was changed between build A and build B must have caused it to break.

Ashmarhan, 2021-03-07:
Onwards and upwards. My major sticking point was the lack of current firmware out of the box. I cant understand why they dont include it with the screens like they do with the motherboards.

closed 2021-03-07 by Ashmarhan