bigtreetech/BIGTREETECH-TouchScreenFirmware #1684: [BUG] BIGTREE_TFT24_V1.1.26.x.bin not updateable (master)

the-real-orca, 2021-03-07:
### Description

When updating my TFT via SD card, the bootloader is running to `Updating: 99%` but then the screen gets black and nothing happens.

### Steps to reproduce

1. checkout current master branch (2021-03-07)
2. copy precompiled *BIGTREE_TFT24_V1.1.26.x.bin, config.ini* and *TFT24* to SD card
3. insert SD card into TFT24-V1.1 and reset board

**Expected behavior**
Update firmware, icons and fonts.

**Actual behavior**
Update to 99%, then screen get blank and update process stops.

### Hardware Variant
Bigtreetech  TFT24-V1.1

### TFT Firmware Version & Main Board Firmware details
V1.1.26 release is working

It seems to be broken between 8baf77ed426ca3cb151f97ea82d8c5c0e401d3af (2021-03-04) and 5c9fd7830d48108933772a0acb52e668d956105b (2021-03-06).
Updating with 8baf77ed426ca3cb151f97ea82d8c5c0e401d3af works fine.

bttguy, 2021-03-07:
Same for me. Any clue about the issue?

AngryShadowNANO, 2021-03-07:
same issue witht BTT TFT24 V1.1
only gets to 98%, then gets black screen.

guruathwal, 2021-03-07:
There is some problem after #1562. Please check if the problem does not exist in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/f021fa3aed7ffaf053fd7bbebbeb1a20e800f1f1. You will have to build the firmware to test.

bttguy, 2021-03-07:
I am familiar with building the fw with the latest commits included, but I don't know hot I can exclude specific commits and use a previous state of the fw. Can you help me with that, so I can help to troubleshoot.

guruathwal, 2021-03-08:
You need to open the commit history here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commits/master
then click on the `<>` button to browse the repo at that specific commit.
[image]

bttguy, 2021-03-08:
Thank you for the guidance @guruathwal ! Using the mentioned state of the fw (f021fa3 - Cleanup and readuce ram) its working as normal.
You mentioned that we need to build the fw for this. I used the precompiled one with the previous "release" I downloaded, am I did it right or I need to compile the fw in vs code regardless the built in auto compile function tha exist for a while?
(I only have a test board right now, not a working printer, but it boots up, loads the icon and the menu navigation is good, cant say anything about print progress or temp detection tough I assume they are also working)
If you need any more testing with different commits, let me know and I can test them!

guruathwal, 2021-03-08:
@bttguy  good to know that. you need to compile only when you need to change anything other than the settings in the config.ini file. After https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/f021fa3aed7ffaf053fd7bbebbeb1a20e800f1f1 there is some problem with the compiling framework, this is why I asked you to compile it. The bug will be resolved very soon then there will no need to compile for stock firmware.

oldman4U, 2021-03-08:
Last but not least... verified on my TFT24 v1 that your PR is fine Gurmeet but after that it is not working anymore. Everything works fine on the TFT28v1

AngryShadowNANO, 2021-03-09:
works now with the master folders, but i think there could at least be a 2 more pictures on the images folder from the guideline from the readme file instrucktion with the renaming TFT folder and adding the bmp and font folders inside to the **step2:** area from readme text

oldman4U, 2021-03-09:
Hi.

step 2 describes that you have to insert the SD card into the TFT and restart the printer or reset the TFT.

Which elements have to be copied onto the SD card is described in step 1 and there the TFTxx folder is part of it. Why would it be important to mention what's inside the TFTxx folder? There is also no renaming needed.

Please let me know

Thank you

AngryShadowNANO, 2021-03-09:
well because the theme folders are only contains bmp and font ... what about the TFT folder from `BIGTREETECH-TouchScreenFirmware-f021fa3aed7ffaf053fd7bbebbeb1a20e800f1f1\TFT` not needed then? 
and yes it was the `**element2**` i meant by.

oldman4U, 2021-03-09:
Not sure where you got this link from (BIGTREETECH-TouchScreenFirmware-f021fa3aed7ffaf053fd7bbebbeb1a20e800f1f1\TFT) but you copy ONE TFT folder to the SD card, which contains the user interface theme you would like to use and the fonts. Currently there are 4 interfaces available and you can use any of them and they all contain the same font folder in addition to the bmp's. Then you copy the config.ini and the firmware (self or precompiled) from the same firmware version on the SD card and use this to update the firmware.

One firmware
One config.ini
One TFTxx folder from the user interface you like most
and optionally a language file.

Voila

oldman4U, 2021-03-09:
Next version of the ReadMe will be changed slightly because of your input:

step 1: Copy your self compiled firmware or the BIGTREE_TFT*_V*.*.*.bin, plus the TFT*` folder **from your preferred interface theme**, as well as the config.ini to the root of a blank SD card that is <8GB and formatted as FAT32:

Thank you

bttguy, 2021-03-09:
Any update on this bug @Msq001? I even tried to compile my own fw.bin, but that doesnt seemed to help either. Dont want to hurry anybody, just interested in how bad is this bug, and what to expect when can I use my TFT with the latest changes :)

Msq001, 2021-03-10:
Hello, Sorry for this fault. #1703 has fixed

bttguy, 2021-03-10:
@Msq001 Thank you for the fast fix!! :)

the-real-orca, 2021-03-11:
Update with the current master is working again. **Thanks for the fast response!**

However I got an error with the new config.ini: 
```
	Updating Configuration ...
	Invalid Value in: rrf_macros_on: 0
```

bigtreetech, 2021-03-11:
> Update with the current master is working again. **Thanks for the fast response!**
> 
> However I got an error with the new config.ini:
> 
> ```
> 	Updating Configuration ...
> 	Invalid Value in: rrf_macros_on: 0
> ```

This is caused by the latest PR of RRF compatibility. will be fixed soon.

PR #1703 (2021-03-10): fix TFT24 start flash address
### Description
fix TFT24 start flash address

### Benefits

### Related Issues
fix #1684