bigtreetech/BIGTREETECH-TouchScreenFirmware #1687: [FR] Make the LED/Lights menu in TFT mode more feature complete

lightmaster, 2021-03-07:
**Is your feature request related to a problem? Please describe.**
Can not set a Custom Light using the TFT Touch Screen mode without resorting to using terminal and `M150` command. If the average user that doesn't know much about gcode commands wants to change the LEDs to something besides Red, Green, Blue, and White, then they have to switch to Marlin mode and can then have full access to Marlin's much more fleshed out Lights menu. Furthermore, I propose expanding the preset colors to also include the other primary and secondary colors used by Marlin (ROY G BIV colors).

**Describe the solution you'd like**
Kinda feel that's pretty self explanatory from above, but I'd like the TFT Touch Screen LED menu to be more fleshed out and have the options that Marlin Mode has.

**Describe alternatives you've considered**
Encourage anyone wanting to play around with the LED colors to use Marlin Mode, or continue to require them to understand GCode and RGB color codes

**Additional context**
Add any other context or screenshots about the feature request here.

oldman4U, 2021-03-07:
How many of this ecustom colours should be possible and how would such custom colours look like on screen, custom buttons?

I can see this a lot of work for little benefit and there are custom gCode commands which could be used for that.

my 2 cents

bigtreetech, 2021-03-08:
Maybe it can add a custom button in the LED Color menu. After clicking in, it can directly set the value of each R/G/B by the keyboard?
It's a good idea, and It can reuse Custom icon without adding new icons avoid taking up more flash memory.
We are going to release a new version of firmware in the near future. The main task at present is to test and fix bugs. If you are willing to submit PR to add this function , I will be verry happy to test it.

lightmaster, 2021-03-08:
Going back to the LED menu, I didn't realize how few space there is without shrinking the icons and potentially causing issues with smaller screens and bigger fingers....

So what about a Custom icon like @bigtreetech described, and potentially a second icon called "Other" to fill the final open spot and it directs to a menu that has a list various other colors. This menu would look like the Custom GCode menu.

DaStivi, 2021-03-08:
i think it would be ah nice idea... I'd might have another one, do add some sort of option to auto enable lightning from the features menu? like to turn on white light from the start without having to go into the menu's or use the gcode

lightmaster, 2021-03-08:
> 
> 
> i think it would be ah nice idea... I'd might have another one, do add some sort of option to auto enable lightning from the features menu? like to turn on white light from the start without having to go into the menu's or use the gcode

You mean when the printer turns on? My NeoPixel turns on at boot. If you have an RGB strip and it doesn't, that's probably a better thing to bring up in the Marlin Github. If you have NeoPixel, check your config: https://github.com/MarlinFirmware/Marlin/blob/b26516b3ac7cd7d15a751af4b299a38ac8ac7a57/Marlin/Configuration_adv.h#L1107

oldman4U, 2021-03-17:
Hi. What is the status of the feature request?

bigtreetech, 2021-03-18:
> Hi. What is the status of the feature request?

Waiting for implementation

digant73, 2021-04-18:
Now it is implemented

PR #1853 (2021-04-22): New LED Color menu + some code optimization
**NEW FEATURES:**
* added custom LED menu:
 1) it is now possible to set all the parameters for M150 gcode (machine RGB LED)
 2) the set LED color is previewed at runtime. Any change is scheduled to be applied every 1 second at maximun, just to avoid useless traffic (to send M150 gcode) every time a component is changed
 3) RGB component colors and RGB color are also displayed just as color pickup. Obviously RGB color on display is limited to RGB 565 format while machine LED uses the wider range 0-255 for each component
 4) the LED main menu allows t [... cut]
closed 2021-04-22 by bigtreetech