bigtreetech/BIGTREETECH-TouchScreenFirmware #1689: [BUG] Issue with M600 when printing from SD/USB TFT

Guilouz, 2021-03-07:
### Description

There is a recent issue with the M600 when printing via SD / USB from the TFT (this has worked before).

M600 is considered like a simple pause, TFT does not indicate that the hotend has been parked (but it comes well parked
), no retraction of the filament is performed, no popup to ask the user to purge again or to restart the print. There is only the start button which allows you to resume printing as a simple pause.

No issue when printing from Octoprint, M600 plays the job as it should and the TFT displays the necessary popups.

### Steps to reproduce

Just print from SD/USB with M600 in Gcode file

oldman4U, 2021-03-07:
Do you know which PR broke it?

Guilouz, 2021-03-07:
> Do you know which PR broke it?

I tried to find it but without success:(

kisslorand, 2021-03-07:
Check _interfaceCmd.c_ for:

        #ifdef NOZZLE_PAUSE_M600_M601
          case 600: //M600/M601 pause print
          case 601:
            if (isPrinting())
            {
              setPrintPause(true, false);
              // prevent sending M600/M601 to marlin
              purgeLastCmd();
              return;
            }
            break;
        #endif

and see what happens if you comment `purgeLastCmd()`.

labotecno, 2021-03-08:
> Check _interfaceCmd.c_ for:
>
> ```
>     #ifdef NOZZLE_PAUSE_M600_M601
>       case 600: //M600/M601 pause print
>       case 601:
>         if (isPrinting())
>         {
>           setPrintPause(true, [...]

yes will be a good thing to check every step of the script M600 like lcd grafic way .could be possible ?

Guilouz, 2021-03-08:
> Check _interfaceCmd.c_ for:
>
> ```
>     #ifdef NOZZLE_PAUSE_M600_M601
>       case 600: //M600/M601 pause print
>       case 601:
>         if (isPrinting())
>         {
>           setPrintPause(true, [...]

I'm testing this tonight. Thanks.

Guilouz, 2021-03-08:
> Check _interfaceCmd.c_ for:
>
> ```
>     #ifdef NOZZLE_PAUSE_M600_M601
>       case 600: //M600/M601 pause print
>       case 601:
>         if (isPrinting())
>         {
>           setPrintPause(true, [...]

Just try with purgeLastCmd() commented and it's worse, the screen freeze when M600 is supposed to trigger.

kisslorand, 2021-03-09:
Try what happens if you send `M25 P`.
`M600` just sends `M600`, nothong more, nothing less.

Guilouz, 2021-03-09:
> ADVANCED_PAUSE

Yes ADVANCED_PAUSE is enabled, all working fine when printing from Octoprint, just issue with SD/USB from TFT.

I will try with M25 P, with M125 in terminal no issue, I have 'Nozzle parked' popup, 'PurgeMore' etc...

kisslorand, 2021-03-09:
This ticket is about `M600`. I will look into it. In fact I wanted to do that today but I am too tired.

Guilouz, 2021-03-09:
> This ticket is about `M600`. I will look into it. In fact I wanted to do that today but I am too tired.

Yes M600, but it’s understood as a simple M25 pause when printing from SD / USB, that's why I wanted to test the M125 which behaves like an M600.

kisslorand, 2021-03-09:
No, M25 is triggered only if printed from Onboard SD. M600 triggers the same as if you would press "Pause".

Guilouz, 2021-03-10:
> No, M25 is triggered only if printed from Onboard SD. M600 triggers the same as if you would press "Pause".

Precisely the M600 is not supposed to trigger the same thing as a pause. That's you mean ?

A simple pause : press "pause" on tft, if PARK_HEAD_ON_PAUSE is enable, hotend is parked until you press "Resume"

With M600 : when is detected, TFT show a popup "Nozzle parked", filament is retracted, the heating of the hotend is stopped (safe if the user is not in front of his machine), TFT ask to user to heat hotend, when temperature is reached, a purge of filament is done, TFT ask to user if he want to purge more or continue the print. It's the same process as with a detection of filament sensor.

bigtreetech, 2021-03-10:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/78a7560b3678f4456e130191a945af54fbf055ec/TFT/src/User/Configuration.h#L302
Can solve your problem from comment out here?

digant73, 2021-03-10:
@Guilouz aren't you sure that in the past it worked because you commented out #define NOZZLE_PAUSE_M600_M601 in Configuration.h?
As far as I remember the emulated M600 (so handled by TFT) was always the same (a simple pause).
If you comment out #define NOZZLE_PAUSE_M600_M601 the M600 gcode is sent and handled by Marlin (if you configured it properly)

Guilouz, 2021-03-10:
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/78a7560b3678f4456e130191a945af54fbf055ec/TFT/src/User/Configuration.h#L302
> 
> 
> Can solve your problem from comment out here?

#define NOZZLE_PAUSE_M600_M601 is not commented in my configration.h

kisslorand, 2021-03-10:
>  no retraction of the filament is performed,

Are you sure? Not even a few mm?

closed 2021-03-10 by Guilouz

oldman4U, 2021-03-10:
Do I understand this right that for one it works in case #define NOZZLE_PAUSE_M600_M601 is Deactivated and for the other one it works in case it is activated....?

oldman4U, 2021-03-10:
Thank you. Makes sense.

digant73, 2021-03-11:
> > Do I understand this right that for one it works in case #define NOZZLE_PAUSE_M600_M601 is Deactivated and for the other one it works in case it is activated....?
> 
> When disabled : M600 working as expected with configuration set in Marlin (nozzle parked, filament retraction, hotend heating deactivated until the user intervenes, possibility to purge more when the filament is inserted
> 
> When enabled : M600 behaves like a classic pause, like when pressing pause from the screen, just parked.

With M600 enabled you can also purge/retract with the extrude menu. So you should be able to do the same as the M600 handled by Marlin. Just using the features / menus provided by the TFT

thsrp-bob-roger, 2021-03-11:
> With M600 enabled you can also purge/retract with the extrude menu. So you should be able to do the same as the M600 handled by Marlin. Just using the features / menus provided by the TFT

No, because you have a big retraction just before printing resumes and thus ruining the printing. But now with the M600 deactivated you have access to a real M600 with its configuration which was made in Marlin and it works perfectly ... In fact it is a bit strange and confusing to have to deactivate the M600 to have a REAL M600? ! Logic would have it the other way around ..?

digant73, 2021-03-11:
Just to collect feedback for further improvements on emultaed M600, so the main problem with the emulated M600 is the excessive rectraction before the resume. Is everything else fine or there is something else that could be improved/fixed?
M600 is not supported on all the mainboard and in the past it was not properly handled by Marlin. For that reason by default emulated M600 was enabled.
The idea is to make the choice emulated M600 / real M600 configurable in config.ini in a further release

thsrp-bob-roger, 2021-03-11:
> Now that M600 is properly handled by Marlin, the idea is to make the choice emulated M600 / real M600 configurable in config.ini in a further release

Yessss great idea !!