bigtreetech/BIGTREETECH-TouchScreenFirmware #1701: [BUG] controller fan option interferes with part cooling fan reading

dielectric-code, 2021-03-09:
### Description
When enabling a controller fan in config.ini the part fan reading is wrong.

### Steps to reproduce

1. set fan_count : 1 and  fan_crtl_count: 1 in config.ini
2. update config.ini
3. set the part cooling fan to any value i.e. 100%,
4. the display will alternate between F0 and Fctrl
5. reading of Fctrl is ok
6. reading for F0 (part cooling fan) will read OK for a second or two, then read 0

**Expected behavior**

Correct reading of F0 and Fctrl
**Actual behavior**

Wrong reading of F0 after a second or two
### Hardware Variant

TFT35 E3 V3
### TFT Firmware Version & Main Board Firmware details

firmware form Copy folder compiled on 8 march 2121

radek8, 2021-03-09:
I can confirm, after the last few PRs, the fan speed is not displayed on the status screen.

oldman4U, 2021-03-10:
dielectric-code

Which mainboard do you have. Are you sure that you have to activate the ctrl fan? Using a single part cooling fan I can control the fan without a problem using the Touchscreen and Marlin Emulation mode on an SKR E3 DIP without having the ctrl count set to 0. Haven't tried it with my SKR PRO, which has more controllable fans.

bigtreetech, 2021-03-10:
I confirmed it and it will be fixed soon.

oldman4U, 2021-03-11:
Hi. Fix is available. Please check

closed 2021-03-11 by bigtreetech

dielectric-code, 2021-03-11:
Works perfectly now. Thanks, that was quick!

PR #1712 (2021-03-11): cleanup & fix controller_fan
### Requirements

* Filling out this template is required. Pull Requests without a clear description may be closed at the maintainers' discretion.

### Description

### Benefits

### Related Issues
fix #1701