bigtreetech/BIGTREETECH-TouchScreenFirmware #1713: [Q] Debuging with JLINK does not work out of the box

kind3r, 2021-03-11:
I've been trying to connect to the TFT35_V3_0 using a JLINK with PlatformIO to try and debug some issues but it did not work out of the box. Uploading firmware worked (so connection is OK) but when starting debug I always got an error complaining about missing program/firmware. After more investigations I noticed that the elf produced after linking does not have the `.elf` extension and that is the reason why the JLINK debugger fails to start. Changing the `executable` inside `.vscode\launch.json` did not work and that file is automatically generated anyway. So I added a copy command inside the `buildroot\scripts\short_out_filename.py` to add the required .elf extension. This results in 2 files, one with .elf and one without.

Since I'm new to debugging MCUs, my questions are: 
- any reason why the elf file has no extension ?
- is there a better way of doing this ?

guruathwal, 2021-04-30:
I do not have a JLINK for debugging but I checked the `.elf` file. After compiling the '.elf' file is created and name properly on my system.
[image]

oldman4U, 2021-05-06:
Any news? Gurmeet tries to help you. Would be nice to answer him.

kind3r, 2021-05-06:
Unfortunately I did not have time to re-test but at the time of writing the .elf had no extension at least on MacOS. I will get back with more info after the weekend.

oldman4U, 2021-05-07:
I just got this under macOS 10.15.7:
[image]

Have a nice weekend

kind3r, 2021-05-07:
[image]