bigtreetech/BIGTREETECH-TouchScreenFirmware #1720: [BUG] Problem with the Printing.sys file

radek8, 2021-03-11:
Unable to confirm pop-up asking if you want to continue printing after the interrupt.
The screen freezes when clicked

[image]

kisslorand, 2021-03-11:
What was the cause of the interruption?

radek8, 2021-03-11:
Stornovaný tisk, ale to není důležité.
Na tlačítko se vygeruje soubor Printing.sys s infornmací kde byl tisk přerušen. Po zvolení nového tisku, vyskočení dotazu chcete pokračovat v přerušeném tisku.
Po kliknutí na kterékoli tlačítko displej zamrzne.

kisslorand, 2021-03-11:
Huh?

radek8, 2021-03-11:
The Printing.sys file is generated if the feature is enabled
Power Loss Recovery

kisslorand, 2021-03-11:
Can you show the content of the Printing.sys file? @digant73 is on this bug.

radek8, 2021-03-11:
[Printing.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6125893/Printing.zip)

digant73, 2021-03-11:
it happened also to me. In my case the content of the file was:
SD:/very_very_very_log_long_long_very_very_very_long_long_long_file.gcode
so more than 63 chars.
The TFT freezes without prompting the "Continue printing?" dialog

radek8, 2021-03-11:
put these two files on the card and select print
[Uploading Archive.zip…]()

digant73, 2021-03-11:
used also you files. the problem is reading the Printing.sys file. On my printer the TFT freezes on "Loading" screen.
While in your TFT the dialog box was shown but then the TFT freezes

radek8, 2021-03-11:
It didn't freeze before the translation was corrected.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1676
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1680

digant73, 2021-03-11:
Ok, so it should more easy to solve the bug

bigtreetech, 2021-03-12:
fixed in #1724 .

PR #1724 (2021-03-12): fix PLR freeze
### Requirements

* Filling out this template is required. Pull Requests without a clear description may be closed at the maintainers' discretion.

### Description
`loadLabelText((uint8_t*)okTxt, LABEL_CONFIRM);` function will always to modify MAX_LANG_LABEL_LENGTH of RAM, when the array is only 50, this function will access overflow and tamper with some unknown RAM data.

### Benefits
fix #1720

### Related Issues
closed 2021-03-12 by bigtreetech