bigtreetech/BIGTREETECH-TouchScreenFirmware #1723: [FR] Separate config.ini for RRF

bttguy, 2021-03-12:
**Is your feature request related to a problem? Please describe.**
RRF compatibility is starting to became an active project #1400 . As RRF is a different firmware, it requires different settings in the current config.ini to be able to used. I think the current situation could be improved.

**Describe the solution you'd like**
I could imagine two different config files like config.ini_Malrin and config.ini_RRF. This could be beneficial both for easier testing and for new users that would want to set up their TFT's for RRF. This could also give the possibility of some restructureing of the config.ini_RRF as some of the lines are not affect our usecase, only Marlin mode. It could lead to less user error as well that came from a wrong config.

**Describe alternatives you've considered**
Edit config.ini each time I test a new commit. Or save a tuned config in a hope that no new line added to it, that could cause a possible error.

**Additional context**
Since Bigtreetech released their RRF compatible board, and there are a lot of users who already use BTT board with BTT TFT's with RRF it would beneficial for new users as well as for BTT if some quality of life improvements could be made for RRF users.
In exchange of a simple rename (or somebody could come up with a coder sollution not needed as with the different fw.bins) and some extra caution (but with so many fw.bins it is a proven doable thing) both testing and user experience would be better.
It would also require a bit getting used to from long time users, and in case of a bigger code change two config.ini-s need to be maintained, but this is why I ask it, to discuss the possible pitfalls I dont thought of. At my level of knowledge config.ini is just an overwrite file not a hard configuration.h change, so it works in my head, but open for any thoughts.

In case this is a welcomed change I would be happy to help to set up and maintain a new config.ini_RRF as I am also a RRF user and with a BTT TFT, and in case some codeing is needed I can seek for some help as well as help in the documentation changes this FR makes necessary.

bigtreetech, 2021-03-12:
Thank you for your idea!  config.ini file for RRF will be added in the next RRF PR.

bttguy, 2021-03-12:
Thank you, sounds great! Here are some changes that is required in the first place:

baudrate: 4 (57600)
onboard_sd_support: 1
serial_always_on: 1
rrf_macros: 1

For config.g the following lines required:
M575 P1 S3 B57600

Let me know if you need any help or testing!

closed 2021-03-19 by bigtreetech

bttguy, 2021-03-19:
Thank you for the merge! @bigtreetech and @Msq001

PR #1743 (2021-03-19): rrf auto detect remote print
### Requirements
Need to add the following command to simulate the marlin response in the sys file
* `M409 K"job.file.fileName"` in `start.g`
* `M118 P2 S"//action:cancel"` in `stop.g`
* `M118 P2 S"//action::paused"` in `pause.g`
* `M118 P2 S"//action::prompt_begin Resuming"` in `resume.g`

* `M575 P1 S2 B57600` in `config.g` of RRF to set baudrate to 57600(The baudrate of the TFT and motherboard must be the same)
### Description
The TFT will auto goto printing menu when motherboard start printing by DWC or USB

### Benefits

### Related Issues
fix #1723