bigtreetech/BIGTREETECH-TouchScreenFirmware #1730: [FR] Option to have GUI upside down (rotate 180°)

radry, 2021-03-14:
It would be good to have an option to rotate the GUI 180° so the TFT can be used "upside down".
This would add more flexibility how to mount the screen on the printers.

bttguy, 2021-03-14:
@radry I don't have my TFT in front of me atm, but you can do it in the settings. "Rotate UI" if I remember right, you can even rotate it using the TFT itself. Let me know if you found it.

kisslorand, 2021-03-14:
> It would be good to have an option to rotate the GUI 180° so the TFT can be used "upside down".
> This would add more flexibility how to mount the screen on the printers.

Let me edit this for you:
"It would __is__ good to have an option to rotate the GUI 180° so the TFT can be used "upside down".
This __adds__ more flexibility how to mount the screen on the printers."

radry, 2021-03-14:
Good to know

radry, 2021-03-14:
Good to know

closed 2021-03-14 by radry