bigtreetech/BIGTREETECH-TouchScreenFirmware #1731: filament runout Gcode issue.

sinole1987, 2021-03-14:
Hi 
I have this problem when filament runs out and I change filament and purge it. the nozzle moves to the top of the print and purges again. I wish I could set up so when I press resume, it primes at park then and moves to the print and just prints. I don't know where should I change the Gcode. I have limited programing skills. 

// Pause Settings
#define NOZZLE_PAUSE_RETRACT_LENGTH               15  // (mm)
#define NOZZLE_RESUME_PURGE_LENGTH                16  // (mm)
#define NOZZLE_PAUSE_X_POSITION     (X_MIN_POS + 10)  // (mm) Must be an integer
#define NOZZLE_PAUSE_Y_POSITION     (Y_MIN_POS + 10)  // (mm) Must be an integer
#define NOZZLE_PAUSE_Z_RAISE                      20  // (mm)
#define NOZZLE_PAUSE_E_FEEDRATE                  600  // (mm/min) retract & purge feedrate
#define NOZZLE_PAUSE_XY_FEEDRATE                6000  // (mm/min) X and Y axes feedrate
#define NOZZLE_PAUSE_Z_FEEDRATE                  600  // (mm/min) Z axis feedrate

I can set the pruge to zero. but the normal pause will not work great. and I don't think this is the code that runs in filament runout pause. Please help me fix my filament runout. I just lost a 10 hour print.

kisslorand, 2021-03-14:
Pr #1718 should fix this

oldman4U, 2021-03-14:
4min

New record?

👍🏻

digant73, 2021-03-14:
@sinole87 #PR1718 fixes that bug. Furthermore You can eventually enable M600 feature provided by Marlin (you must enable on marlin all the settings reported on top of the config.ini file). You are currently using the M600 feature emulated by the TFT

sinole1987, 2021-03-14:
thanks everyone, I prefer least amount of emulation possible. i know my way around marlin better than the tft program.

sinole1987, 2021-03-14:
> @sinole87 #PR1718 fixes that bug. Furthermore You can eventually enable M600 feature provided by Marlin (you must enable on marlin all the settings reported on top of the config.ini file). You are currently using the M600 feature emulated by the TFT

I have M600 activated in marlin, but in tft mode it runs the emulation. is there a way for TFT mode to perform marlin M600 directly, instead of emulation?

kisslorand, 2021-03-14:
For the moment disable `#define NOZZLE_PAUSE_M600_M601` in "Configuration.h".

sinole1987, 2021-03-14:
> For the moment disable `#define NOZZLE_PAUSE_M600_M601` in "Configuration.h".

then it will run marlin algorithm on filament runout?

digant73, 2021-03-14:
Yes

sinole1987, 2021-03-14:
BTW, thank you guys for the hard work that you do on  this page. IF it wasn't for yall, i would have gone back to Prusa LCD screen. much love

sarvenn, 2021-03-16:
> For the moment disable `#define NOZZLE_PAUSE_M600_M601` in "Configuration.h".

Could it be added to config.ini and feature menu to choose marlin or emulated m600?

digant73, 2021-03-16:
@my PR with making emulated M600 configurable on config.ini was merged on BTT main branch so you can use it

oldman4U, 2021-03-17:
Hi sinole

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-03-17 by sinole1987